espressif/esp-idf #13307: [ESP32 S3] Multi hubs and multi devices through wifi

Docfch, 2024-03-04:
### Is your feature request related to a problem?

This library can't work as multi hubs and multi devices and establish wifi communication.

### Describe the solution you'd like.

i want to establish a multi hubs and multi devices through wifi communication

Xiehanxin, 2024-03-04:
hi @Docfch could you please provide more details? like logs

Docfch, 2024-03-04:
Hi @Xiehanxin ,
I am currently working on a project where I use a USB cable for communication between my stm32s in vendor-specific device mode connected to hubs and a PC software with libusb, and I want to replace it with a connection of my board to the USB of the esp32 s3 to communicate through WiFi. (see image below)

[image]

igrr, 2024-03-04:
For a single hub layer, this is currently in progress over at https://github.com/espressif/esp-idf/issues/12554 — you can subscribe to the issue to get notified about the updates. We currently do not have support for nested hubs on our roadmap, though.

tore-espressif, 2025-04-07:
USB HUBs, including nested HUBs, are now supported in esp-idf.

Keep in mind that ESP32-S2/S3 have only 8 USB channels, so in case you need nested HUBs, ESP32-P4, with 16 channels, is much more suitable.

Feel free to reopen if you have more questions

closed 2025-04-07 by tore-espressif