espressif/esp-idf #13308: esp32 esp32 s3 gatt server_table 与手机没法配对

huangdiajd, 2024-03-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

4.4.2

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

esp32 gatt server_table 与手机没法配对

### What is the actual behavior?

esp32 gatt server_table 与手机没法配对

### Steps to reproduce.

esp32 gatt server_table 与手机没法配对

### Build or installation Logs.

```plain
esp32 gatt server_table 与手机没法配对
```

### More Information.

esp32 gatt server_table 与手机没法配对****

zhp0406, 2024-03-07:
@huangdiajd 
请使用 esp-idf/examples/bluetooth/bluedroid/ble/gatt_security_server 与手机配对。
esp-idf/examples/bluetooth/bluedroid/ble/gatt_server 和 esp-idf/examples/bluetooth/bluedroid/ble/gatt_server_service_table 没有打开 SMP。

zhp0406, 2024-03-08:
@huangdiajd 
你用 esp-idf/examples/bluetooth/bluedroid/ble/gatt_security_server 试了吗？

zhp0406, 2024-03-08:
@huangdiajd 
esp-idf/examples/bluetooth/bluedroid/ble/gatt_server 和 esp-idf/examples/bluetooth/bluedroid/ble/gatt_server_service_table  直接打开 SMP 也是不行的，因为还没有设置 SMP 相关的参数

zhp0406, 2024-03-08:
@huangdiajd 
如果你使用  esp-idf/examples/bluetooth/bluedroid/ble/gatt_security_server  也有问题，请提供一下  esp-idf/examples/bluetooth/bluedroid/ble/gatt_security_server  ESP32* 端的 log，谢谢

Indastri, 2024-04-23:
Solution for the issue was provided as there was no reply from the user closing this issue. If you encounter any further issues feel free to reopen this or create a new issue.

Cheers!

closed 2024-04-23 by Indastri