espressif/esp-idf #13310: ESP32C6 - WiFi 6 - Targeted Wait Time (TWT) - Require Functions: Wakeup Event / Auto Light Sleep Duration

djangoa, 2024-03-04:
### Is your feature request related to a problem?

After many days or trying to get TWT functional, I just can't get the current implementation to be useful and save enough power to warrant using it over deep sleep. This has been highlighted indirectly previously several months ago:

"Until now, we don't have a way to get the light sleep duration and the next wake up trigger. And we will enable a twt wakeup event, which is currently in progress." _Originally posted by @xuxiao111 in https://github.com/espressif/esp-idf/issues/12441#issuecomment-1775029218_

### Describe the solution you'd like.

New functionality:

1) Providing a WIFI_EVENT (e.g. "WIFI_EVENT_ITWT_AWAKE") enabling an application to register a callback function via esp_event_handler_instance_register() when WiFi is awake after a TWT has expired.

2) Providing a function to ascertain the remaining automatic light sleep duration until next targeted wakeup.

### Describe alternatives you've considered.

I've tried working round this issue using functions in _esp_wifi/include/esp_wifi_he.h_ such as:

_esp_err_t esp_wifi_sta_itwt_teardown(int flow_id);
esp_err_t esp_wifi_sta_itwt_setup(wifi_twt_setup_config_t *setup_config);_

and

_esp_err_t esp_wifi_sta_itwt_suspend(int flow_id, int suspend_time_ms);_

to temporarily disable TWT but neither work. Invoking the first 2 functions to teardown and then setup again TWT more than once causes the Wifi station to disconnect. Using suspend disables TWT but never re-enables when using either the suspend_time_ms or calling the function again with suspend_time_ms=0 as per the function brief. Both are probably bugs that warrant their own reports?

### Additional context.

Is there any indication when this functionality will be complete?

djangoa, 2024-03-06:
In reference to https://github.com/espressif/esp-idf/issues/13312#issuecomment-1980948036, I've managed to create a workaround to this issue using the _actual_suspend_time_ms_ when invoking _esp_wifi_sta_itwt_suspend(int flow_id, int suspend_time_ms);_ to ascertain the remaining sleep time until the net targeted wake.

It's not ideal as I have to calculate the next remaining sleep time by adding the previous _actual_suspend_time_ms_ to the TWT wake interval and subtracting the max execution time of my application and the call to _esp_wifi_sta_itwt_suspend()_. Additionally to calculate the time I have no choice but to suspend.

I've also managed to get the _esp_wifi_sta_itwt_teardown()_ and _esp_wifi_sta_itwt_setup()_ to function with out causing superfluous disconnects. I'm also not sure how either as previously wifi would reliably disconnect and reconnect when tearing down, invoking a TCP connection and setting up a TWT agreement after a single iteration. Maybe it's a timing issue when tearing down too close to a TWT interval expiring?

xuxiao111, 2024-03-07:
Hi @djangoa,
we will provide a WIFI_EVENT (WIFI_EVENT_TWT_AWAKE) when WiFi is awake. And now it‘s in the process of code review. we will merge it ASAP.

djangoa, 2024-03-07:
@xuxiao111 Thanks for the update, looking forward to the result.

tom-van, 2025-03-30:
Any progress in this issue? I would appreciate WIFI_EVENT_TWT_AWAKE as well.

xuxiao111, 2025-03-31:
Hi @tom-van, the TWT wakeup event functionality has been supported in commit 27f61966cdd65da99138eaa61b4c02ed5b75fcb0. You can call “esp_err_t esp_wifi_sta_twt_config(wifi_twt_config_t *config);” to enable this feature.

gbarranovac, 2025-05-22:
Dear All, how are you?

I am currently using ESP-IDF 5.4.1 at my ESP32C6 but I cant make the WIFI_EVENT_TWT_AWAKE work.

I am using the itwt example and did the following changes:

```
Added in wifi_itwt(void) function with other handlers:
      ESP_ERROR_CHECK(esp_event_handler_instance_register(WIFI_EVENT,
                      WIFI_EVENT_TWT_WAKEUP,
                      &itwt_wkp_handler,
                      NULL,
                      NULL));
```

And added the handler function:
```
static void itwt_wkp_handler(void *arg, esp_event_base_t event_base,
                               int32_t event_id, void *event_data)
{
    wifi_event_sta_twt_wakeup_t *wake = (wifi_event_sta_twt_wakeup_t *) event_data;
    ESP_LOGI(TAG, "<WIFI_EVENT_ITWT_WAKEUP>flow_id:%d, twt_type:%d", wake->flow_id, wake->twt_type);
}
```

The problem is that it nevers enters the handler function. Am I doing something wrong?

xuxiao111, 2025-05-23:
Hi @gbarranovac,
the current design logic defaults to not posting TWT event notifications. To enable them, you need to call "esp_wifi_sta_twt_config". You can refer to the following configuration as an example.

[image]