espressif/esp-idf #13311: ESP-IDF example i2s_recorder failing on opening SDcard file

jessica-brown, 2024-03-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2

### Espressif SoC revision.

esp32

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-WROOM-32D

### Power Supply used.

USB

### What is the expected behavior?

I am attempting to run the esp-idf example i2s_recorder. I can flash the code but get an error opening a file to write to. 

### What is the actual behavior?

When the code attempts to open a file it fails with the following:
[image]

The code function where it fails is:

```
void record_wav(uint32_t rec_time)
{
    // Use POSIX and C standard library functions to work with files.
    int flash_wr_size = 0;
    ESP_LOGI(TAG, "Opening file");

    uint32_t flash_rec_time = BYTE_RATE * rec_time;
    const wav_header_t wav_header =
        WAV_HEADER_PCM_DEFAULT(flash_rec_time, 16, CONFIG_EXAMPLE_SAMPLE_RATE, 1);

    // First check if file exists before creating a new file.
    struct stat st;
    if (stat(SD_MOUNT_POINT"/record.wav", &st) == 0) {
        // Delete it if it exists
        unlink(SD_MOUNT_POINT"/record.wav");
    }

    // Create new WAV file
    FILE *f = fopen(SD_MOUNT_POINT"/record.wav", "a");
    if (f == NULL) {
        ESP_LOGE(TAG, "Failed to open file for writing");
        return;
    }
```

### Steps to reproduce.

I am flashing the code on a Ubuntu virtual machine (Oracle VM VirtualBox) hosted on Windows. 
The ESP32 is connected to the host via USB. 
The SD card is a Netac 8GB microSDHC. 
The SD card adapter I am using is a Adafruit Micro SD SPI or SDIO Card Breakout Board - 3V. 
I have no pull up resistors on the SD card lines currently.

igrr, 2024-03-04:
Could you please check if the workaround suggested in this comment helps? https://github.com/espressif/esp-idf/issues/12968#issuecomment-1888668680

jessica-brown, 2024-03-04:
Unfortunately this didn't work. 
I edited the esp-idf/components/driver/sdmmc/sdmmc_host.c file line to increase the timeout to 500 ms and rebuilt the project but there was no change to the error.

maxbalan, 2024-03-10:
I am having similar issue with the example from the repo for SD card
tried lowering the speed to 4MHz and increase command timeout to 1s 

**Note:** if I take out the SD card and put it back in the first run of the app will run just fine, but as soon as I RST and re-run the app it will fail

```
#define PIN_NUM_MISO  13
#define PIN_NUM_MOSI  11
#define PIN_NUM_CLK   12
#define PIN_NUM_CS    10
```

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037faba
0x4037faba: spi_ll_enable_mosi at /home/max/esp/esp-idf/components/hal/esp32s3/include/hal/spi_ll.h:1001
 (inlined by) spi_hal_prepare_data at /home/max/esp/esp-idf/components/hal/spi_hal_iram.c:180

SPIWP:0xee
mode:DIO, clock div:4
load:0x3fce3818,len:0x18e4
load:0x403c9700,len:0x4
load:0x403c9704,len:0xda4
load:0x403cc700,len:0x3084
entry 0x403c9934
I (41) boot: ESP-IDF v5.1.2-dirty 2nd stage bootloader
I (41) boot: compile time Mar 10 2024 15:00:33
I (41) boot: Multicore bootloader
I (45) boot: chip revision: v0.2
I (48) qio_mode: Enabling QIO for flash chip WinBond
I (54) boot.esp32s3: Boot SPI Speed : 20MHz
I (59) boot.esp32s3: SPI Mode       : QIO
I (64) boot.esp32s3: SPI Flash Size : 16MB
I (68) boot: Enabling RNG early entropy source...
I (74) boot: Partition Table:
I (77) boot: ## Label            Usage          Type ST Offset   Length
I (85) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (92) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (100) boot:  2 factory          factory app      00 00 00010000 00100000
I (107) boot: End of partition table
I (111) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=0e710h ( 59152) map
I (136) esp_image: segment 1: paddr=0001e738 vaddr=3fc93200 size=018e0h (  6368) load
I (139) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=2b5f0h (177648) map
I (191) esp_image: segment 3: paddr=0004b618 vaddr=3fc94ae0 size=010f0h (  4336) load
I (193) esp_image: segment 4: paddr=0004c710 vaddr=40374000 size=0f1ach ( 61868) load
I (223) boot: Loaded app from partition at offset 0x10000
I (223) boot: Disabling RNG early entropy source...
I (234) cpu_start: Multicore app
I (235) cpu_start: Pro cpu up.
I (235) cpu_start: Starting app cpu, entry point is 0x4037538c
0x4037538c: call_start_cpu1 at /home/max/esp/esp-idf/components/esp_system/port/cpu_start.c:157

I (0) cpu_start: App cpu up.
I (253) cpu_start: Pro cpu start user code
I (253) cpu_start: cpu freq: 160000000 Hz
I (254) cpu_start: Application information:
I (256) cpu_start: Project name:     sd-card
I (261) cpu_start: App version:      1
I (266) cpu_start: Compile time:     Mar 10 2024 15:00:30
I (272) cpu_start: ELF file SHA256:  dcd29f1791073649...
I (278) cpu_start: ESP-IDF:          v5.1.2-dirty
I (283) cpu_start: Min chip rev:     v0.0
I (288) cpu_start: Max chip rev:     v0.99 
I (293) cpu_start: Chip rev:         v0.2
I (298) heap_init: Initializing. RAM available for dynamic allocation:
I (305) heap_init: At 3FC964B8 len 00053258 (332 KiB): DRAM
I (311) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (318) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (324) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (332) spi_flash: detected chip: winbond
I (335) spi_flash: flash io: qio
I (339) sleep: Configure to isolate all GPIO pins in sleep state
I (346) sleep: Enable automatic switching of GPIO sleep configuration
I (353) app_start: Starting scheduler on CPU0
I (358) app_start: Starting scheduler on CPU1
I (358) main_task: Started on CPU0
I (368) main_task: Calling app_main()
I (368) example: Initializing SD card
I (378) example: Using SPI peripheral
I (378) example: Mounting filesystem
I (388) gpio: GPIO[10]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (538) example: Filesystem mounted
Name: 
Type: SDIO
Speed: 4.00 MHz (limit: 4.00 MHz)
Size: 7680MB
CSD: ver=2, sector_size=512, capacity=15728640 read_bl_len=9
SCR: sd_spec=2, bus_width=5
I (548) example: Opening file /sdcard/hello.txt
I (588) example: File written
E (3598) sdmmc_cmd: sdmmc_erase_sectors: sdmmc_send_cmd (ERASE) returned 0x107
E (3598) diskio_sdmmc: sdmmc_erase_sectors failed (263)
E (3648) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x108
E (3648) diskio_sdmmc: sdmmc_write_blocks failed (264)
I (3648) example: Renaming file /sdcard/hello.txt to /sdcard/foo.txt
E (3698) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x108
E (3698) diskio_sdmmc: sdmmc_write_blocks failed (264)
E (3698) example: Rename failed
I (3698) main_task: Returned from app_main()
```

any subsequent tries to RST and re-run the app prints this 

```
I (377) main_task: Started on CPU0
I (387) main_task: Calling app_main()
I (387) example: Initializing SD card
I (397) example: Using SPI peripheral
I (397) example: Mounting filesystem
I (407) gpio: GPIO[10]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (617) sdmmc_sd: sdmmc_init_sd_if_cond: send_if_cond (1) returned 0x108
E (617) vfs_fat_sdmmc: sdmmc_card_init failed (0x108).
I (617) gpio: GPIO[10]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (627) example: Failed to initialize the card (ESP_ERR_INVALID_RESPONSE). Make sure SD card lines have pull-up resistors in place.
I (637) main_task: Returned from app_main()
```

jessica-brown, 2024-03-10:
Hi, where did you change the sd card speed?

jessica-brown, 2024-03-10:
@maxbalan also how did you change the default pin configuration in the SD card example? Thanks

maxbalan, 2024-03-12:
hi @jessica-brown both speed and time can be changed using the `host` instance

```
    sdmmc_host_t host = SDSPI_HOST_DEFAULT();
    host.max_freq_khz = 4000;
```

chaning the PINs in this example you have a couple options:
- can use either `menuconfig` and set desired params 
- you can edit the `kconfig` file and set them as default values there, check the link >  https://github.com/espressif/esp-idf/blob/release/v5.2/examples/storage/sd_card/sdspi/main/Kconfig.projbuild
- and you can just set them manually in the code here or the link above > https://github.com/espressif/esp-idf/blob/release/v5.2/examples/storage/sd_card/sdspi/main/sd_card_example_main.c#L28

maxbalan, 2024-03-12:
hi @igrr I have a hunch on what is going on, haven't investigated it much but the SD card (SPI) example was working fine a couple months ago, and since the first run works fine and subsequent fail now since I upgraded to latest version, unfortunatly did not took a note of what version I was running before.

I tried multiple SD cards and after running the example and all failing, my PC stopped recognising the SD card saying it was corrupted and required a format, so this points me that something must've changed in the library and communication protocol that is corrupting the card after first run. 

Just an idea. I will have to start dawngrading the IDF version I am using and see if it helps, that might prove my point.

xuhongv, 2024-03-21:
> hi @igrr I have a hunch on what is going on, haven't investigated it much but the SD card (SPI) example was working fine a couple months ago, and since the first run works fine and subsequent fail [...]

I meet the same question.