espressif/esp-idf #13312: ESP32-C6 - WiFi 6 - Target Wait Time / "itwt" Example - Suspend time incorrect and does not resume

djangoa, 2024-03-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.2

### Espressif SoC revision.

ESP32-C6-MINI-1

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

SparkFun Qwiic Pocket Development Board - ESP32-C6

### Power Supply used.

USB

### What is the expected behavior?

When running the itwt example (or executing `esp_wifi_sta_itwt_suspend(flow_id, suspend_time_ms);` in your application) with power management turned off and the wake duration set to approximately 30 seconds, issuing a suspend command on the CLI for 10s should suspend TWT for ~10s. Issuing two suspend commands with a _suspend_time_ms=0_ should suspend indefinitely with a subsequent suspend command resuming TWT.

### What is the actual behavior?

Suspend time is ~246.996s instead of ~10 seconds. Issuing a second suspend command with _suspend_time_ms=0_ does not resume and errors with "already suspended". See INFO in Debug Logs for details.

### Steps to reproduce.

1. Set the following in the sdkconfig:
```
- # CONFIG_PM_ENABLE is not set
- CONFIG_EXAMPLE_ITWT_WAKE_INVL_EXPN=13
- CONFIG_EXAMPLE_ITWT_WAKE_INVL_MANT=36621
```

3. Build & Flash
4. Issue on the CLI:
```
itwt --suspend=1 -i 1 -s 10000
itwt --suspend=1 -i 1 -s 0
itwt --suspend=1 -i 1 -s 0
```

### Debug Logs.

```plain
I (2610) itwt: <WIFI_EVENT_ITWT_SETUP>twt_id:0, flow_id:1, trigger-enabled, announced, wake_dura:255, wake_dura_unit:0, wake_invl_e:13, wake_invl_m:36621
I (2622) itwt: <WIFI_EVENT_ITWT_SETUP>target wake time:2105092931584, wake duration:65280 us, service period:299999232 us
...
itwt> itwt --suspend=1 -i 1 -s 10000
I (49746) twt: (itwt)suspend, flow_id:1, all_twt:0, suspend:10000 ms, err:0x0
I (49746) itwt: <WIFI_EVENT_ITWT_SUSPEND>status:0, flow_id_bitmap:0x2, actual_suspend_time_ms:[0 246996 0 0 0 0 0 0]
itwt> W (296742) wifi:(itwt)suspend_fid_bitmap:0x2, resume_fid_bitmap:0x2
...
itwt> itwt --suspend=1 -i 1 -s 0
I (422736) twt: (itwt)suspend, flow_id:1, all_twt:0, suspend:0 ms, err:0x0
I (422737) itwt: <WIFI_EVENT_ITWT_SUSPEND>status:0, flow_id_bitmap:0x2, actual_suspend_time_ms:[0 0 0 0 0 0 0 0]
itwt> itwt --suspend=1 -i 1 -s 0
E (426972) wifi:(itwt)flow_id:1 is already suspended
I (426973) itwt: <WIFI_EVENT_ITWT_SUSPEND>status:258, flow_id_bitmap:0x0, actual_suspend_time_ms:[0 0 0 0 0 0 0 0]
I (426974) twt: (itwt)suspend, flow_id:1, all_twt:0, suspend:0 ms, err:0x102
...
```

### More Information.

With the default (255 << 10) ~500ms wait time , the suspend time of **_9780_** is correct:
```
itwt> itwt --suspend=1 -i 1 -s 10000
I (10328) twt: (itwt)suspend, flow_id:1, all_twt:0, suspend:10000 ms, err:0x0
I (10328) itwt: <WIFI_EVENT_ITWT_SUSPEND>status:0, flow_id_bitmap:0x2, actual_suspend_time_ms:[0 9780 0 0 0 0 0 0]
itwt> W (20108) wifi:(itwt)suspend_fid_bitmap:0x2, resume_fid_bitmap:0x2
```
but still exhibits the same bug when issuing two suspend commands with _suspend_time_ms=0_.

djangoa, 2024-03-06:
Update on this issue, after further experimentation with different TWT wake intervals and:

_/**
  \* @brief     Send a TWT Information frame to AP for suspending/resuming established iTWT agreements.
  \*
  \* @param[in]    flow_id The value range is [0, 7].
  \*                       FLOW_ID_ALL indicates suspend all individual TWT agreements
  \* @param[in]    suspend_time_ms If the value is 0, indicates the specified flow_id or all established agreements will be suspended until resume by users.
  \*                               If the value is greater than 0, indicates the specified flow_id or all established agreements will be suspended until suspend_time_ms timeout, unit: ms.
  \*
  \* @return
  \*    - ESP_OK: succeed
  \*    - ESP_ERR_WIFI_NOT_INIT: WiFi is not initialized by esp_wifi_init
  \*    - ESP_ERR_WIFI_NOT_STARTED: WiFi is not started by esp_wifi_start
  \*    - ESP_ERR_WIFI_CONN: WiFi internal error, station or soft-AP control block wrong
  \*    - ESP_ERR_WIFI_NOT_ASSOC: WiFi is not associated
  \*    - ESP_ERR_INVALID_ARG: invalid argument
  \*/
esp_err_t esp_wifi_sta_itwt_suspend(int flow_id, int suspend_time_ms);_

The _actual_suspend_time_ms_ returned inside the suspension event handler for a given flow id:

_wifi_event_sta_itwt_suspend_t *suspend = (wifi_event_sta_itwt_suspend_t *) event_data;
uint32_t actual_suspend_time_ms = suspend->actual_suspend_time_ms[flow_id];_

won't be less than or is rounded to the remaining time to the next wake interval. So if the service period is "299999232us" and you try and suspend for 10000ms and the remaining time left until the next wake interval is > 10000ms then you suspend for longer than anticipated. Furthermore, I assume that when _suspend_time_ms_ is 0 a consecutive call will return "_already suspended_" if the nearest wake interval has yet to expire.

This isn't explained clearly in the Brief, API or documentation and should probably be updated to avoid confusion for others.

xuxiao111, 2024-03-07:
Hi @djangoa, Thanks for reporting.
Q1: the “issuing two suspend commands with a suspend_time_ms=0”
For our design, the "suspend_time = 0" means suspended until the TWT session is resumed, and if you set suspend time twice, will return the error "flow_id:X is already suspended".  And you can set "suspend_time = 10" to resume the TWT session.

Q2: the actual suspend time is incorrect
The actual suspend time depends on whether the AP supports Flexible TWT Schedule. If it is supported, the actual suspend time will be the number you set. If not, will consider the twt service period.

djangoa, 2024-03-07:
@xuxiao111 Thank you for taking time to explain the implementation and for all your work on enabling TWT.

If you are able to add this information to the brief of the function in a subsequent release, I think that would help future users avoid confusion.

Regarding Q2, is it possible to provide a function to figure out if an AP supports a "Flexible TWT Schedule"?

xuxiao111, 2024-03-08:
Hi @djangoa, thanks for your advice.
we will add this information.
And for "Flexible TWT Schedule", maybe it's better to add it to "wifi_event_sta_itwt_suspend_t".