espressif/esp-idf #13313: esp32-s3 i2c failure in esp-idf 5.2

ahemberger, 2024-03-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2

### Espressif SoC revision.

esp32-s3

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

n/a

### Power Supply used.

USB

### What is the expected behavior?

I recently upgraded my setup from esp-idf 5.1.2 to 5.2, and am trying to write a driver for a magnetic encoder that configures the chip via i2c. This code/driver runs fine on an esp32-c3 seeed module I have, but fails the same way on two different esp32-s3 boards I've tried. I don't understand the error that's being offered. I am using:

`#include "driver/i2c_master.h"`

and the failure occurs when I try creating a new master bus like so:

```
    // https://docs.espressif.com/projects/esp-idf/en/v5.2/esp32/api-reference/peripherals/i2c.html#i2c_master_controller
    // We are configuring the i2c master bus first.
    i2c_master_bus_config_t i2c_master_config = {
        .clk_source = I2C_CLK_SRC_DEFAULT,
        .i2c_port = I2C_MASTER_PORT,
        .scl_io_num = I2C_MASTER_SCL_IO,        // SCL pin
        .sda_io_num = I2C_MASTER_SDA_IO,        // SCL pin
        .glitch_ignore_cnt = 7,                 // a default from esp's docs
        .flags.enable_internal_pullup = true,
    };

    i2c_master_bus_handle_t bus_handle;
    i2c_new_master_bus(&i2c_master_config, &bus_handle);
```
For other boards, the above code runs fine - a new master bus handle is created and I can probe it for i2c devices. 

### What is the actual behavior?

Instead of a new master bus handle, my code seems to fail when trying to create the new master bus, providing an output message that I don't find helpful for troubleshooting. It does not crash outright, but instead provides this error message:

```
assert failed: xTaskRemoveFromEventList tasks.c:3826 (pxUnblockedTCB)

Backtrace: 0x403758e2:0x3fc9a2b0 0x40379ed1:0x3fc9a2d0 0x40380371:0x3fc9a2f0 0x4037bec2:0x3fc9a410 0x4037a744:0x3fc9a430 0x4037a7f6:0x3fc9a470 0x40376865:0x3fc9a490 0x40376945:0x3fc9a4c0 0x420082b3:0x3fc9a4e0 0x42006e61:0x3fc9a500 0x420069d2:0x3fc9a520 0x4200f77e:0x3fc9a540 0x4200ef4d:0x3fc9a560 0x4200efaa:0x3fc9a580 0x4200f5e9:0x3fc9a5b0 0x420136b7:0x3fc9a5e0 0x4201307a:0x3fc9a600 0x4200f7f5:0x3fc9a920 0x4201b565:0x3fc9a950 0x403800e5:0x3fc9a980 0x42009b8d:0x3fc9a9d0 0x4200b6fe:0x3fc9aa30 0x4200a925:0x3fc9aa90 0x4200ae88:0x3fc9aac0 0x4200834c:0x3fc9ab10 0x4037a8c1:0x3fc9ab60
0x403758e2: panic_abort at /Users/me/dev/github/esp-idf-v5.2/components/esp_system/panic.c:472
0x40379ed1: esp_system_abort at /Users/me/dev/github/esp-idf-v5.2/components/esp_system/port/esp_system_chip.c:93
0x40380371: __assert_func at /Users/me/dev/github/esp-idf-v5.2/components/newlib/assert.c:81
[... 111 lines cut ...]
0x420082b3: uart_write at /Users/me/dev/github/esp-idf-v5.2/components/vfs/vfs_uart.c:209
0x42006e61: console_write at /Users/me/dev/github/esp-idf-v5.2/components/vfs/vfs_console.c:73 (discriminator 1)
0x420069d2: esp_vfs_write at /Users/me/dev/github/esp-idf-v5.2/components/vfs/vfs.c:482 (discriminator 4)
0x4200f77e: __swrite at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin21.1/xtensa-esp-elf/src/newlib/newlib/libc/stdio/stdio.c:94
0x4200ef4d: __sflush_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin21.1/xtensa-esp-elf/src/newlib/newlib/libc/stdio/fflush.c:224
0x4200efaa: _fflush_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin21.1/xtensa-esp-elf/src/newlib/newlib/libc/stdio/fflush.c:278
 (inlined by) _fflush_r at /builds/idf/crosstool-NG/.build/src/newlib-git-af24ceb7/newlib/libc/include/stdio.h:417
0x4200f5e9: __sfvwrite_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin21.1/xtensa-esp-elf/src/newlib/newlib/libc/stdio/fvwrite.c:251
0x420136b7: __sprint_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin21.1/xtensa-esp-elf/src/newlib/newlib/libc/stdio/vfiprintf.c:429
 (inlined by) __sprint_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin21.1/xtensa-esp-elf/src/newlib/newlib/libc/stdio/vfprintf.c:399
0x4201307a: _vfprintf_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin21.1/xtensa-esp-elf/src/newlib/newlib/libc/stdio/vfprintf.c:1774 (discriminator 1)
0x4200f7f5: vprintf at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin21.1/xtensa-esp-elf/src/newlib/newlib/libc/stdio/vprintf.c:34
0x4201b565: esp_log_writev at /Users/me/dev/github/esp-idf-v5.2/components/log/log.c:215
0x403800e5: esp_log_write at /Users/me/dev/github/esp-idf-v5.2/components/log/log.c:225
0x42009b8d: gpio_config at /Users/me/dev/github/esp-idf-v5.2/components/driver/gpio/gpio.c:408 (discriminator 1)
0x4200b6fe: i2c_common_set_pins at /Users/me/dev/github/esp-idf-v5.2/components/driver/i2c/i2c_common.c:235
0x4200a925: i2c_param_master_config at /Users/me/dev/github/esp-idf-v5.2/components/driver/i2c/i2c_master.c:642
0x4200ae88: i2c_new_master_bus at /Users/me/dev/github/esp-idf-v5.2/components/driver/i2c/i2c_master.c:799
0x4200834c: i2c_scan_task at /Users/me/dev/esp/i2c-scan/main/i2c-scan.c:62
0x4037a8c1: vPortTaskWrapper at /Users/me/dev/github/esp-idf-v5.2/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134
```

### More Information.

I have tried specifying different pins for I2C, and tried a few different esp32-s3 boards from different manufacturers, all with the same result.

mythbuster5, 2024-03-06:
Seems that 2048 in your task create is not sufficient. Please try 4096.

ahemberger, 2024-03-06:
Thank you! Increasing the task allocation to 4096 does indeed make these errors go away. However, my microcontroller still seems unable to communicate via i2c with an attached device. 

When I run the above code on an esp32-c3, I see it output the following:

```
Scanning I2C bus...
Found device at address 54 (0x36)
Write address: 108 (0x6C)
Read address: 109 (0x6D)
```
However, running the same code on an identical circuit setup using an s3 produces:

```
Scanning I2C bus...
Scanning I2C bus...
Scanning I2C bus...
Scanning I2C bus...
Scanning I2C bus...
Scanning I2C bus...
```
I've tried different pins, but can't seem to produce a different result.

mythbuster5, 2024-03-07:
Don't know you are using v5.2.what. But I tried on the latest 5.2 branch esp32s3 probe works fine with SCL=9 SDA=10 port=0/1. So, is it possible to try latest 5.2? And paste logic analyzer on I2C bus if possible. That helps us go forward.

chipweinberger, 2024-03-10:
I'm also have i2c errors in v5.1.2

```
	  > E (19574) i2c: ack error
	  > E (19574) i2c: i2c_master_cmd_begin failed
```

AFAIK this code used to work in v4.4.X. 

Still debugging it.

GordonWen, 2024-03-11:
The problem seems is the device address.
I try it with a I2C oled with ssd1306.
The device is 0b01111000, and the last bit is used for R/W. 
We usually called the device address is 0x78 = 0x01111000. And when I use 0x78 with old i2c driver, it's works. But if I use the 0x78 with new i2c_master driver, it's wrong.
So I changed the address to 0b0111100 = 0x3C, it's works again!
`
i2c_master_bus_config_t i2c_conf = {
		.clk_source = I2C_CLK_SRC_DEFAULT,
		.i2c_port = I2C_NUM_1,
		.scl_io_num = OLED_SCL_PIN,
		.sda_io_num = OLED_SDA_PIN,
		.glitch_ignore_cnt = 7,
		.flags.enable_internal_pullup = true,
	};
	i2c_master_bus_handle_t i2c_bus_handle;
	ESP_ERROR_CHECK(i2c_new_master_bus(&i2c_conf, &i2c_bus_handle));

	i2c_device_config_t i2c_dev_cfg = {
		.dev_addr_length = I2C_ADDR_BIT_LEN_7,
		.device_address = 0x3C,
		.scl_speed_hz = 400000,
	};
	i2c_master_dev_handle_t i2c_oled_handle;
	ESP_ERROR_CHECK(i2c_master_bus_add_device(i2c_bus_handle, &i2c_dev_cfg, &i2c_oled_handle));
`

mythbuster5, 2024-03-14:
@ahemberger Did you success on s3 on latest 5.2?

mythbuster5, 2024-03-14:
Hi, @ahemberger I tried with my esp32s3 board on latest v5.2 with your code and your io pins. That I cannot reproduce the issue. I'm using ssd1306 as a device. The code is almost as same as yours as following:

```
    i2c_master_bus_config_t i2c_mst_config_1 = {
        .clk_source = I2C_CLK_SRC_DEFAULT,
        .i2c_port = 0,
        .scl_io_num = 9,
        .sda_io_num = 10,
        .glitch_ignore_cnt = 7,
        .flags.enable_internal_pullup = true,
    };
    i2c_master_bus_handle_t bus_handle;

    TEST_ESP_OK(i2c_new_master_bus(&i2c_mst_config_1, &bus_handle));
    for (uint8_t addr = 0; addr < 127; addr++) {
        esp_err_t error_status = i2c_master_probe(bus_handle, addr, -1);

            // If we received the ACK bit, we found something! Print its address.
            if (error_status == ESP_OK) {
                printf("Found device at address %d (0x%02X)\n", addr, addr);

                uint8_t write_address = (addr << 1) | 0;
                uint8_t read_address = (addr << 1) | 1;

                printf("Write address: %d (0x%02X)\n", write_address, write_address);
                printf("Read address: %d (0x%02X)\n", read_address, read_address);
                
            }
    }
```
If you still fail on v5.2, it would be helpful if you paste a logic analyzer picture around your truly address, I want to see what really happens.

(If there is no further respond in several days, I'm going to conclude this issue~) Thank you.

ahemberger, 2024-03-14:
Hi! 

I confess I did not previously own a logic analyzer, so I bought one to try to address this request and better-understand what was going on. The answer is ultimately both embarassing and infuriating (TL;DR, idf 5.2 and 5.2.1 both work fine - the initial observation you made about using a larger stack size is actually what unstuck me). But if you're curious:

The esp32-s3 board I have been using is this one from seeed:
https://www.amazon.com/dp/B0BYSB66S5?psc=1&ref=ppx_yo2ov_dt_b_product_details

The pinout diagram (about halfway down the product page) shows the pins along the right edge of the board: D10, D9, D8, D7, which are the pins I was trying to use - actually correspond (rather unintuitively, I think) with GPIO pins 9, 8, 7, and 44). The pins on the esp32-c3 seed board have identically-named digital pins, but the GPIO pins also correspond to them. So, basically, I was accidentally using the incorrect pins and didn't notice until my logic analyzer showed me no communication happening where I expected some to be. 

So, in short, I'm an idiot and your initial fix worked for me. Thank you, and I'm sorry? :)

Alvin1Zhang, 2024-03-17:
Thanks for reporting, and sharing the updates, feel free to reopen.

closed 2024-03-17 by Alvin1Zhang