espressif/esp-idf #13314: idf 5.2 是否可以支持p4芯片

shenxiaozheng, 2024-03-05:
### Is your feature request related to a problem?

[image]

suda-morris, 2024-03-05:
P4 has very **limited preview** support in esp-idf v5.2 release branch, most of the P4 drivers should only work on the FPGA. If you have the esp32p4 **chip**, we suggest you to use the master branch.

BTW, currently esp32p4 is still in the "preview" stage, you should append `--preview` to the `idf.py` command. e.g.
`idf.py --preview set-target esp32p4`

shenxiaozheng, 2024-03-05:
我在p4上烧写hello_world程序，发生下面的错误，请指点
A fatal error occurred: Failed to write to target Flash after seq 0 (result was 01050000: Requested resource not found)
CMake Error at run_serial_tool.cmake:66 (message):
  
  /home/test/.espressif/python_env/idf5.2_py3.9_env/bin/python;;/home/test/esp/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32p4
  failed.

dobairoland, 2024-03-06:
I suggest to close this in favor of https://github.com/espressif/esp-idf/issues/13323. Newer esptool versions with full ESP32-P4 support will eventually propagate to v5.2 as well but given that v5.2 will have only limited support for ESP32-P4 in general, it is not recommended to target it with v5.2.

closed 2024-03-07 by suda-morris