espressif/esp-idf #13315: Flash from within docker fails

SimonPVS, 2024-03-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi

I am using esp-idf 5.1.2 and i followed the 'using-docker-container' tutorial. 
It is possible to build a project inside the docker, but i am not able to flash the binary to my device.
The host os is windows 10.

Flashing fails with:

` *  Executing task: /opt/esp/python_env/idf5.1_py3.8_env/bin/python /opt/esp/idf/components/esptool_py/esptool/esptool.py -p /dev/ttyUSB1 -b 460800 --before default_reset --after hard_reset --chip esp32 write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB 0x1000 bootloader/bootloader.bin 0x10000 COM1xx.bin 0x8000 partition_table/partition-table.bin 

esptool.py v4.7.dev3
Serial port /dev/ttyUSB1

A fatal error occurred: Could not open /dev/ttyUSB1, the port is busy or doesn't exist.
([Errno 13] could not open port /dev/ttyUSB1: [Errno 13] Permission denied: '/dev/ttyUSB1')

Hint: Try to add user into dialout group: sudo usermod -a -G dialout $USER

 *  The terminal process "/opt/esp/python_env/idf5.1_py3.8_env/bin/python '/opt/esp/idf/components/esptool_py/esptool/esptool.py', '-p', '/dev/ttyUSB1', '-b', '460800', '--before', 'default_reset', '--after', 'hard_reset', '--chip', 'esp32', 'write_flash', '--flash_mode', 'dio', '--flash_freq', '40m', '--flash_size', '2MB', '0x1000', 'bootloader/bootloader.bin', '0x10000', 'COM1xx.bin', '0x8000', 'partition_table/partition-table.bin'" terminated with exit code: 2. `

What can I try to resolve this?

dobairoland, 2024-03-05:
Please see the note in the https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-docker-image.html#using-the-image-interactively section. Using remote serial port is a way to overcome this limitation on Windows.

SimonPVS, 2024-03-05:
Thanks, using the esp tool esp_rfc2217_server.exe it works.

Maybe the tutorial should be updated because this states it should work with usbipd: https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/using-docker-container.md

dobairoland, 2024-03-05:
cc @brianignacio5

brianignacio5, 2024-03-06:
Are you sure that the serial port was properly bind and attached to WSL with usbipd? You can use the `ESP-IDF: Select port to use` command to see available serial ports. I have tested it recently and it does work as long as the device was properly bind and attached.

Will add documentation using esp tool esp_rfc2217 for an alternative solution to usbipd.

SimonPVS, 2024-03-06:
Apologies for using the wrong repository.

The COM port seemed to be attached to WSL. (I checked it as proposed in the tutorial)
But I think there is something wrong permission wise. I could not figure out what..