espressif/esp-idf #13316: esp-idf make defconfig uses `env/esp-idf/components/mdns/Kconfig` when `/components/mdns/KConfig` is available

rretanubun, 2024-03-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.6-1-gb550086d0e

### Operating System used.

Linux

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

The `KConfig` options and limits specified in `/components/mdns/KConfig` should be used when present instead of `env/esp-idf/components/mdns/Kconfig` that came with esp-idf

### What is the actual behavior?

During the build, the limit is set back to the `default` value of the `env/esp-idf/components/mdns/Kconfig` and the code is built with an unexpected MAX_MDNS_SERVICE

### Steps to reproduce.

1. Copy the mdns component from `env/esp-idf/components/mdns` to `/components/mdns`
2. Change the CONFIG_MDNS_MAX_SERVICES  in `/components/mdns/KConfig` to have a larger `range` like this
```     config MDNS_MAX_SERVICES
         int "Max number of services"
-        range 1 64
+        range 1 255
         default 10```
3. Create an `sdkconfig.myproject` that sets this value to `227`
4. run these steps. `cp sdkconfig.myproject sdkconfig.defaults && make defconfig`

### Build or installation Logs.

```plain
warning: user value 227 on the int symbol MDNS_MAX_SERVICES (defined at env/esp-idf/components/mdns/Kconfig:3) ignored due to being outside the active range ([1, 64]) -- falling back on defaults
```

igrr, 2024-03-05:
Hi @rretanubun, unfortunately I couldn't reproduce this issue. After following these steps, `make defconfig` has succeeded and I got `CONFIG_MDNS_MAX_SERVICES=227` in the resulting `sdkconfig` file.

Could you please run `make list-components` and verify that the `mdns` component is taken from the project components directory? Can you reproduce this issue also with a `hello_world` example, or only with your custom project?

rretanubun, 2024-03-05:
Hi @igrr : Thanks for the reply. Here is my `make list components` output:
```
make list-components
Python requirements from /home/rretanubun/work/mmb/bayview/env/esp-idf/requirements.txt are satisfied.
GENCONFIG
Loading defaults file /home/rretanubun/work/mmb/bayview/sdkconfig.defaults...
warning: user value 227 on the int symbol MDNS_MAX_SERVICES (defined at env/esp-idf/components/mdns/Kconfig:3) ignored due to being outside the active range ([1, 64]) -- falling back on defaults
App "bayview" version: 0.10.8
*******************************************************************************
COMPONENT_DIRS (components searched for here)
/home/rretanubun/work/mmb/bayview/components
/home/rretanubun/work/mmb/bayview/env/esp-idf/examples/common_components/protocol_examples_common
/home/rretanubun/work/mmb/bayview/env/esp-idf/components
/home/rretanubun/work/mmb/bayview/main
*******************************************************************************
TEST_COMPONENTS (list of test component names)

*******************************************************************************
TEST_EXCLUDE_COMPONENTS (list of test excluded names)
 
*******************************************************************************
COMPONENT_PATHS (paths to all components):
<snip>
/home/rretanubun/work/mmb/bayview/env/esp-idf/components/mdns
```

So yes, the project appears to be `misconfigured`, I was expecting `/home/rretanubun/work/mmb/bayview/components/mdns`
Any hints on where I can check next? (I have not tried with `hello_world` example).

igrr, 2024-03-05:
The list of component paths is built over here: https://github.com/espressif/esp-idf/blob/7a00499cc9ae66a81ab1720a1ca1c50f2b1a04b7/make/project.mk#L205

Looking at this code, I would check that 1) there is a component.mk file in mdns directory, 2) MULTI_COMPONENT_DIRS includes the `components` subdirectory of your project — you can edit the project.mk file to print this variable.

You can also add a debug print of the sub-expression over there which gets the list of candidate paths:
```make
$(foreach cd,$(COMPONENT_DIRS),\
                            $(if $(findstring $(cd),$(MULTI_COMPONENT_DIRS)),\
                                 $(abspath $(dir $(wildcard $(cd)/mdns/component.mk))),)\
                            $(if $(findstring $(cd),$(SINGLE_COMPONENT_DIRS)),\
                                 $(if $(filter mdns,$(notdir $(cd))),$(cd),),)
```
(I have substituted `$(comp)` for `mdns`)

closed 2024-03-06 by rretanubun