espressif/esp-idf #13317: Failed assert in BLE NIMBLE during init: ble_hs_locked_by_cur_task

danfleck, 2024-03-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2

### Espressif SoC revision.

ESP32-S3

### Operating System used.

macOS

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

I expect it to initialize NIMBLE BLE correctly using FreeRTOS: nimble_port_freertos_init in ESP-IDF v5.2. The code works in ESP-IDF v5.1.2. 

### What is the actual behavior?

The code asserts with: assert failed: ble_hs_id_addr ble_hs_id.c:223 (ble_hs_locked_by_cur_task()). See stack trace below.

### Steps to reproduce.

1. Call nimble_port_freertos_init(taskMainFunction)
2. In taskMainFunction call "nimble_port_run()"

### Debug Logs.

```plain
13:28:12 E TRACE 0x40374cf2: /Users/redacted/esp/esp-idf/components/esp_system/panic.c:472
13:28:12 E TRACE 0x40382441: /Users/redacted/esp/esp-idf/components/esp_system/port/esp_system_chip.c:93
13:28:12 E TRACE 0x4038b9a6: /Users/redacted/esp/esp-idf/components/newlib/assert.c:81
13:28:12 E TRACE 0x42019dd2: /Users/redacted/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_id.c:223 (discriminator 1) 
13:28:12 E TRACE 0x42020c16: /Users/redacted/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_pvcy.c:214
13:28:12 E TRACE 0x4201db9e: /Users/redacted/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_startup.c:385
13:28:12 E TRACE 0x4201a40c: /Users/redacted/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:350
13:28:12 E TRACE 0x4201a61f: /Users/redacted/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:676
13:28:12 E TRACE 0x4201a62b: /Users/redacted/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:576
13:28:12 E TRACE 0x420da597: /Users/redacted/esp/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/src/npl_os_freertos.c:451
13:28:12 E TRACE 0x403770ae: /Users/redacted/esp/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:185
<<My code below here>> 

static void BleHostTask(void *param) {
  log("BLE Host Task Started");
  /* This function will return only when nimble_port_stop() is executed */
  nimble_port_run();

  nimble_port_freertos_deinit();
}

nimble_port_freertos_init(BleHostTask); // <--- during this call I get the assert
```

### More Information.

This code works under ESP-IDF version 5.1.2.

rahult-github, 2024-03-05:
Hi @danfleck , recently a [commit](https://github.com/espressif/esp-nimble/commit/4c281ee888f100373420bc7b4fbd0c4da4fb929c) was pushed to release/v5.2 to handle such scenario. Can you please confirm that you are on latest 5.2 code ?

danfleck, 2024-03-05:
Hi Rahul,

Thanks for the quick reply. You are right, that commit is not in the code I have. I just rechecked out using the command: 
`git clone -b v5.2 --recursive https://github.com/espressif/esp-idf.git esp-idf-v5.2`

which is noted as the recommended way to get 5.2 on here: 
[https://github.com/espressif/esp-idf/releases/tag/v5.2](https://github.com/espressif/esp-idf/releases/tag/v5.2)

So this seems like more of a GIT/documentation issue.

rahult-github, 2024-03-06:
Hi @danfleck  , i think the change is merged after the 5.2 release tag you are referring to was made . So when the next 5.2.1 release tag is available, and if pulled, the change would be available directly.

closed 2024-03-06 by rahult-github