espressif/esp-idf #13319: Callback for esp_wifi_80211_tx()

Syphontwo, 2024-03-05:
### Is your feature request related to a problem?

The current implementation of `esp_wifi_80211_tx()` is fire and forget.  The problem is that sometimes the device will employ RTS/CTS in order to deliver the arbitrary frame instead of a direct transmission on the line.  While RTS/CTS is important to prevent collision, there can be confusion to the user about what was actually sent, or if the transmission was even successful.

### Describe the solution you'd like.

`esp_wifi_80211_tx()` should have an optional, or required, callback which will report the status of the transmission.  This callback should include information such as:
Was the transmission successful
was RTS / CTS used
was RTS / CTS successful in delivering the frame

Other useful bits that I would love to see:
Did the receiving device ACK (if that's possible within the current implementation)
What was the checksum of the final frame
what was the time on wire
how many total frames were sent in order to deliver the requested payload
What was the RTS frame, and details of the responding CTS (signal strength, channel, just like the rx_ctrl in a promiscuous callback)

### Describe alternatives you've considered.

Allowing the user direct control over the use of RTS / CTS in order to ensure the exact frame is delivered as desired.  The issue here is users may be confused when they are not able to ever successfully deliver a frame on a busy line.

Making the function blocking.  When `esp_wifi_80211_tx()` is called, it could be a blocking function that will not continue execution until the frame is properly sent and the response could be in the form of `esp_err_t` in line with other functions.  The idea of a blocking function isn't really desirable, and `esp_err_t` isn't really employable without blocking.

### Additional context.

related to this bug report https://github.com/espressif/esp-idf/issues/10668

zhangyanjiaoesp, 2024-03-20:
@Syphontwo we will consider this feature request.

VNovytskyi, 2024-10-15:
I agree with @Syphontwo. It will be great to have some callback for esp_wifi_80211_tx. The callback must be fired when the PHY layer has been transmitted frame to the air. Alternatively, it may be a function like esp_wifi_80211_empty or esp_wifi_80211_ready.
I think there is some function which allows us to do this thing but it is private. I found some functions which possibly can do this, but no documentation has been provided about them:
- ic_register_net80211_tx_cb
- ic_register_rx_cb

Without this function, I can not determine when I can perform the next call of esp_wifi_80211_tx, for example:
```
    esp_err_t err = esp_wifi_80211_tx(WIFI_IF_AP, data, 1500, false);
    if (err != ESP_OK) {
      vTaskDelay(pdMS_TO_TICKS(10));
    }
```
Are there any updates?

mintisan, 2024-12-27:
The addition of this feature will greatly enhance the utility of esp_wifi_80211_tx, highly supportive.

closed 2025-04-14 by espressif-bot