espressif/esp-idf #13320: Host Vendor Class API

Docfch, 2024-03-05:
### Is your feature request related to a problem?

I am currently working on a project where I use a USB cable for communication between my stm32 in vendor-specific device mode and a PC software with libusb, and I want to replace it with a connection of my board to the USB of the esp32 s3 to communicate through WiFi.
For this, I wanted to know when will we have the complete host vendor class API library?

Dazza0, 2024-03-06:
> For this, I wanted to know when will we have the complete host vendor class API library?

That's the [USB Host Library](https://docs.espressif.com/projects/esp-idf/en/v5.2/esp32s3/api-reference/peripherals/usb_host.html#usb-host). All host class drivers are implemented using the USB Host Library API. Thus, any vendor specific class drivers using the same API as well.

Dazza0, 2024-03-19:
@Docfch Closing this issue as there is not Vendor Class API we can provide. If you plan on implementing your own Vendor specific class driver, please refer to the USB Host Library document linked above.