espressif/esp-idf #13321: Nimble init problem on SDK 5.2 commit Merge branch 'bugfix/serial_jtag_v5.2' into 'release/v5.2'

filzek, 2024-03-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Using esp32 there is a problem to init nimble 

The example on SPP Server build and when run there is a failure on init a nimble ble.

Commit
 Merge branch 'bugfix/serial_jtag_v5.2' into 'release/v5.2'

rahult-github, 2024-03-06:
Hi @filzek ,
 
 For the commit mentioned: 
 

> Commit
> Merge branch 'bugfix/serial_jtag_v5.2' into 'release/v5.2'

This maps to the commit : 

> commit 7ec8266e776298686dcaf26ab36b65b08af3b73a (HEAD -> release/v5.2)
> Merge: 4f6cca6f08 194b616c36
> Author: Marius Vikhammer <marius.vikhammer@espressif.com>
> Date:   Mon Mar 4 15:20:20 2024 +0800
> 
>     Merge branch 'bugfix/serial_jtag_v5.2' into 'release/v5.2'
>     
>     fix(usb_serial_jtag): Fix issue that boot failed on esp32c3 (for v5.2)
>     
>     See merge request espressif/esp-idf!29381

I synced to same and tested with esp32 at my end, and i don't see any issue. 

---- 

I (31) boot: ESP-IDF v5.2-305-**g7ec8266e77**-dirty 2nd stage bootloader
I (31) boot: compile time Mar  6 2024 10:10:31
I (32) boot: Multicore bootloader
I (37) boot: chip revision: v1.0
I (40) boot.**esp32**: SPI Speed      : 40MHz
I (45) boot.**esp32**: SPI Mode       : DIO
I (50) boot.**esp32**: SPI Flash Size : 2MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1b68ch (112268) map
I (144) esp_image: segment 1: paddr=0002b6b4 vaddr=3ffbdb60 size=04964h ( 18788) load
I (151) esp_image: segment 2: paddr=00030020 vaddr=400d0020 size=5d188h (381320) map
I (282) esp_image: segment 3: paddr=0008d1b0 vaddr=3ffc24c4 size=00250h (   592) load
I (283) esp_image: segment 4: paddr=0008d408 vaddr=40080000 size=1a008h (106504) load
I (343) boot: Loaded app from partition at offset 0x10000
I (343) boot: Disabling RNG early entropy source...
I (354) cpu_start: Multicore app
I (363) cpu_start: Pro cpu start user code
I (363) cpu_start: cpu freq: 160000000 Hz
I (363) cpu_start: Application information:
I (366) cpu_start: Project name:     spp_server
I (371) cpu_start: App version:      qa-test-v5.2.1-20240223-4-g7ec8
I (378) cpu_start: Compile time:     Mar  6 2024 10:10:26
I (385) cpu_start: ELF file SHA256:  2c5849029...
**I (390) cpu_start: ESP-IDF:          v5.2-305-g7ec8266e77-dirty**
I (396) cpu_start: Min chip rev:     v0.0
I (401) cpu_start: Max chip rev:     v3.99 
I (406) cpu_start: Chip rev:         v1.0
I (411) heap_init: Initializing. RAM available for dynamic allocation:
I (418) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (424) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (430) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (436) heap_init: At 3FFC5830 len 0001A7D0 (105 KiB): DRAM
I (442) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (449) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (455) heap_init: At 4009A008 len 00005FF8 (23 KiB): IRAM
I (463) spi_flash: detected chip: generic
I (466) spi_flash: flash io: dio
W (470) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (484) coexist: coex firmware version: 77cd7f8
I (489) main_task: Started on CPU0
I (499) main_task: Calling app_main()
I (529) BTDM_INIT: BT controller compile version [0f0c5a2]
I (539) BTDM_INIT: Bluetooth MAC: 24:6f:28:80:2c:2a
I (539) phy_init: phy_version 4791,2c4672b,Dec 20 2023,16:06:06
I (8I (859) NimBLE: BLE server UART_task started

I (859) NimBLE: BLE Host Task Started
I (859) main_task: Returned from app_main()
I (899) NimBLE: Device Address: 
I (899) NimBLE: 24:6f:28:80:2c:2a
I (899) NimBLE: 

I (899) NimBLE: GAP procedure initiated: advertise; 
I (909) NimBLE: disc_mode=2
I (909) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=0 adv_itvl_max=0
I (919) NimBLE: 

------- 

So, can you please share more information : 
 1. Please provide console log for what error you observing .
 2. Also please provide your sdkconfig so as to confirm if any extra feature / functionality has been enabled.
 3. Please ensure "git submodule update --init --recursive -f " has been done so as to pull all the needed submodules.

filzek, 2024-03-06:
Now we have update to SDK v5.2.1 as develop still in progress
Nimble has initiate perfect onto this version, so we will close the ticket

closed 2024-03-06 by filzek