espressif/esp-idf #13323: 使用最新的master分支，无法烧写

shenxiaozheng, 2024-03-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

 v5.3-dev-2381-g9f4e8eb0cd

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

~/esp/esp-idf/examples/get-started/hello_world$ idf.py  -p /dev/ttyUSB0 flash 
Executing action: flash
Running ninja in directory /home/test/esp/esp-idf/examples/get-started/hello_world/build
Executing "ninja flash"...
[1/5] cd /home/test/esp/esp-idf/examples/get-started/hello_world/build/esp-idf/esptool_py && /home/test/.espressif/python_env/idf5.3_py3...rted/hello_world/build/partition_table/partition-table.bin /home/test/esp/esp-idf/examples/get-started/hello_world/build/hello_world.bin
hello_world.bin binary size 0x2ab40 bytes. Smallest app partition is 0x100000 bytes. 0xd54c0 bytes (83%) free.
[1/1] cd /home/test/esp/esp-idf/examples/get-started/hello_world/build/bootloader/esp-idf/esptool_py && /home/test/.espressif/python_env...check_sizes.py --offset 0x8000 bootloader 0x2000 /home/test/esp/esp-idf/examples/get-started/hello_world/build/bootloader/bootloader.bin
Bootloader binary size 0x5080 bytes. 0xf80 bytes (16%) free.
[4/5] cd /home/test/esp/esp-idf/components/esptool_py && /snap/cmake/1366/bin/cmake -D IDF_PATH=/home/test/esp/esp-idf -D "SERIAL_TOOL=/...TORY=/home/test/esp/esp-idf/examples/get-started/hello_world/build -P /home/test/esp/esp-idf/components/esptool_py/run_serial_tool.cmake
esptool.py --chip esp32p4 -p /dev/ttyUSB0 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x2000 bootloader/bootloader.bin 0x10000 hello_world.bin 0x8000 partition_table/partition-table.bin
esptool.py vv4.8.dev1
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP32-P4 (revision v0.0)
Features: High-Performance MCU
Crystal is 40MHz
MAC: 00:00:00:00:00:00
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00002000 to 0x00007fff...
Flash will be erased from 0x00010000 to 0x0003afff...
Flash will be erased from 0x00008000 to 0x00008fff...
Compressed 20608 bytes to 12356...
Writing at 0x00002000... (100 %)
Wrote 20608 bytes (12356 compressed) at 0x00002000 in 0.3 seconds (effective 481.2 kbit/s)...
File  md5: 4675a4beb7092ca8f3068faa25e01491
Flash md5: 8fa0a279615eb2b92e2ae11304f8c77d
MD5 of 0xFF is 8fa0a279615eb2b92e2ae11304f8c77d

A fatal error occurred: MD5 of file does not match data in flash!
CMake Error at run_serial_tool.cmake:66 (message):
  
  /home/test/.espressif/python_env/idf5.3_py3.9_env/bin/python;;/home/test/esp/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32p4
  failed.

FAILED: CMakeFiles/flash /home/test/esp/esp-idf/examples/get-started/hello_world/build/CMakeFiles/flash 
cd /home/test/esp/esp-idf/components/esptool_py && /snap/cmake/1366/bin/cmake -D IDF_PATH=/home/test/esp/esp-idf -D "SERIAL_TOOL=/home/test/.espressif/python_env/idf5.3_py3.9_env/bin/python;;/home/test/esp/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32p4" -D "SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;write_flash;@flash_args" -D WORKING_DIRECTORY=/home/test/esp/esp-idf/examples/get-started/hello_world/build -P /home/test/esp/esp-idf/components/esptool_py/run_serial_tool.cmake
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/test/esp/esp-idf/examples/get-started/hello_world/build/log/idf_py_stderr_output_2968 and /home/test/esp/esp-idf/examples/get-started/hello_world/build/log/idf_py_stdout_output_2968

### What is the actual behavior?

idf.py  -p /dev/ttyUSB0 flash

### Steps to reproduce.

1. Step
2. Step
3. Step
...

peterdragun, 2024-03-12:
Hi @shenxiaozheng,
I have tried multiple esp32-p4 and I was not able to reproduce the issue on any of them. Based on the fact that your chip does not have a MAC address set I assume that it is some engineering sample. The response from the flash is all 0xFF, so maybe it is a hardware issue? Do you have any other board that you can try this on? Or can you maybe provide a little more details about your setup? Can you please run `esptool.py flash_id` command if the chip can communicate with the flash properly?

shenxiaozheng, 2024-03-13:
Thank you very much for your answer. It is indeed a hardware issue, and I have resolved it. Thank you for your help.

radimkarnis, 2024-03-13:
@shenxiaozheng could you please share what the issue was? It could help anyone having similar issues in the future. Thank you!

I will close this ticket, I am glad you could resolve it.

closed 2024-03-13 by radimkarnis

shenxiaozheng, 2024-03-14:
This is due to unstable voltage of the flash. Thank you.

jeff-getlucky, 2024-03-14:
> 这是由于闪光灯电压不稳定造成的。谢谢。

可以问一下在哪里能买到p4吗

dobairoland, 2024-03-18:
For the sake of completeness and in case anybody else would get the same issue: burning the `PXA0_TIEH_SEL_0` eFuse to 2 also can help. But better contact the Business Support team before doing this. 

- Please be aware that burning eFuses are an irreversible operation and you might damage your chip.
- This comment is related to only engineering samples of ESP32-P4 which were distributed without pre-burned eFuses. Mass production chips will have all eFuses ready-for-use.

Espressif-yangxi, 2024-03-26:
> > 这是由于闪光灯电压不稳定造成的。谢谢。
> 
> 可以问一下在哪里能买到p4吗

@jeff-getlucky  您好，若您有 P4 产品需求，可[联系乐鑫](https://www.espressif.com/zh-hans/contact-us/sales-questions)客户支持团队获取 https://www.espressif.com/zh-hans/contact-us/sales-questions。我们将竭诚为您提供支持。

Espressif-yangxi, 2024-03-26:
> > 这是由于闪光灯电压不稳定造成的。谢谢。
> 
> 可以问一下在哪里能买到p4吗

@jeff-getlucky 您好，若您有 P4 产品需求，可[联系乐鑫]( https://www.espressif.com/zh-hans/contact-us/sales-questions)客户支持团队获取 https://www.espressif.com/zh-hans/contact-us/sales-questions。我们将竭诚为您提供支持。

Espressif-yangxi, 2024-03-26:
> > 这是由于闪光灯电压不稳定造成的。谢谢。
> 
> 可以问一下在哪里能买到p4吗

@jeff-getlucky  您好，若您有 ESP32-P4 需求，可[联系乐鑫]( https://www.espressif.com/zh-hans/contact-us/sales-questions)客户支持团队。我们将竭诚为您提供支持。