espressif/esp-idf #13324: Get ID's of devices currently connected to Zigbee coordinator

allexoK, 2024-03-06:
### Is your feature request related to a problem?

Hello, 
sorry if this feature is already implemented, I just can't find any info about it.
I can't find a way to check which Zigbee End devices, are currently connected to Esp32-C6 coordinator.

### Describe the solution you'd like.

It would be nice to have some kind of function, which returns IDs of currently connected devices. Usage example is beneficial. 

### Describe alternatives you've considered.

I tried tracking connection and disconnection to the network. The only thing I see is ESP_ZB_ZDO_SIGNAL_DEVICE_ANNCE, that get's triggered, but when I reset coordinator I'dont know which devices are currently in the network.

xieqinan, 2024-04-15:
@allexoK ，

The esp-zigbee currently does not support this feature, but we have plans to implement it in the future. Thank you for your patience.

allexoK, 2024-04-15:
Thanks for the info!

closed 2024-04-15 by allexoK

Shatodor, 2025-02-20:
Any updates on implementing this feature?

xieqinan, 2025-02-20:
@Shatodor ,

The `esp_zb_nwk_get_next_neighbor()` API can be used to check which devices are connected to the current device. This API is supported in `esp-zigbee-lib` and `esp-zboss-lib` version 1.3.2 and above.  

The latest `zigbee-lib` version is 1.6.3. Please refer to the [release notes](https://github.com/espressif/esp-zigbee-sdk/blob/main/RELEASE_NOTES.md) for more details.