espressif/esp-idf #13325: In USB Host mode, can the ESP32 format the Udisk?

rtek1000, 2024-03-06:
### Is your feature request related to a problem?

Hello, I did a [project](https://github.com/rtek1000/Datalogger_2039) with STM32F407VGT6, but I'm interested in trying the ESP32-S3 instead of the STM32.

For this I need 2 USB Host ports, so I thought about using 2 ESP32-S3, or some converter like CH376 (but it seems that this IC does not have Udisk formatting in the internal firmware).

A USB keyboard port with built-in touch, which the ESP32-S3 can [already operate](https://github.com/espressif/esp-idf/issues/12667). And another port for Udisk, but I also need the formatting function.

### Describe the solution you'd like.

If there is already an option to format Udisk, is there any specific documentation or example code?

### Describe alternatives you've considered.

Perhaps the Udisk formatting functions of the STM32 code (FatLib) could be adapted for the ESP32-S3

### Additional context.

Code I used for STM32F4:

```C++
HAL_StatusTypeDef Udisk_format(uint8_t fs_type) {
	if (Appli_state_HS != APPLICATION_READY) {
		UsrLog("Appli_state_HS Not APPLICATION_READY");
		return HAL_BUSY;
	}

	if ((fs_type != FM_FAT32) && (fs_type != FM_EXFAT)) {
		return HAL_ERROR;
	}

	//if (lcd_screen == LcdScreenUsbFormat) {
	uint16_t FF_MAX_SS = 4096;
	BYTE work[FF_MAX_SS];

	uint8_t fres = 0;

	DWORD myLba[4] = { 100, 0, 0, 0 };

	UsrLog("f_fdisk Start...");

	fres = f_fdisk(0, myLba, (void*) work);

	if (fres != FR_OK) {
		UsrLog("f_fdisk ERROR: %u", fres);
		return HAL_ERROR;
	} else {
		UsrLog("f_fdisk OK");
	}

	UsrLog("f_mkfs Start...");

	// fs_type: FM_EXFAT, FM_FAT32
	fres = f_mkfs((const TCHAR*) USBDISKPath, fs_type, 0, work, sizeof work);

	if (fres != FR_OK) {
		UsrLog("f_mkfs ERROR: %u", fres);
		return HAL_ERROR;
	} else {
		UsrLog("f_mkfs OK");
	}
	//}

	return HAL_OK;
}
```

- f_fdisk() and f_mkfs() are from ff.c file: "[FatFs - Generic FAT file system module  R0.12c](http://elm-chan.org/fsw/ff/)"

leeebo, 2024-03-07:
@rtek1000 By default, if you set `esp_vfs_fat_mount_config_t->format_if_mount_failed = true`, the driver will internally format the udisk **only when mounts failed** (happens when udisk FAT files is bronken)

If you want to force trigger the format operate, please refer [msc_format_storage](https://github.com/leeebo/esp-usb/blob/76a723acc5a403046104e3cf8d69acedea5214a8/host/class/msc/usb_host_msc/src/msc_host_vfs.c#L28)

leeebo, 2024-03-07:
For the 2 USB Host ports requirement, you may get interests of the USB Hub support. https://github.com/espressif/esp-idf/issues/12554

tore-espressif, 2024-12-16:
@rtek1000 The external HUB is already supported in latest master branch.

Explicit function for run-time formatting of connected USB disks was also added https://github.com/espressif/esp-usb/pull/101

Feel free to reopen if you have any more questions!

closed 2024-12-16 by tore-espressif