espressif/esp-idf #13327: ESP32-S3-DevKitC-1  host  msc

vwwwvwww, 2024-03-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I (202) cpu_start: Project name:     usb-msc
I (207) cpu_start: App version:      v5.2-dirty
I (212) cpu_start: Compile time:     Mar  7 2024 11:00:23
I (218) cpu_start: ELF file SHA256:  7201ffe3d...
I (223) cpu_start: ESP-IDF:          v5.2-dirty
I (228) cpu_start: Min chip rev:     v0.0
I (233) cpu_start: Max chip rev:     v0.99 
I (238) cpu_start: Chip rev:         v0.1
I (243) heap_init: Initializing. RAM available for dynamic allocation:
I (250) heap_init: At 3FC95B70 len 00053BA0 (334 KiB): RAM
I (256) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (262) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (268) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (276) spi_flash: detected chip: generic
I (279) spi_flash: flash io: dio
W (283) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (296) sleep: Configure to isolate all GPIO pins in sleep state
I (303) sleep: Enable automatic switching of GPIO sleep configuration
I (311) main_task: Started on CPU0
I (321) main_task: Calling app_main()
I (321) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2 
I (331) example: Waiting for USB flash drive to be connected

你好,  想咨询一下ESP32-S3-DevKitC-1这块开发板是不是只需要插上u盘就可以做usb host msc的 u盘读写实验了?  我用otg线连接u盘,  Monitor一直在等待连接,,          还是说ESP32-S3-DevKitC-1这个开发板要另外做一些设置才能用usb host.

peter-marcisovsky, 2024-03-07:
Hello @vwwwvwww 

Can you please explain how are you connecting your flash drive to the development board?

Did you just connect your flash drive to the USB port on the dev board? If yes, that unfortunately would not work. There is a protection diode which does not deliver power to a USB device.

Please specify your connection so I can give you further instructions how to make it working. Thx.

vwwwvwww, 2024-03-07:
[image]

i connect  ESP32-S3-DevKitC-1   like this.  

should I change the development board?

peter-marcisovsky, 2024-03-07:
The discussion on the forum thread I posted says: 

> As shown in the picture above, the USB interface of esp32-s3-devkit can only input 5V by default and cannot output 5V because there is a diode to prevent reverse direction. If you want to output 5V, you can short-circuit the diode on the development board yourself.

To make your setup working, you can either do that (to short circuit the diode, as explained in the forum).

Or you can connect your flash disk using a similar cable to [this](https://www.adafruit.com/product/4449l) one. With that cable you can connect the female jumpers to the pin headers of the dev board. Thus connecting VCC, GND, D+ and D- of the flash disk directly to the pin headers.

vwwwvwww, 2024-03-07:
Thank you very much. Your response is very helpful to me

peter-marcisovsky, 2024-03-07:
I am glad I could help. 

Feel free to ask more questions, or to close the issue if your problem is solved.

closed 2024-03-13 by peter-marcisovsky