espressif/esp-idf #13328: prvSelectHighestPriorityTaskSMP in tasks.c throws a panic

norventa, 2024-03-07:
Back to prvSelectHighestPriorityTaskSMP in tasks.c. If a task is currently being executed it won't be scheduled to run. When my app panics at:
configASSERT(xTaskScheduled == pdTRUE); /* At this point, a task MUST have been scheduled */ 
Adding a call to vTaskList just before that line shows IDLE0 is being executed (X) and xCurCoreID == 0.
So my question is: what prevents this situation happening?

_Originally posted by @norventa in https://github.com/espressif/esp-idf/issues/11461#issuecomment-1980746598_

norventa, 2024-03-07:
It shows the fault on 5.2, I haven't tried it on any other version. Increase the outer loop iterations of main to 1000.

0xjakob, 2024-03-07:
When I increase the iterations, I get a different error while deleting the tasks. It looks quite like a memory corruption. Do you have a backtrace of the application (the output of `idf.py` after the crash)? With the backtrace we might get a hint if we're talking about the same issue here or if it's something else.

BTW., I remove the watchdog code from the provided example, it didn't change anything.

0xjakob, 2024-03-08:
@norventa Please check the return value of `xTaskCreatePinnedToCore()`. In the example, it's not checked and hence the application continues until it later accesses uninitialized memory. Seems like `xTaskCreatePinnedToCore()` simply runs out of memory. The reason why it runs out of memory is that the memory of the old tasks is deleted by the idle task which is not run fast enough in this case. Adding a vTaskDelay(1) each time after all the tasks are created (or deleted) fixes the crash. Do you check the return values of all calls to `xTaskCreatePinnedToCore()` in your application? Will it still crash even though `pdPASS` is returned?

norventa, 2024-03-14:
Sorry for the long delay in responding.
Your point about my example is quite valid. Checking the return values pointed to a problem with the stack size. When I increased that the example ran without a problem.
However my application still exhibits the fault: the function prvSelectHighestPriorityTaskSMP in tasks.c throws a panic at the line:
`configASSERT(xTaskScheduled == pdTRUE); /* At this point, a task MUST have been scheduled */`
Using jtag debugging I have discovered that when the panic occurs there is always a task stopped at one of the printf variants. I suggest to you that somewhere in the call chain, at or below one of these functions, a block occurs but does not correctly update pxReadyTasksLists of tasks.c.
Why this might be the case and printf... blocks, I can only imagine it uses a shared buffer and some locking mechanism?
My app has to run for some time before this panic happens and if I add a task that loops repeatedly calling taskYIELD() the panic does not happen.

[image]

0xjakob, 2024-03-15:
@norventa We can take a look, but we also need to know which ESP-IDF version are you using, i.e., which tag or commit ID. When I checkout `v5.2`, the line numbers do not match the ones from your screenshot. Also, could you send the skdconfig file generated in the build forlder? I'm asking because for timing-sensitive issues these details sometimes make the difference between reproducing a crash and running fine.

0xjakob, 2024-03-15:
One more question: Are you using idle task hooks registered by `esp_register_freertos_idle_hook_for_cpu()` or the FreeRTOS idle task hook `vApplicationIdleHook()`?

norventa, 2024-03-15:
@0xjakob The version is 5.2 however I doctored the function prvSelectHighestPriorityTaskSMP in tasks.c as below to capture the panic in my debugger. My code formatter may have changed line position too but this is the only change to that file. I have also included sdkconfig.
My app doesn't use Arduino code but vanilla idf. Does that mean esp_register_freertos_idle_hook_for_cpu or vApplicationIdleHook is used? Otherwise how do I find out which it is?

`#if (configNUMBER_OF_CORES > 1)
// DPB:
#define DPB_DEBUG
#if defined(DPB_DEBUG)
#include <esp_task_wdt.h>
#endif

static void
prvSelectHighestPriorityTaskSMP(void)
{
    /* This function is called from a critical section. So some optimizations are made */
    BaseType_t uxCurPriority;
    BaseType_t xTaskScheduled     = pdFALSE;
    BaseType_t xNewTopPrioritySet = pdFALSE;
    BaseType_t xCurCoreID         = portGET_CORE_ID();

    /* Search for tasks, starting form the highest ready priority. If nothing is
         * found, we eventually default to the IDLE tasks at priority 0 */

    for (uxCurPriority = uxTopReadyPriority; uxCurPriority >= 0 && xTaskScheduled == pdFALSE; uxCurPriority--)
    {
        /* Check if current priority has one or more ready tasks. Skip if none */
        if (listLIST_IS_EMPTY(&(pxReadyTasksLists[uxCurPriority])))
        {
            continue;
        }

        /* Save a copy of highest priority that has a ready state task */
        if (xNewTopPrioritySet == pdFALSE)
        {
            xNewTopPrioritySet = pdTRUE;
            uxTopReadyPriority = uxCurPriority;
        }

        /* We now search this priority's ready task list for a runnable task.
             * We always start searching from the head of the list, so we reset
             * pxIndex to point to the tail so that we start walking the list from
             * the first item */
        pxReadyTasksLists[uxCurPriority].pxIndex = (ListItem_t *)&(pxReadyTasksLists[uxCurPriority].xListEnd);

        /* Get the first item on the list */
        TCB_t *pxTCBCur;
        TCB_t *pxTCBFirst;
        listGET_OWNER_OF_NEXT_ENTRY(pxTCBCur, &(pxReadyTasksLists[uxCurPriority]));
        pxTCBFirst = pxTCBCur;

        do
        {
            /* Check if the current task is currently being executed. However, if
                 * it's being executed by the current core, we can still schedule it.
                 * Todo: Each task can store a xTaskRunState, instead of needing to
                 *       check each core */
            UBaseType_t x;

            for (x = 0; x < configNUMBER_OF_CORES; x++)
            {
                if (x == xCurCoreID)
                {
                    continue;
                }
                else if (pxCurrentTCBs[x] == pxTCBCur)
                {
                    /* Current task is already being executed. Get the next task */
                    goto get_next_task;
                }
            }

[... 1999 lines cut ...]
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_TIME_SYSCALL_USE_RTC_HRT=y
CONFIG_ESP32_TIME_SYSCALL_USE_RTC_FRC1=y
# CONFIG_ESP32_TIME_SYSCALL_USE_RTC is not set
# CONFIG_ESP32_TIME_SYSCALL_USE_HRT is not set
# CONFIG_ESP32_TIME_SYSCALL_USE_FRC1 is not set
# CONFIG_ESP32_TIME_SYSCALL_USE_NONE is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
# CONFIG_ESP32_ULP_COPROC_ENABLED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
# End of deprecated options
`

norventa, 2024-03-15:
@0xjakob My app just threw a panic and `esp_register_freertos_idle_hook_for_cpu` was in the call chain.

0xjakob, 2024-03-18:
@norventa If your application is registering idle hook (callback) functions `esp_register_freertos_idle_hook_for_cpu()`, and one of these idle hook functions does a blocking call, then this would explain the assert you're describing above:
```
configASSERT(xTaskScheduled == pdTRUE); /* At this point, a task MUST have been scheduled */
```

Could you check all the functions which are registered as idle hook functions via `esp_register_freertos_idle_hook_for_cpu()`?

norventa, 2024-03-18:
Nope, not called anywhere withing the project apart from within FreeRTOS's startup. Is there a way to iterate idle hooks programmatically?

0xjakob, 2024-03-18:
The idle hooks are called either every FreeRTOS tick (which usually occurs with 100Hz) or repeatedly as fast as possible. Hence, I would like to know how the functions which are registered using `esp_register_freertos_idle_hook_for_cpu()` look like, in particular, if these functions call a blocking function as this will trigger exactly the assert you pointed out above.

norventa, 2024-03-24:
@0xjakob Hypothetically, if I had many tasks and the task slot is 1ms per task, if the idle task took longer than 1ms to perform its duty and the task scheduler switched it out, would it possible for the scheduler to have no tasks ready to run?

0xjakob, 2024-05-02:
@norventa No tasks ready to run means that none of the tasks is in "ready" state (compare [FreeRTOS tasks states](https://www.freertos.org/RTOS-task-states.html)). This is why the IDLE task shall never block. If there is no other task than the IDLE task ready to run, the IDLE task should be scheduled, and re-scheduled if it has been running before. In other words, it is not (so much) a problem if the IDLE task is running a long time, but it definitely is a problem if it blocks, because that's where the scheduler may not find a ready task. 

Please note, however, that offloading too much work to the IDLE task is not a good idea in general, even if it does not encounter blocking states. The IDLE task performs some other work than user callbacks, too. E.g., it is cleaning up memory after a task has been deleted.

0xjakob, 2024-06-26:
@norventa Is there any update regarding this issue? If not, we'd like to close it for now (could be re-opened later, of course).

norventa, 2024-06-27:
@0xjakob I have tried to find the source of my problem but couldn't nail it down. I refactored some of my code and it went away. I believe it is possible for a 'well behaved' app to throw this exception but it would be rare. In anycase I am happy for you to close this issue. Thank you for your help.

0xjakob, 2024-07-02:
@norventa We're closing this issue then. If you find any example that still triggers the assert, please let use know!

closed 2024-07-02 by 0xjakob