espressif/esp-idf #13330: MQTT publish hangs some times

br101, 2024-03-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1

### Espressif SoC revision.

ESP32-C3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C3-DevKitC-02, Custom hardware

### Power Supply used.

USB

### What is the expected behavior?

The MQTT component esp_mqtt_client_publish() function should either publish or fail after a timeout.

### What is the actual behavior?

esp_mqtt_client_publish() hangs indefinitely (does not return), blocking the calling thread.

### Steps to reproduce.

Unfortunately I can not reproduce it easily. It happens in a production environment, which may be have poor Wifi at times, and quite a lot of data is transmitted by esp_mqtt_client_publish(). Publish is called from several threads simultaneously, and it could be that MQTT has to reconnect due to bad wifi signal or connection problems.

### Debug Logs.

```plain
Sorry, not available...
```

### More Information.

The MQTT connection is using SSL and there may be not a lot of free continuous memory available in these cases.

I'm not sure which way to go to mitigate the problem. Would it make sense to call publish only from one thread?

SolidStateLEDLighting, 2024-03-07:
My instinct on this would be marshal all the messages over to one task so you have direct control over the loading of the local client.   You never know when something behind the scenes is not as thread (task) safe and you thought it might be.

The second thing would be to make sure that your client is passing through it's process_loop() on the correct periodic basis to make sure the connection remains open and active.   Typically, if you may be expecting an incoming message,  you would pass through a process_loop() often enough to receive that incoming message without delay.    My experience is with the AWS MQTT system so your approach may differ.

br101, 2024-03-08:
Thank you for your feedback and your suggestions! In the end it was my own problem. I had one task that could get suspended while publishing... sorry

closed 2024-03-08 by br101