espressif/esp-idf #13331: ESP32-C5 Status

JonRobo, 2024-03-07:
### Is your feature request related to a problem?

Currently using the ESP32 in a product, but many clients are switching to 5GHz networks and we will need the ESP32-C5 to stay relevant. I would like to check in to see how the ESP32-C5's development is going. Thank you

Alvin1Zhang, 2024-03-08:
Thanks for reporting, the support for ESP32-C5 is already in progress in IDF, and the tracking issue will be created soon, similar to the [ESP32-P4](https://github.com/espressif/esp-idf/issues/12996). Will update again once the tracking issue is created. Thanks.

closed 2024-03-08 by Alvin1Zhang

Espressif-yangxi, 2024-03-12:
@JonRobo Hi JonRobo, If you need more information about the ESP32-C5, feel free to [reach out to Espressif's customer support team](https://www.espressif.com/en/contact-us/sales-questions). thanks.

MichelangeloGuaitolini, 2024-04-17:
Hi @Espressif-yangxi, I am interested in information about the ESP32-C5 as well. 
I emailed the customer support team throught the very same link the you suggested here more than one month ago but I didn't get any reply. Would you be able to give me any update? 
Thank you in advance.

Espressif-yangxi, 2024-04-18:
> Hi @Espressif-yangxi, I am interested in information about the ESP32-C5 as well.
> I emailed the customer support team throught the very same link the you suggested here more than one month ago but I didn't get any reply. Would you be able to give me any update?
> Thank you in advance.
Hi @Mike9277, could you kindly write to sales@espressif.com & yangxi@espressif.com again, we will have a double check about your email, and share you ESP32-C5 documents or more information you are interested in.
Thank you.