espressif/esp-idf #13332: TWAI has no defaults for 26 Mhz XTAL (like the popular LoRa32), probably showstopper for many

nicklasb, 2024-03-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi Espressif! 

If I twai (hehe) to get TWAI working on the LoRa32 it fails to build since there is no section for XTAL in xtal_types.h for this 26 Mhz XTALs.
 
As these values are at least beyond me to (properly) figure out, it is probably a show stopper for most with similar boards, thus it would be appreciated if someone calculated some defaults for 26Mhz XTALs and posted here, or better, added to xtal_types.h.

Thanks!

wanckl, 2024-03-08:
> twai (hehe)

hehe

wanckl, 2024-03-08:
@nicklasb  Thank you for report! I'll update later :+1:

nicklasb, 2024-03-21:
> @nicklasb Thank you for report! I'll update later 👍

I guess forgot to ask what "later" meant. :-)

(note that awaiting inclusion in the actual source I will gladly try out any settings you may propose here)

wanckl, 2024-03-22:
@nicklasb Sorry for late reply, But due to integer divider limitation, just littile of macros are available for 26M xtal, and don't have a  test because lack this board, you can have a try first.
```c
#if CONFIG_XTAL_FREQ == 26   // TWAI_CLK_SRC_XTAL = 26M
#define TWAI_TIMING_CONFIG_25KBITS()    {.clk_src = TWAI_CLK_SRC_DEFAULT, .quanta_resolution_hz = 400000, .brp = 0, .tseg_1 = 11, .tseg_2 = 4, .sjw = 3, .triple_sampling = false}
#define TWAI_TIMING_CONFIG_50KBITS()    {.clk_src = TWAI_CLK_SRC_DEFAULT, .quanta_resolution_hz = 1000000, .brp = 0, .tseg_1 = 15, .tseg_2 = 4, .sjw = 3, .triple_sampling = false}
#define TWAI_TIMING_CONFIG_100KBITS()   {.clk_src = TWAI_CLK_SRC_DEFAULT, .quanta_resolution_hz = 1000000, .brp = 0, .tseg_1 = 6, .tseg_2 = 3, .sjw = 3, .triple_sampling = false}
#define TWAI_TIMING_CONFIG_125KBITS()   {.clk_src = TWAI_CLK_SRC_DEFAULT, .quanta_resolution_hz = 1000000, .brp = 0, .tseg_1 = 5, .tseg_2 = 2, .sjw = 1, .triple_sampling = false}
#define TWAI_TIMING_CONFIG_250KBITS()   {.clk_src = TWAI_CLK_SRC_DEFAULT, .quanta_resolution_hz = 1000000, .brp = 0, .tseg_1 = 2, .tseg_2 = 1, .sjw = 1, .triple_sampling = false}
#define TWAI_TIMING_CONFIG_1MBITS()     {.clk_src = TWAI_CLK_SRC_DEFAULT, .quanta_resolution_hz = 13000000, .brp = 0, .tseg_1 = 8, .tseg_2 = 4, .sjw = 3, .triple_sampling = false}
#endif
```

nicklasb, 2024-03-22:
Great! Will try right away!

nicklasb, 2024-03-22:
On the TWAI_TIMING_CONFIG_1MBITS i am getting ESP_ERR_INVALID_ARG when calling twai_driver_install, so that seems to be a bit off. 
I am getting further with TWAI_TIMING_CONFIG_250KBITS, at least then it starts and seems to send something that my logic analyzer (set to 250 KBits/s) picks up and can parse, but the other ESP32 that is a 40 Mhz XTAL (and can talk to a T-Beam) doesn't seem to react or interact much with it.

wanckl, 2024-03-25:
@nicklasb  I realized my mistake, On esp32 it depend on **NOT** `XTAL` freq but `TWAI_CLK_SRC_DEFAULT`, they are different on ESP32,,So for the correction, just replace block in `twai_types.h` as follow:
```c
#if APB_CLK_FREQ == (32*1000000)   // TWAI_CLK_SRC_DEFAULT = 32M
#define TWAI_TIMING_CONFIG_25KBITS()    {.clk_src = TWAI_CLK_SRC_DEFAULT, .quanta_resolution_hz = 400000, .brp = 0, .tseg_1 = 11, .tseg_2 = 4, .sjw = 3, .triple_sampling = false}
#define TWAI_TIMING_CONFIG_50KBITS()    {.clk_src = TWAI_CLK_SRC_DEFAULT, .quanta_resolution_hz = 1000000, .brp = 0, .tseg_1 = 15, .tseg_2 = 4, .sjw = 3, .triple_sampling = false}
#define TWAI_TIMING_CONFIG_100KBITS()   {.clk_src = TWAI_CLK_SRC_DEFAULT, .quanta_resolution_hz = 2000000, .brp = 0, .tseg_1 = 15, .tseg_2 = 4, .sjw = 3, .triple_sampling = false}
#define TWAI_TIMING_CONFIG_125KBITS()   {.clk_src = TWAI_CLK_SRC_DEFAULT, .quanta_resolution_hz = 4000000, .brp = 0, .tseg_1 = 23, .tseg_2 = 8, .sjw = 3, .triple_sampling = false}
#define TWAI_TIMING_CONFIG_250KBITS()   {.clk_src = TWAI_CLK_SRC_DEFAULT, .quanta_resolution_hz = 4000000, .brp = 0, .tseg_1 = 11, .tseg_2 = 4, .sjw = 3, .triple_sampling = false}
#define TWAI_TIMING_CONFIG_500KBITS()   {.clk_src = TWAI_CLK_SRC_DEFAULT, .quanta_resolution_hz = 8000000, .brp = 0, .tseg_1 = 11, .tseg_2 = 4, .sjw = 3, .triple_sampling = false}
#define TWAI_TIMING_CONFIG_800KBITS()   {.clk_src = TWAI_CLK_SRC_DEFAULT, .quanta_resolution_hz = 16000000, .brp = 0, .tseg_1 = 15, .tseg_2 = 4, .sjw = 3, .triple_sampling = false}
#define TWAI_TIMING_CONFIG_1MBITS()     {.clk_src = TWAI_CLK_SRC_DEFAULT, .quanta_resolution_hz = 16000000, .brp = 0, .tseg_1 = 11, .tseg_2 = 4, .sjw = 3, .triple_sampling = false}

#elif APB_CLK_FREQ == (80*1000000)   // TWAI_CLK_SRC_DEFAULT = 80M
#define TWAI_TIMING_CONFIG_25KBITS()    {.clk_src = TWAI_CLK_SRC_DEFAULT, .quanta_resolution_hz = 625000, .brp = 0, .tseg_1 = 16, .tseg_2 = 8, .sjw = 3, .triple_sampling = false}
#define TWAI_TIMING_CONFIG_50KBITS()    {.clk_src = TWAI_CLK_SRC_DEFAULT, .quanta_resolution_hz = 1000000, .brp = 0, .tseg_1 = 15, .tseg_2 = 4, .sjw = 3, .triple_sampling = false}
#define TWAI_TIMING_CONFIG_100KBITS()   {.clk_src = TWAI_CLK_SRC_DEFAULT, .quanta_resolution_hz = 2000000, .brp = 0, .tseg_1 = 15, .tseg_2 = 4, .sjw = 3, .triple_sampling = false}
#define TWAI_TIMING_CONFIG_125KBITS()   {.clk_src = TWAI_CLK_SRC_DEFAULT, .quanta_resolution_hz = 2500000, .brp = 0, .tseg_1 = 15, .tseg_2 = 4, .sjw = 3, .triple_sampling = false}
#define TWAI_TIMING_CONFIG_250KBITS()   {.clk_src = TWAI_CLK_SRC_DEFAULT, .quanta_resolution_hz = 5000000, .brp = 0, .tseg_1 = 15, .tseg_2 = 4, .sjw = 3, .triple_sampling = false}
#define TWAI_TIMING_CONFIG_500KBITS()   {.clk_src = TWAI_CLK_SRC_DEFAULT, .quanta_resolution_hz = 10000000, .brp = 0, .tseg_1 = 15, .tseg_2 = 4, .sjw = 3, .triple_sampling = false}
#define TWAI_TIMING_CONFIG_800KBITS()   {.clk_src = TWAI_CLK_SRC_DEFAULT, .quanta_resolution_hz = 20000000, .brp = 0, .tseg_1 = 16, .tseg_2 = 8, .sjw = 3, .triple_sampling = false}
#define TWAI_TIMING_CONFIG_1MBITS()     {.clk_src = TWAI_CLK_SRC_DEFAULT, .quanta_resolution_hz = 20000000, .brp = 0, .tseg_1 = 15, .tseg_2 = 4, .sjw = 3, .triple_sampling = false}
#endif  //APB_CLK_FREQ
```

nicklasb, 2024-03-25:
Hi! 
Nice! Those settings worked for 250 KBits/s and 1 MBits at least! 
I did not try any of the other speed, if you want me too, I can walk through them at a later time, I don't have time to sit through 7 more rebuilds right now. 

I didn't add it to twai_types.h as then idf.py will refuse to build it (because of https://github.com/espressif/idf-component-manager/issues/40#issuecomment-1897929443 , yeah, that is me being annoyed there. :-) ) and I'd rather do it so that I know the dependencies works. 
But I just added it above the code, and that worked. I would assume it would work wherever you put it. It also worked with the DevKit v5 i used as a test peer. 

So if understand it correctly, the above code would be able to replace all the default code in twai_types and work for all boards? It would basically always be better to go by APB_CLK_FREQ?

nicklasb, 2024-03-25:
I am also noting that the low-speed configs are not available, is this intentional?

wanckl, 2024-03-26:
@nicklasb  

Yeah, this way is work for all boards and more correct than using `APB_CLK_FREQ`

For speed lower than 25K, it is available for ESP32-**ECO2** and later other chips

wanckl, 2024-03-28:
@nicklasb My bad, but a bit different, in one word, for esp32 APB is more correct then XTAL

I update the comment above, now should be full correct

wanckl, 2024-04-09:
@nicklasb Hi, If no problem, can you close this issue?

closed 2024-04-09 by nicklasb