espressif/esp-idf #13335: InstrFetchException with in IDF v5 and mbedtls v3

HamzaHajeir, 2024-03-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5 and newer

### Espressif SoC revision.

..

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Any

### Power Supply used.

USB

### What is the expected behavior?

I'm maintaining a TCP/TLS library for Arduino users, and using [LwIP ALTCP](https://github.com/espressif/esp-idf/issues/11556) virtual layer of LwIP to run both plain TCP and TLS with minimal code changes.

After developing and integrating it, this worked perfectly with no known issues, except a [corner one](https://github.com/espressif/esp-idf/issues/11738).

### What is the actual behavior?

However, upgrading to ESP-IDF v5, with the accompanying major releases (mbedtls v3) and (GCC 12.2) seems to cause an unknown exception that happens after calling `mbedtls_ssl_read()` and targets the buffer provided and managed by lwip. So when calling for `pbuf_free` for the destination buffer happens the exception.

### Steps to reproduce.

I've uprooted the minimum code from the library to make a minimal esp-idf reproducible code, and made it to connect to  an MQTT local server which enables both plain TCP and secured TLS with standard ports (1883,8883).

Now I've established a [repository](https://github.com/HamzaHajeir/esp-idf-v5-issue) that holds both programs, the first to ensure it's working in v4.4, and the other with the bug running IDF v5.1.

Both programs have a combined of WiFi station and Blink examples within the running code with similar code in both.

Both programs have the corresponding lwip library, with my fixes and patches applied in both under `components` directory, but with a port to mbedtls v3 applied to the v5.1 example, the port patch can be found [here](https://github.com/HamzaHajeir/esp-lwip/commit/06ffa197068981493743fe5321d64014f8473308).

We're interested with the TLS one.
So to reproduce:
1. Change `MQTT_CERT` into your installed certificate, in TCP.cpp.  
Alternitavely, install the certificate and key provided [here](https://gist.github.com/HamzaHajeir/b07d8b4a0dbf2f497a9c199c4961da9a) into the MQTT server program. A sample mosquitto config file is added.
2. Change the WiFi credentials in main.cpp.
3. Change the local IP of your machine running the MQTT server:
```cpp
    ip_addr_t ip = IPADDR4_INIT_BYTES(192,168,100,4);
```
4. Might need to change the  ESP-IDF environment under `./.vscode/settings.json`, as I've identified the ESP-IDF environment for both (IDF path and toolchain versions,...).

### Debug Logs.

```plain
In the program running successfully, nothing useful done other than connecting to the MQTT server successfully.

[... 70 lines cut ...]
0x420156bd: tcpip_thread_handle_msg at C:/Users/Hamza/Documents/esp-idf/Projects/WiFi/station-v5.1/components/lwip/lwip/src/api/tcpip.c:174 (discriminator 4)
0x4201574c: tcpip_thread at C:/Users/Hamza/Documents/esp-idf/Projects/WiFi/station-v5.1/components/lwip/lwip/src/api/tcpip.c:148
0x4037ffe1: vPortTaskWrapper at C:/Users/Hamza/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162
```
```

### More Information.

As the provided example uses MQTT as application, the issue is reproducible even if it's an HTTP client that runs over TLS, over the first call to `mbedtls_ssl_read()`.

To summarize links:
1. The esp-idf projects repo: https://github.com/HamzaHajeir/esp-idf-v5-issue
2. Mbedtls v3 port patch*: https://github.com/HamzaHajeir/esp-lwip/commit/06ffa197068981493743fe5321d64014f8473308
3. Local server certificates (built for testing): https://gist.github.com/HamzaHajeir/b07d8b4a0dbf2f497a9c199c4961da9a
4. An explanation of LwIP ALTCP provided by me (With a request to support it in IDF): https://github.com/espressif/esp-idf/issues/11556

\* The provided lwip component adds a check to the return value of `mbedtls_ssl_flush_output` over that. and it also adds some debugging prints which are deactivated in default by a preprocessor.

I've tried to use GCC 12.2 to build an [Arduino project](https://github.com/HamzaHajeir/H4Plugins_Env) over a precompiled sdk of (esp-idf version 4.4.5) (Patched Arduino v2.x [by adding ALTCP](https://github.com/HamzaHajeir/esp32-arduino-lib-builder/tree/lwip-tls-1)), the issue doesn't appear.

Compiling with GCC 12.2 was done by identifying the toolchain in `platformio.ini` (The config file).

My comment to this last one: It could be appear that GCC 12.2 is completely innocent and isolated, but we shall not forget that ESP-IDF v4.4.5 SDK is being built by GCC 8.4.0.

This issue have stopped me progressing for 6 months until now!

HamzaHajeir, 2024-04-05:
Hi @mahavirj, 

Per the server, it doesn't necessarily be an MQTT server, it can be a normal TCP echo server secured with TLS, which should be sufficient for replicating the issue.

Is there anything else I can provide to accelerate investigating the issue?

mahavirj, 2024-04-08:
@HamzaHajeir 

Few things to note from the crash log shared here:

1. CPU is trying to fetch instruction from invalid address: `EXCVADDR: 0x00000220` and hence the exception
2. Probably this address is coming from this address (line above the crash dump)
```
[0x3FCB9004] 0x00000000     0:  20 02 00 00                                        ...
```
3. Is `_raw_recv` the call responsible to execute some sort of callback? 
4. Maybe putting up a watchpoint on `0x3FCB9004` can help if the crash log is reproducible

In general, I think this use-case needs to be narrowed down further. I do not see this to be directly IDF related because you are using different networking stack, netif integration layer etc. Please see if you can refer to our debug docs and see from where the crash originates, that will help.

Thanks.

HamzaHajeir, 2024-04-08:
Hi @mahavirj 

Thanks for the hint, the printed binary is a dumphex of the copied buffer and maintained by the application, I can confirm by the decoded stack trace that the exception is not emitted from that copied buffer, but by calling `pbuf_free(p)` (line 84), one can checkout the concerning code:

https://github.com/HamzaHajeir/esp-idf-v5-issue/blob/8ac83b816e566cbfe541e7c41f1f055bc0423f8c/station-v5.1/main/TCP.cpp#L77-L85

It's interesting that both copied buffer and the buffer hold that same value (It should hold the same value, as it's a copy). What's interesting is the packet buffer `pbuf *p` becomes dangerous to free just after calling `mbedtls_ssl_read()`, one can try adding `pbuf_free(buf)` after ([components\lwip\lwip\src\apps\altcp_tls\altcp_tls_mbedtls.c:489](https://github.com/HamzaHajeir/esp-idf-v5-issue/blob/8ac83b816e566cbfe541e7c41f1f055bc0423f8c/station-v5.1/components/lwip/lwip/src/apps/altcp_tls/altcp_tls_mbedtls.c#L489)).

> 3. Is `_raw_recv` the call responsible to execute some sort of callback?

It should, but in this minimal code, it does not.

> 4. Maybe putting up a watchpoint on `0x3FCB9004` can help if the crash log is reproducible

Per my earlier explanation, the issue is not related to this location, as it's a copy from another buffer which becomes 'infected'.

> In general, I think this use-case needs to be narrowed down further.

I've [tried my best](https://github.com/espressif/esp-idf/issues/13069) to narrow it down further by building IDF v5.x based project with mbedtls 2.x, but with no luck at all.

In another way, we can determine whether it's the compiler or not by building each of the examples with the other one's compiler: station-v5.1 with GCC 8.4.0 and station-v4.4 with GCC 12.2.0. I tried, but had no success to change compiler in building esp-idf projects, as there's a check in IDF build environment that prevents so.

Your thoughts?

HamzaHajeir, 2024-05-10:
I've found a change made in ESP-IDF v5.0 that relates to [lwip pbufs management](https://github.com/espressif/esp-idf/commit/dbc91f73e64548a8a81a88ba2998380fc9328226).

What led me to find that is printing `custom_free_function` of the pbuf, which it fails on its calling after being corrupted:

The logic is:
- `pbuf_free` is being called.
- Deallocates it by the `custom_free_function`.
- There's a chanined buffer which had been concatenated by mbedtls [callback](https://github.com/HamzaHajeir/esp-lwip/blob/06ffa197068981493743fe5321d64014f8473308/src/apps/altcp_tls/altcp_tls_mbedtls.c#L493-L542) in lwip altcp_tls_mbedtls.c
- Deallocating it by the `custom_free_function` which is corrupted.
- 
```
pbuf_free(0x3fcb341c)
pbuf_free: deallocating 0x3fcb341c
pbuf_free: custom_free_function 0x4207595c
0x4207595c: esp_pbuf_free at C:/Users/Hamza/esp/esp-idf/components/esp_netif/lwip/netif/esp_pbuf_ref.c:34
 
pbuf_free(0x3fcbc3a8)
pbuf_free: deallocating 0x3fcbc3a8
pbuf_free: custom_free_function 0x5f207525
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
 
Core  1 register dump:
PC      : 0x5f207525  PS      : 0x00060930  A0      : 0x8202693e  A1      : 0x3fca8980
A2      : 0x3fcbc3a8  A3      : 0x5f207525  A4      : 0x00000000  A5      : 0x00000000
A6      : 0x00000000  A7      : 0x3fcbc3a8  A8      : 0x820169c7  A9      : 0x3fca8930  
A10     : 0x3fcbc3a8  A11     : 0x3fca8ca8  A12     : 0x0000001a  A13     : 0x000005ec
A14     : 0x00000182  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xffffffff
```

After calling `mbedtls_ssl_read()`, any call to `pbuf_free()` of a newly created buffer result in such behaviour:
```
pbuf_free(0x3fcbb260)
pbuf_free: deallocating 0x3fcbb260
pbuf_free: custom_free_function 0x4207598c
0x4207598c: esp_pbuf_free at C:/Users/Hamza/esp/esp-idf/components/esp_netif/lwip/netif/esp_pbuf_ref.c:34
 
Allocating another buffer, then freeing it immediately
pbuf_alloc(length=1516)
pbuf_alloc(length=1516) == 0x3fcbbfb4
pbuf_free(0x3fcbbfb4)
pbuf_free: deallocating 0x3fcbbfb4
pbuf_free(0x3fcba7cc)
pbuf_free: deallocating 0x3fcba7cc
pbuf_free: custom_free_function 0x3fcba82c
```

To which extent that contribute to the issue?
@mahavirj 
CC: @david-cermak

david-cermak, 2024-05-10:
This looks like an bug in the custom pbuf/free function, as we've probably never expected (nor experienced?) the chained pbufs to be free'd. I'll take a look and fix it.

HamzaHajeir, 2024-05-11:
> This looks like an bug in the custom pbuf/free function, as we've probably never expected (nor experienced?) the chained pbufs to be free'd. I'll take a look and fix it.

May God forgive you, David :-)

HamzaHajeir, 2024-05-27:
Hi @david-cermak, there's an important finding I've just have, there's an issue somewhere in allocating the actual buffer pointer, where in my case, it's 16-bytes distance of the `struct pbuf*` itself.

This pointer is the same with `pbuf_custom->custom_free_function`  (Just after the `pbuf struct*` as it should be [if it's custom](https://github.com/HamzaHajeir/esp-lwip/blob/90009cc2b0b487ab12688778f5fffc27e9b2b8c3/src/include/lwip/pbuf.h#L245-L250)), and when the application tries to copy data into `pbuf->payload`, it overwrites the `custom_free_function` pointer!

I've tried to find where it gets marked with `PBUF_FLAG_IS_CUSTOM` with no success!

- Here's a memory instance of the program running ESP-IDF v5.1:

> `struct pbuf* p` --> 0x3fcb6c24
> `p->payload` --> 0x3fcb6c34
> `p->flags` --> 3 (p->flags&PBUF_FLAG_IS_CUSTOM==PBUF_FLAG_IS_CUSTOM)

- While in a program running on IDF v4.4, because of the [added preprocessor](https://github.com/HamzaHajeir/esp-lwip/blob/8290c3b8f2adaf82aa45ec992b87f16205f2689b/src/include/lwip/pbuf.h#L225-L228), the payload is 24-bytes distant, and is NOT custom:

> `struct pbuf* p` --> 0x3fcabdec
> `p->payload` --> 0x3fcabe04
> `p->flags` --> 1 (p->flags&PBUF_FLAG_IS_CUSTOM==0)

However, in both cases, allocating `struct pbuf` is done by the same call: 
```c
struct pbuf *buf = pbuf_alloc(PBUF_RAW, PBUF_POOL_BUFSIZE, PBUF_POOL);
```

We can focus on only one question: Why is it created with the PBUF_FLAG_IS_CUSTOM flag in ESP-IDF v5.0 upwards?

Best,
Hamza Hajeir

HamzaHajeir, 2024-06-02:
It is important to note that it's not about chained pbufs. It's all about wrongly flagging the created pbuf with `PBUF_FLAG_IS_CUSTOM`, which is the root cause of the issue.

HamzaHajeir, 2024-06-08:
Hi there,

Here's the last update which I'll close the issue with, the issue was related to the custom pbuf management by esp-idf, but not their fault.

One of the patched lwip commits was aimed to [copy TCP PUSH flags to the receiving pbufs](https://github.com/HamzaHajeir/esp-lwip/blob/017168a333164a38b4b5c0c85c05172bb9c1087b/src/apps/altcp_tls/altcp_tls_mbedtls.c#L407), which was working okay with no customization. But after that being the case for esp-idf, the `PBUF_FLAG_IS_CUSTOM` was also copied. 

This was the root cause of the weird issue which delayed me 8+ months of progress, Gods will :).

closed 2024-06-08 by HamzaHajeir