espressif/esp-idf #13336: [ESP32] ULP periodic wakeup causes hang after some time

Tristaan, 2024-03-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2

### Espressif SoC revision.

ESP32-D0WD

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-DevKitC

### Power Supply used.

USB

### What is the expected behavior?

Wake from deep sleep indefinitely.

1. Connect to wifi
2. Print data to uart
3. boot ulp and go into deep sleep
4. blink 10 times from ulp
5. wake from ulp and halt ulp
6. Repeat from 1.

And do this forever

### What is the actual behavior?

The ESP32 hangs somewhere between bootloader and ULP Wake. The blinking LED should be ON after the WAKE instruction but it shuts down and the chip does not boot.

### Steps to reproduce.

1. `git clone https://github.com/Tristaan/esp32_ulp_test`
2. Connect a led to IO2
3. Configure wifi connect in `main.c`
4. Build & Flash with 5.2/master (8.3.2024)
5. Observe the LED turning off & the chip freezing after about 3-10 boots from ULP

### Debug Logs.

```plain
--
```

### More Information.

Can't seem to get debugging to work in VSCode IDE with ARM-USB-TINY. some weird error messages and google is of no help. Maybe this is known, but not so important now.

The chip is totally frozen and won't do anything until a hard reset is triggered. Shouldn't a specific WDT run?

Can it be that RTC MEM gets used up and this causes a freeze? this is just an assumption.

Alvin1Zhang, 2024-03-08:
Thanks for reporting.

sudeep-mohanty, 2024-03-14:
Hi @Tristaan, Thanks for reporting the issue to us. I can confirm that when I run your application from https://github.com/Tristaan/esp32_ulp_test, I can see that the main CPU does get woken up by the ULP a few times and then it doesn't. We will take a look into it.