espressif/esp-idf #13337: Incorrect FLASH SIZE in binary image header

ammaree, 2024-03-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3-dev-2457-g8e863fa9a8

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.1)

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

MakerFabs S3 Parallel TFT with Touch 16 (3.5")

### Power Supply used.

USB

### What is the expected behavior?

Correct FLASH Size of 16MB to be in binary image header

### What is the actual behavior?

Error message is:
`spi_flash Detected size(16384k) larger than the size in the binary image header(1024k). Using the size in the binary image header.`

Correct flash size (16MB) and PSRAM size (2MB) is specified in sdkconfig.h

### Steps to reproduce.

[sdkconfig.json](https://github.com/espressif/esp-idf/files/14539145/sdkconfig.json)

### Debug Logs.

```plain
SHA-256 comparison failed:
Calculated: 1bfd3875881504ab06b415fd4d9b6662096ac47528ce77b5e2b7079eb7fb6f1a
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403c89a0
I (45) boot: ESP-IDF v5.3-dev-2457-g8e863fa9a8-dirty 2nd stage bootloader
I (46) boot: compile time Mar  8 2024 16:02:01
I (46) boot: Multicore bootloader
D (48) bootloader_flash: non-XMC chip detected by SFDP Read (C8), skip.
D (54) bootloader_flash: mmu set block paddr=0x00000000 (was 0xffffffff)
I (61) boot: chip revision: v0.1
D (64) qio_mode: Probing for QIO mode enable...
D (68) qio_mode: Raw SPI flash chip id 0xc84018
D (72) qio_mode: Manufacturer ID 0xc8 chip ID 0x4018
I (77) qio_mode: Enabling default flash chip QIO
D (81) qio_mode: Initial flash chip status 0x2
D (85) qio_mode: QIO mode already enabled in flash
D (90) qio_mode: Enabling QIO mode...
[... 256 lines cut ...]
1.628: #0 main esp_littlefs Searching for existing filesystem for partition "lfs"
1.656: #0 main esp_littlefs Existing filesystem "lfs" not found
1.677: #0 main esp_littlefs failed to read addr 00000000, size 00000080, err 258
1.698: #0 main esp_littlefs mount failed, LFS_ERR_IO (-5). formatting...
1.719: #0 main esp_littlefs Formatting partition at "0x00310000"
1.740: #0 main esp_littlefs Searching for existing filesystem for partition "0x00310000"
1.761: #0 main esp_littlefs Found existing filesystem "0x00310000" at index 0
1.782: #0 main esp_littlefs Formatting filesystem
1.803: #0 main esp_littlefs failed to read addr 00001000, size 00000080, err 258
1.824: #0 main esp_littlefs Failed to format filesystem
1.845: #0 main esp_littlefs format failed
1.866: #0 main esp_littlefs Failed to initialize LittleFS
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x4200e17b
0x4200e17b: app_main at /Users/andremaree/DevSpace/z-projects/irmacs/main/main.cpp:91 (discriminator 1)

file: "./main/main.cpp" line 91
func: void app_main()
expression: halSTORAGE_InitFS()

abort() was called at PC 0x4037c987 on core 0
```

### More Information.

Have also tried on an EspressIf BOX with 16MB FLASh and 8MB PSRAM, exact same problem.

mythbuster5, 2024-03-13:
The kconfig link is invalid. BTW can you read the image header via `esptool.py read_flash 0x0000 0x200 xxx.bin >> xxx.txt`?

AxelLin, 2024-05-19:
> The kconfig link is invalid.

Looks valid to me: https://github.com/espressif/esp-idf/files/14539145/sdkconfig.json

ammaree, 2024-07-15:
I think this issue has "solved itself" somehow, not sure but keeping an eye open.
Will update again during next 2 weeks...

mythbuster5, 2025-01-20:
Thanks, if there is nothing update, I will close this issue?

closed 2025-04-07 by Alvin1Zhang