espressif/esp-idf #13341: [Question] is Secure Download Mode worth the hassle? given that Flash Encryption can still be hacked?

chipweinberger, 2024-03-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm considering leaving Secure Download Mode off. It adds a lot of hassle. 

Flash Encryption can never be fully secure anyway, that is just reality: https://www.youtube.com/watch?v=wfZHQocTsZo 

I'm trying to answer the question: how much easier will it be to hack my device if I leave Secure Download Mode off, in comparison to doing the above hack from Black Hat?

chipweinberger, 2024-03-09:
Same with these fuses.

> DIS_DOWNLOAD_MANUAL_ENCRYPT which disables flash encryption operation when running in UART bootloader boot mode.
> 
> DIS_DOWNLOAD_ICACHE and DIS_DOWNLOAD_DCACHE which disables the entire MMU flash cache when running in UART bootloader mode.
> 
> HARD_DIS_JTAG and DIS_USB_JTAG which disables JTAG.
> 
> DIS_LEGACY_SPI_BOOT which disables Legacy SPI boot mode

If I skip these steps, will my device still require some effort to crack?

chipweinberger, 2024-03-09:
From [this thread](https://esp32.com/viewtopic.php?t=23235), I found a good answer for what happens if you *don't* enable SECURE_DOWNLOAD_MODE.

> Additionally, I read in the docs and on other forum posts: when enabling flash encryption and secure boot v2, and not restricting the UART download mode ("full UART"/SECURE_INSECURE_ALLOW_DL_MODE) should not expose any known vulnerabilities? Restricting it will only (and I quote) "reduce the attack surface of an attacker" (see https://www.esp32.com/viewtopic.php?p=73686#p73686). So in other words, enabling flash encryption and secure boot V2, while allowing full UART, should in theory be safe? (Agreed, disabling it would probably be better, but that's no option for us). However, if we manage (with your help0 to get it in this secure mode, we'll be glad to do so.

> Well, it's secure in that there shouldn't be any *direct* ways to grab the unencrypted flash using the insecure download mode... however, do note that download mode can still access the entirety of RAM which depending on how your application works may lead to attacks. An attacker may for example be able to reset the ESP32 at just the right time that server credentials are being handled (and as such are in RAM), then dump the RAM and lift server credentials from there, perhaps using that to nab a firmware update straight from the server.

Yes, it makes it much easier to exploit. But still requires a motivated attacker, perhaps similar motivation to the Black Hat exploits.

chipweinberger, 2024-03-09:
So I'll answer my own question. The hack from Jan 2024 doesn't seem that hard to pull off.

Tools will probably come out that make it even easier. plug-n-play.

So in this sense, I don't think Secure Download Mode or any of the other fuses are worth burning.

I might still enable flash encryption just to make it a little harder I suppose.

mahavirj, 2024-03-11:
@chipweinberger 

> I'm considering leaving Secure Download Mode off.

Secure ROM DL mode disables the arbitrary stub code execution on the device through ROM DL mode. Since, there is no formal verification that happens on the stub code, if the ROM DL mode is kept enabled then there is an easy way to run arbitrary (untrusted) firmware on the device (through internal memory). For this stub code, if other UARL DL mode related configuration (e.g., disable ICACHE/DCACHE) bits are not protected then it can easily decrypt the entire flash contents.

> DIS_DOWNLOAD_MANUAL_ENCRYPT

This bit disables the encrypted flash writes through UART DL mode. In the Secure DL mode, basic flash writes are still supported and if `DIS_DOWNLOAD_MANUAL_ENCRYPT` is not burned then encrypted flash writes are also possible. If secure boot is enabled, then access to the private key would be required to flash an arbitrary firmware on the device.

> HARD_DIS_JTAG and DIS_USB_JTAG which disables JTAG.

Keeping JTAG enabled can provide full access to the internal memory, peripherals and CPU registers (at-runtime). Ephemeral keys (session specific) can be easily retrieved if there is full access to internal memory of the target.

> So I'll answer my own question. The hack from Jan 2024 doesn't seem that hard to pull off.

So if you look at the technical report (specifically the one targeting XTS-AES design) then it clearly says that:

1. This attack requires external flash emulator interfaced to the target to circumvent the flash speed, erase-cycles limitations
2. After interfacing the emulator, the sample collection spanning over several days is required
3. CPA followed by mounting careful voltage glitch injection is required to execute the shell code on the target 

First thing that we recommended in the advisory is to consider using the SIP with flash pins internally terminated. This can act as the first level of defense, unless the SIP itself is first de-capped to expose the flash pins.

> So in this sense, I don't think Secure Download Mode or any of the other fuses are worth burning.

I would still not recommend this option, it provides a much simpler way to run un-trusted code on the device. 

Hope this helps!

chipweinberger, 2024-03-11:
Thank you for your very great answer. This is incredibly helpful. 

Okay, I will probably use the recommended settings!

closed 2024-03-11 by chipweinberger