espressif/esp-idf #13343: Support both Quad and Octal SPI RAM in one image

silverchris, 2024-03-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,
Is there a way to support both Octal and Quad PSRAM from one image/OTA?
I have a way to detect my hardware from efuse bits, and it would be optimal to support multiple boards with one image.
I don't see any documentation on late PSRAM init. or if these can be changed at run time instead of compile

silverchris, 2024-04-14:
No thoughts on this?

OverTree, 2025-05-15:
I also want this function.

lanteeu, 2025-06-25:
I support request !. Supporting both Quad and Octal PSRAM from one image would be usefull.  I also need to support multiple hardware versions with a single firmware, and currently there seems to be no clean way to do that.
Looking forwarrd good news about implementation this feature.

nebkat, 2025-07-24:
@Icarus113 I've had a very brief look at `esp_psram` and on the surface it would appear this could be achieved by simply renaming `esp_psram_impl_enable` to respective `esp_psram_quad/octal/hex_enable` and calling them sequentially (with similar handling for the 3 other `esp_psram_impl_*` functions). Only other place that seems to reference `CONFIG_SPIRAM_MODE_***` is `mspi_timing_tuning` which is mostly `#ifdef`-ed out on default settings.

I suspect there may be more to it than that, but could you shed some light on whether this would be doable?

We are currently considering migrating a number of product lines from `ESP32-S3-WROOM-1-N16R2` to `ESP32-S3-WROOM-2-N32R16V` but we would very much like to avoid doubling the number of builds required to support them.

ginkgm, 2025-11-22:
Hi @silverchris , @JacekDob, @mutatrum, @nebkat , @lanteeu , @OverTree 

This needs some work and is still in our backlog.

One of the (minor?) blocker is that there is no official method for software to determine whether there is Octal/Quad PSRAM connected. So even if we provide a way to choose at runtime, you will still need to handle the recognition problem and provide the code for the system to choose from Octal or Quad. This is a hardware customization problem. You may consult our sales to see if there is a way to update the hardware: https://www.espressif.com/en/contact-us/sales-questions