espressif/esp-idf #13345: Incomplete Git Repository Detection

jonahbron, 2024-03-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1

### Espressif SoC revision.

n/a

### Operating System used.

Linux

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

n/a

### Power Supply used.

USB

### What is the expected behavior?

n/a

### What is the actual behavior?

n/a

### Steps to reproduce.

1. Build this Nix flake: https://github.com/mirrexagon/nixpkgs-esp-dev

### More Information.

The way in which esp.py detects whether it's in a git repository for version detection appears to pick up non-`esp-idf` repositories that it might happen to exist inside of.

https://github.com/mirrexagon/nixpkgs-esp-dev/issues/5

This causes secondary issues for Nix where the derivation isolation collides with Git's attempts at ensuring the ownership of the Git repo is what is expected.

https://github.com/mirrexagon/nixpkgs-esp-dev/pull/39

dobairoland, 2024-05-23:
We plan to remove the use of git for detecting ESP-IDF version. That will solve this issue as well. 

> to pick up non-esp-idf repositories that it might happen to exist inside of.

But I don't see why this should be necessary. ESP-IDF directory should be left as is and everything else can be added into the project's directory (which can be anywhere else on the disk).

dobairoland, 2024-06-06:
Change of plans. We have to keep the use of git but will try to ignore all git related failures.

closed 2024-06-17 by espressif-bot