espressif/esp-idf #13347: Stack overflow in esp_log_buffer_hexdump_internal when targeting linux 64 bit

matthew-8925, 2024-03-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2

### Espressif SoC revision.

Linux

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

N/A

### Power Supply used.

USB

### What is the expected behavior?

When using `idf.py --preview set-target linux` and running code on Linux that uses ESP_LOG_BUFFER_HEXDUMP it should print without crashing.

### What is the actual behavior?

When using `idf.py --preview set-target linux` and running on Linux on code useing ESP_LOG_BUFFER_HEXDUMP then a stack overflow is detected and the program exits.

The cause is the data buffer passed to ESP_LOG_BUFFER_HEXDUMP contains bytes with the top bit set.
That seems to be signed extended into a 64 bit word when compiled for Linux and sprintf prints 8 chars instead of the expected 2.

https://github.com/espressif/esp-idf/blob/60a2bf6a68867fd468a250966230a067f8d97ea2/components/log/log_buffers.c#L129

Replacing the line with this resolved the issue.
`ptr_hd += sprintf(ptr_hd, " %02x", (unsigned char)ptr_line[i]);`

### Steps to reproduce.

```c++
#include <esp_log.h>
void example(){
  char temp[16] = {0};
  temp[2] = temp[1] = 0xFF;
  ESP_LOG_BUFFER_HEXDUMP("TAG", temp, 16, ESP_LOG_WARN);
}
```

The 0xFF gets sign extended and prints as ffffffff in ESP_LOG_BUFFER_HEXDUMP. That exceeds the expected line length, smashing stack variables.

### Debug Logs.

```plain
$ uname -sp
Linux x86_64
$ ./build/test.elf
W (19511609) TAG: 0x7fdb2fa75bd0   00 ffffffff ffffffff 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*** stack smashing detected ***: terminated
```

matthew-8925, 2024-03-10:
One more fix is needed for Linux targets, the pointer is now 64 bits so extra space is needed in `hd_buffer` for printing `%p`
https://github.com/espressif/esp-idf/blob/60a2bf6a68867fd468a250966230a067f8d97ea2/components/log/log_buffers.c#L123

0xjakob, 2024-03-11:
@matthew-8925 Thanks for pointing this out! We'll take a look at this.

matthew-8925, 2024-03-11:
And a similar issue here raised by AddressSanitizer 
https://github.com/espressif/esp-idf/blob/60a2bf6a68867fd468a250966230a067f8d97ea2/components/log/log_buffers.c#L52

Fix that worked for me is
`sprintf(hex_buffer + 3 * i, "%02x ", (unsigned char)ptr_line[i]);`

closed 2024-03-13 by espressif-bot