espressif/esp-idf #13349: Unstable W5500 on ESP32

binaur1, 2024-03-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have been unable to get the W5500 to function in a stable manner with my ESP32 WeMos LOLIN32 for quite some time.

At the most basic level the ping response is chaotic, indicating the ESP32 isn't correctly responding.  Here is the Wireshark log.

```
734	21.404498	192.168.2.16	192.168.2.32	ICMP	74	Echo (ping) request  id=0x0001, seq=77/19712, ttl=128 (reply in 1154)
1154	24.656223	192.168.2.32	192.168.2.16	ICMP	74	Echo (ping) reply    id=0x0001, seq=77/19712, ttl=64 (request in 734)
1155	24.658772	192.168.2.16	192.168.2.32	ICMP	74	Echo (ping) request  id=0x0001, seq=78/19968, ttl=128 (no response found!)
1177	29.461005	192.168.2.16	192.168.2.32	ICMP	74	Echo (ping) request  id=0x0001, seq=79/20224, ttl=128 (no response found!)
1242	34.462087	192.168.2.16	192.168.2.32	ICMP	74	Echo (ping) request  id=0x0001, seq=80/20480, ttl=128 (no response found!)
1524	42.733543	192.168.2.16	192.168.2.32	ICMP	74	Echo (ping) request  id=0x0001, seq=81/20736, ttl=128 (no response found!)
1551	47.468587	192.168.2.16	192.168.2.32	ICMP	74	Echo (ping) request  id=0x0001, seq=82/20992, ttl=128 (no response found!)
1799	52.469717	192.168.2.16	192.168.2.32	ICMP	74	Echo (ping) request  id=0x0001, seq=83/21248, ttl=128 (no response found!)
1857	57.457995	192.168.2.16	192.168.2.32	ICMP	74	Echo (ping) request  id=0x0001, seq=84/21504, ttl=128 (no response found!)
2225	72.427822	192.168.2.16	192.168.2.32	ICMP	74	Echo (ping) request  id=0x0001, seq=85/21760, ttl=128 (no response found!)
2316	76.971132	192.168.2.16	192.168.2.32	ICMP	74	Echo (ping) request  id=0x0001, seq=86/22016, ttl=128 (reply in 2317)
2317	76.972046	192.168.2.32	192.168.2.16	ICMP	74	Echo (ping) reply    id=0x0001, seq=86/22016, ttl=64 (request in 2316)
2337	77.982148	192.168.2.16	192.168.2.32	ICMP	74	Echo (ping) request  id=0x0001, seq=87/22272, ttl=128 (no response found!)
```

And here are the ping requests
```shell
C:\Program Files (x86)\Nmap>ping 192.168.2.32

Pinging 192.168.2.32 with 32 bytes of data:
Reply from 192.168.2.32: bytes=32 time=3251ms TTL=64
Request timed out.
Request timed out.
Request timed out.

Ping statistics for 192.168.2.32:
    Packets: Sent = 4, Received = 1, Lost = 3 (75% loss),
Approximate round trip times in milli-seconds:
    Minimum = 3251ms, Maximum = 3251ms, Average = 3251ms

C:\Program Files (x86)\Nmap>ping 192.168.2.32

Pinging 192.168.2.32 with 32 bytes of data:
Request timed out.
Request timed out.
Request timed out.
Request timed out.

Ping statistics for 192.168.2.32:
    Packets: Sent = 4, Received = 0, Lost = 4 (100% loss),

C:\Program Files (x86)\Nmap>ping 192.168.2.32

Pinging 192.168.2.32 with 32 bytes of data:
Request timed out.
Reply from 192.168.2.32: bytes=32 time=1ms TTL=64
Request timed out.
Reply from 192.168.2.16: Destination host unreachable.

Ping statistics for 192.168.2.32:
    Packets: Sent = 4, Received = 2, Lost = 2 (50% loss),
Approximate round trip times in milli-seconds:
    Minimum = 1ms, Maximum = 1ms, Average = 1ms
``` 

I have tried many combinations of SPI clock speeds from 5-20 Mhz.

192.168.2.32 is the ESP and 192.168.2.16 is the computer.  They are both wired to the same switch and the switch connects to the router/gateway at 192.168.2.1

The W5500 is connected with all equal length wires of 6" and no other wires are near that would interfere.  I have a dedicated regulator supplying a stable 3.28V.

ESP32 is powered via USB and voltage stable at 3.32V with minimal current loads.

Using ESP-IDF Version: 5.3.0

This is a significantly reduced issues as i'm having similar problems with all TCP communications.

I have attached the basic ethernet example with support for static IP and the sdkconfig i'm using.

[ethernet_example_main.zip](https://github.com/espressif/esp-idf/files/14551661/ethernet_example_main.zip)

Can anyone see something obvious that would explain such problems?

kostaond, 2024-03-11:
Could you please double check that all the GPIO's are connected correctly (including INT)? Please also try to use shorted wires.

binaur1, 2024-03-12:
I scoped INT line and there is a signal.  Also checked that w5500_isr_handler is called so INT is triggering, but according to the results I don't think enough interrupts are generated, but can't be sure.  I also attempted to shorten the wires, but could only get them to about 3" due to the ESP32 and W5500 form factors on the protoboards.  This didn't change any results.
One last attempt was I simplified the network as a direct connection between Win 10 and ESP32 (both static IP).  Same inconsistent ICMP with the ESP not responding in time or at all.

kostaond, 2024-03-12:
What does ESP log says? Any errors? What W5500 module do you use?

binaur1, 2024-03-13:
No errors in the logs, the last messages were from the GOT_IP event callback.  This seems correct for the basic example.
I have the [W5500 Lite](https://solectroshop.com/en/modulos-ethernet/5220-w5500-lite-ethernet-network-modules-spi-5905323231627.html), more compact model.
I also increased the log output to "debug" and no errors/warnings. I did get these messages.  I notice the two len=74 messages as these correspond to the ICMP request, but 5 were sent.  The others are ARP and mDNS from windows.
```
D (5875) w5500.mac: receive len=60
D (6885) w5500.mac: receive len=60
D (11255) w5500.mac: receive len=83
D (37275) w5500.mac: receive len=83
D (38885) w5500.mac: receive len=60
D (38885) w5500.mac: receive len=74
D (41275) w5500.mac: receive len=83
D (43885) w5500.mac: receive len=74
``` 
```
151	154.714258	192.168.2.10	192.168.2.32	ICMP	74	Echo (ping) request  id=0x0001, seq=110/28160, ttl=128 (reply in 152)
152	154.716233	192.168.2.32	192.168.2.10	ICMP	74	Echo (ping) reply    id=0x0001, seq=110/28160, ttl=64 (request in 151)
153	154.719816	192.168.2.10	192.168.2.32	ICMP	74	Echo (ping) request  id=0x0001, seq=111/28416, ttl=128 (no response found!)
156	159.713071	192.168.2.10	192.168.2.32	ICMP	74	Echo (ping) request  id=0x0001, seq=112/28672, ttl=128 (reply in 157)
157	159.714126	192.168.2.32	192.168.2.10	ICMP	74	Echo (ping) reply    id=0x0001, seq=112/28672, ttl=64 (request in 156)
```

mickeyl, 2024-03-18:
The W5500 driver is pretty solid these days (although I never used the ESP32, only the ESP32S3), I'd be surprised if it wasn't a hardware issue.

Two shots in the dark though:

1. Try powering the W5500 lite with 5V.
2. Did you configure the PMODE to autonegotiate? Try one of the fixed modes and perhaps half-duplex modes.

binaur1, 2024-03-18:
The W5500 is a 3.3V device with 5V I/O tolerance per the documentation.  I will make no attempt to supply 5V.  That being said I did force the mode to Half Duplex and 10Mb and that seems to work.  I get less than 1ms ping times and no packet loss.
Unfortunately this is unacceptable, neither explaining not solving the problem.
My analysis so far doesn't indicate a Noise problem, all lines are short with clean transitions (via scope).

mickeyl, 2024-03-18:
I don't know whether there is a low end of tolerable SPI frequency. I'm running my W5500 @ 32MHz which gets me `iperf` speeds of up to 15Mbps. Since I'm using the GPIO matrix, more than 40MHz is unlikely anyways. Do you have a second W5500 to test with? Preferably a non-lite one. I had a few lite ones that were DOA which all the non-lite ones worked.

binaur1, 2024-03-20:
I tested for where it begins to fail.   I am able to achieve Full duplex at 10Mbps only up to 23Mhz SPI.  As soon as I set it to 24, w5500 fails init.
```
E (573) w5500.mac: w5500_reset(305): reset timeout
E (573) w5500.mac: emac_w5500_init(823): reset w5500 failed
E (583) esp_eth: esp_eth_driver_install(228): init mac failed
``` 

This is similar with 100Mps, except I get missed packets/interrupts.

At this point it seems the w5500 is defective.

kostaond, 2024-03-21:
@binaur1 if you want to use any SPI device at high frequencies, you have to properly design the signal path (PCB). Otherwise, you have troubles with signal integrity. Also note that even if you design the PCB properly, you may need to fine tune driver setting, e.g. [input_delay_ns](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/spi_master.html#_CPPv4N29spi_device_interface_config_t14input_delay_nsE).

We do not observe any such issues in our W5500 tests. Therefore, I'm afraid that your issue is hardware related. As such, I'm closing the issue. Feel free to re-open or open a new if you feel there is SW issue and you are able to provide more information to reproduce.

closed 2024-03-21 by kostaond