espressif/esp-idf #13354: New i2c slave driver: multiple bugs

haydenroche5, 2024-03-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-WROOM-1

### Power Supply used.

USB

### What is the expected behavior?

- `s_i2c_handle_rx_fifo_wm` should not overrun the user's receive buffer.
- `i2c_slave_bus_destroy` should clear the slave's interrupt mask.

### What is the actual behavior?

In `s_i2c_handle_rx_fifo_wm`, we have this code:

```
static IRAM_ATTR void s_i2c_handle_rx_fifo_wm(i2c_slave_dev_handle_t i2c_slave, i2c_slave_receive_t *t)
{
    i2c_hal_context_t *hal = &i2c_slave->base->hal;
    uint32_t rx_fifo_cnt;
    i2c_ll_get_rxfifo_cnt(hal->dev, &rx_fifo_cnt);
    i2c_ll_read_rxfifo(hal->dev, i2c_slave->data_buf, rx_fifo_cnt);
    memcpy(t->buffer + i2c_slave->already_receive_len, i2c_slave->data_buf, rx_fifo_cnt);
    i2c_slave->already_receive_len += rx_fifo_cnt;
    t->rcv_fifo_cnt -= rx_fifo_cnt;
}
```

If `rx_fifo_cnt` is greater than the space left in the user's buffer (as indicated by `t->rcv_fifo_cnt`), the user's buffer will be overrun and `t->rcv_fifo_cnt` will go negative.

There may be a related problem in `s_i2c_handle_complete`. It should be pulling the minimum of `rx_fifo_cnt` and `t->rcv_fifo_cnt` out of the FIFO, not just unconditionally reading `t->rcv_fifo_cnt`.

The other problem is in `i2c_slave_bus_destroy`. There, the interrupt mask isn't being cleared. In the old driver, this was done by `i2c_driver_delete`: `i2c_ll_disable_intr_mask(i2c_context[i2c_num].hal.dev, I2C_LL_INTR_MASK);`. If this isn't cleared, after deleting the slave, we can still get interrupts.

### Steps to reproduce.

To reproduce the first problem, have the master send X bytes to the slave. Have the slave read less than X bytes. Now you've hit the error.

To hit the second problem, have the master send X bytes to the slave. Have the slave read X bytes. Delete the slave. Have the master send X bytes again. The slave RX interrupts will fire again, even though the slave was deleted.

mythbuster5, 2024-03-11:
I'm going to take a look, thanks!

mythbuster5, 2024-03-14:
Hi, @haydenroche5 Very glad to here from you about those i2c slave issues, the fix of this issue is on the way. But I'm little worry about this fix, and I'm going to have some topic to discuss with you.

When I designed this API `i2c_slave_receive`, I'm trying to avoid the severe issue on legacy API `i2c_slave_read_buffer`-- losing data randomly. So `i2c_slave_receive` is strongly depends on interrupt can callback to make it unblocking. But the premise is， user knew how much data he is going receive. Otherwise, another issue would happen.

If master sends 20 bytes data, slave receive 10. (This would be possible after this fix.). But if you wanna to get another 10 bytes data, you need another master transaction to trigger an another interrupt. Otherwise, this 10 bytes data will not be got out anymore.

So, I'm thinking to have another API without any `size` restriction, maybe called `i2c_slave_receive_all`. That means you can get all data that master sent from start to end. And buffer & size will be given in a callback. The only thing that user need to know is the maximum data to receive, and prepare a sufficient buffer ahead of time.

Thanks again for giving so much good advice. I'm taking look in https://github.com/espressif/esp-idf/issues/13377. If you have any other opinion on slave driver, please poke me in that thread.

closed 2024-03-16 by espressif-bot

haydenroche5, 2024-03-18:
> So, I'm thinking to have another API without any size restriction, maybe called i2c_slave_receive_all. That means you can get all data that master sent from start to end. And buffer & size will be [...]

That sounds reasonable to me.

haydenroche5, 2024-05-01:
I see on the master branch that `s_i2c_handle_rx_fifo_wm` has been updated to fix the problem:

```
static IRAM_ATTR void s_i2c_handle_rx_fifo_wm(i2c_slave_dev_handle_t i2c_slave, i2c_slave_receive_t *t)
{
    i2c_hal_context_t *hal = &i2c_slave->base->hal;
    uint32_t rx_fifo_cnt;
    i2c_ll_get_rxfifo_cnt(hal->dev, &rx_fifo_cnt);
    uint32_t fifo_cnt_rd = MIN(t->rcv_fifo_cnt, rx_fifo_cnt);
    i2c_ll_read_rxfifo(hal->dev, i2c_slave->data_buf, fifo_cnt_rd);
    memcpy(t->buffer + i2c_slave->already_receive_len, i2c_slave->data_buf, fifo_cnt_rd);
    i2c_slave->already_receive_len += fifo_cnt_rd;
    t->rcv_fifo_cnt -= fifo_cnt_rd;
}
```

Specifically, the `uint32_t fifo_cnt_rd = MIN(t->rcv_fifo_cnt, rx_fifo_cnt);` line. But I do not see a corresponding change in `s_i2c_handle_complete`.

Further, I think the implementation of `s_i2c_handle_complete` may have a related problem. Imagine we provided a huge buffer for receiving data via i2c_slave_receive. The master then does a transaction and it doesn't fill the buffer (this is a scenario where the slave does NOT know how much data the master will send, which is true for our protocol). Then this line

```
i2c_ll_read_rxfifo(hal->dev, i2c_slave->data_buf, t->rcv_fifo_cnt);
```

will overread the FIFO, because `t->rcv_fifo_cnt` is much larger than `rx_fifo_cnt` (the actual amount of RX data in the FIFO).

I suppose this would be resolved if I had access to the `i2c_slave_receive_all` function you proposed. I am wondering if you can give an update on if that's the direction you all are going, or if you have new plans.

Thanks for all the time and effort!