espressif/esp-idf #13355: mbedtls crashing on invallid MAC when sending lot of messages

bartman101, 2024-03-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

### IDF version
v5.2 commit bf17be96b46bd5ab81582e75fa4a8c682841b649
#### Mbed TLS version
branch mbedtls-3.5.2-idf commit [09bba1](https://github.com/espressif/mbedtls/commit/09bba150d0d822aad2e58d71723f5407da5c21e0)
#### lwIP version
branch 2.1.3-esp commit [f792214](https://github.com/espressif/esp-lwip/commit/f79221431fa9042b3572d271d687de66da7560c4)
### Setup
An STM connected with Ethernet to an esp32-wroom-32e. 
Packets are encrypted by Mbed TLS and sent by lwIP. 

Using freeRTOS with 3 tasks, 
1. the networking task responsible for the networking part
2. the task that handles the incoming and outgoing messages
3. the task that handles the actual messages

The tasks communicate through RTOS queues and wait indefinitely on a new item on the queue to process.

### Information
When trying to send a firmware update from the STM to the ESP we are sending packets with a size of 1000 bytes. The next packet is only sent when the previous packet has been handled so the STM is not spamming the packets. In this ping-pong setup we experience spurious, but consistently reproducible errors in MbedTLS, namely the error below.

### Problem
Sometimes when transferring these packets, the mbedtls_ssl_read function return `-0x7180` (MBEDTLS_ERR_SSL_INVALID_MAC). 
```
W (13510) mbedtls: ssl_msg.c:1662 mbedtls_cipher_auth_decrypt_ext() returned -25344 (-0x6300)
W (13516) mbedtls: ssl_msg.c:4010 ssl_decrypt_buf() returned -29056 (-0x7180)
W (13525) mbedtls: ssl_msg.c:4165 ssl_get_next_record() returned -29056 (-0x7180)
W (13532) mbedtls: ssl_msg.c:5748 mbedtls_ssl_read_record() returned -29056 (-0x7180)
```

It looks like some buffer is overflowing in lwIP but I have tried to enlarge the LWIP_TCPIP_RECVMBOX_SIZE and the LWIP_TCP_RECVMBOX_SIZE but this did not make the error disappear. 

I tried changing some settings in the menuconfig like enlarging the buffers and giving the different tasks more stack size but this all did not help. 

Increasing sizes only seems to delay the problem from happening and decreasing sizes seems to accelerate the occurrence.

What could cause this MBEDTLS_ERR_SSL_INVALID_MAC error? Is there a way to debug the incoming TLS message or Ethernet packets and their associated buffers?

### minimal sdkconfig
```
CONFIG_SECURE_BOOT=y
CONFIG_SECURE_BOOT_SIGNING_KEY="development_key.pem"
CONFIG_SECURE_BOOT_INSECURE=y
CONFIG_SECURE_BOOT_ALLOW_JTAG=y
CONFIG_APP_RETRIEVE_LEN_ELF_SHA=16
CONFIG_ESPTOOLPY_FLASHSIZE_8MB=y
CONFIG_PARTITION_TABLE_CUSTOM=y
CONFIG_PARTITION_TABLE_OFFSET=0x10000
CONFIG_BROKER_MAX_SUBSCRIBERS=7
CONFIG_ETH_MDC_GPIO=32
CONFIG_ETH_MDIO_GPIO=33
CONFIG_ETH_PHY_ADDR=0
CONFIG_LOG_LEVEL_NETWORKINGSERVICE_DEBUG=y
# CONFIG_TWAI_ERRATA_FIX_BUS_OFF_REC is not set
# CONFIG_TWAI_ERRATA_FIX_TX_INTR_LOST is not set
# CONFIG_TWAI_ERRATA_FIX_RX_FRAME_INVALID is not set
# CONFIG_TWAI_ERRATA_FIX_RX_FIFO_CORRUPT is not set
CONFIG_ETH_DMA_BUFFER_SIZE=1048
CONFIG_ETH_DMA_RX_BUFFER_NUM=5
CONFIG_ETH_DMA_TX_BUFFER_NUM=5
CONFIG_ETH_IRAM_OPTIMIZATION=y
CONFIG_ESP_HTTP_CLIENT_ENABLE_DIGEST_AUTH=y
CONFIG_ESP32_REV_MIN_3=y
CONFIG_ESP_SLEEP_POWER_DOWN_FLASH=y
CONFIG_ESP_PHY_REDUCE_TX_POWER=y
CONFIG_ESP_SYSTEM_ESP32_SRAM1_REGION_AS_IRAM=y
CONFIG_ESP_TASK_WDT_PANIC=y
CONFIG_ESP_DEBUG_STUBS_ENABLE=y
CONFIG_ESP_IPC_TASK_STACK_SIZE=2048
# CONFIG_ESP_WIFI_STA_DISCONNECTED_PM_ENABLE is not set
CONFIG_FREERTOS_HZ=1000
CONFIG_FREERTOS_TIMER_TASK_STACK_DEPTH=8196
CONFIG_FREERTOS_WATCHPOINT_END_OF_STACK=y
CONFIG_FREERTOS_ISR_STACKSIZE=2096
CONFIG_LOG_DEFAULT_LEVEL_DEBUG=y
CONFIG_LWIP_LOCAL_HOSTNAME="Nano_Charge_2.0"
CONFIG_LWIP_CHECK_THREAD_SAFETY=y
CONFIG_LWIP_L2_TO_L3_COPY=y
CONFIG_LWIP_IP_FORWARD=y
CONFIG_LWIP_IPV4_NAPT=y
CONFIG_LWIP_STATS=y
CONFIG_LWIP_TCPIP_RECVMBOX_SIZE=64
CONFIG_LWIP_TCP_SYNMAXRTX=6
CONFIG_LWIP_TCP_MSS=1460
CONFIG_LWIP_TCP_SND_BUF_DEFAULT=5840
CONFIG_LWIP_TCP_WND_DEFAULT=5840
CONFIG_LWIP_DEBUG=y
CONFIG_LWIP_DEBUG_ESP_LOG=y
CONFIG_LWIP_NETIF_DEBUG=y
CONFIG_LWIP_PBUF_DEBUG=y
CONFIG_LWIP_SOCKETS_DEBUG=y
CONFIG_LWIP_IP_DEBUG=y
CONFIG_MBEDTLS_SSL_IN_CONTENT_LEN=4096
CONFIG_MBEDTLS_DYNAMIC_BUFFER=y
CONFIG_MBEDTLS_DYNAMIC_FREE_CONFIG_DATA=y
CONFIG_MBEDTLS_DEBUG=y
CONFIG_MBEDTLS_DEBUG_LEVEL_DEBUG=y
CONFIG_MBEDTLS_ECP_RESTARTABLE=y
CONFIG_MBEDTLS_GCM_SUPPORT_NON_AES_CIPHER=y
CONFIG_VFS_SELECT_IN_RAM=y
```

AdityaHPatwardhan, 2024-03-11:
Thanks for you detailed description of the issue.
Based on your observation,
> Increasing sizes only seems to delay the problem from happening and decreasing sizes seems to accelerate the occurrence.
it seems it is indeed related to the memory.

What happens after you get the error?
 Does the error stop the operation completely when it occurs
 or
 it occurs -> some operations succeed in the mean time -> it re-occurs after some time.

I am wondering if enabling heap poisoning would help here.
Have you tried setting [Heap corruption detection](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-heap-corruption-detection) to `Light impact`  and then executing the example?

Thanks,
Aditya

bartman101, 2024-03-11:
Thanks for your reaction Aditya, 
When an unexpected read result is received we clear the socket and set up a new one which then works normally until it crashes again. I will take a look at the heap corruption detection so I will come back on that. 

Thanks,
Bart

mahavirj, 2024-03-12:
@bartman101 

> CONFIG_MBEDTLS_SSL_IN_CONTENT_LEN=4096

Any specific reason that you have reduced the SSL in content length? Please try revering to the default value (16384) once. Reference: https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/kconfig.html#config-mbedtls-ssl-max-content-len

closed 2024-04-01 by Alvin1Zhang