espressif/esp-idf #13357: Disable BT_NIMBLE_SECURITY_ENABLE will cause build failure in v5.2.1.

djytw, 2024-03-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Disable BT_NIMBLE_SECURITY_ENABLE in menuconfig worked fine in version <= 5.2, but failed to build in 5.2.1

### What is the actual behavior?

Disable BT_NIMBLE_SECURITY_ENABLE will cause build failure.

### Steps to reproduce.

1. Use esp-idf v5.2.1 target ESP32
2. cd `examples/bluetooth/nimble/ble_spp/spp_server`
3. `idf.py menuconfig` and goto Component config → Bluetooth → NimBLE Options, press 'N' on 'Enable BLE SM feature'
4. `idf.py build`
5. fail
6. change to v5.2
7. success

### Build or installation Logs.

```plain
In file included from /esp-idf/components/bt/host/nimble/nimble/nimble/host/include/host/ble_gap.h:31,
                 from /esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gap_priv.h:26,
                 from /esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_priv.h:27,
                 from /esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_att_svr.c:27:
/esp-idf/components/bt/host/nimble/nimble/nimble/host/include/host/ble_hs.h:77:37: error: expected identifier or '(' before numeric constant
   77 | #define BLE_HS_ENOTSUP              8
      |                                     ^
/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_sm_priv.h:413:48: note: in expansion of macro 'BLE_HS_ENOTSUP'
  413 | #define ble_sm_alg_aes_cmac(key, in, len, out) BLE_HS_ENOTSUP
      |                                                ^~~~~~~~~~~~~~
/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_sm_priv.h:435:5: note: in expansion of macro 'ble_sm_alg_aes_cmac'
  435 | int ble_sm_alg_aes_cmac(const uint8_t *key, const uint8_t *in, size_t len, uint8_t *out);
      |     ^~~~~~~~~~~~~~~~~~~
In file included from /esp-idf/components/bt/host/nimble/nimble/porting/nimble/include/syscfg/syscfg.h:9,
                 from /esp-idf/components/bt/host/nimble/nimble/porting/nimble/include/os/os.h:33,
                 from /esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_att_svr.c:23:
/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_att_svr.c: In function 'ble_att_svr_rx_notify':
/esp-idf/components/bt/host/nimble/port/include/esp_nimble_cfg.h:847:34: error: 'CONFIG_BT_NIMBLE_SM_SC_LVL' undeclared (first use in this function); did you mean 'CONFIG_BT_NIMBLE_LOG_LEVEL'?
  847 | #define MYNEWT_VAL_BLE_SM_SC_LVL CONFIG_BT_NIMBLE_SM_SC_LVL
      |                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~
/esp-idf/components/bt/host/nimble/port/include/esp_nimble_cfg.h:18:49: note: in expansion of macro 'MYNEWT_VAL_BLE_SM_SC_LVL'
   18 | #define MYNEWT_VAL(_name)                       MYNEWT_VAL_ ## _name
      |                                                 ^~~~~~~~~~~
/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_att_svr.c:2937:9: note: in expansion of macro 'MYNEWT_VAL'
 2937 |     if (MYNEWT_VAL(BLE_SM_SC_LVL) >= 2 && !sec_state.encrypted) {
      |         ^~~~~~~~~~
/esp-idf/components/bt/host/nimble/port/include/esp_nimble_cfg.h:847:34: note: each undeclared identifier is reported only once for each function it appears in
  847 | #define MYNEWT_VAL_BLE_SM_SC_LVL CONFIG_BT_NIMBLE_SM_SC_LVL
      |                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~
/esp-idf/components/bt/host/nimble/port/include/esp_nimble_cfg.h:18:49: note: in expansion of macro 'MYNEWT_VAL_BLE_SM_SC_LVL'
   18 | #define MYNEWT_VAL(_name)                       MYNEWT_VAL_ ## _name
      |                                                 ^~~~~~~~~~~
/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_att_svr.c:2937:9: note: in expansion of macro 'MYNEWT_VAL'
 2937 |     if (MYNEWT_VAL(BLE_SM_SC_LVL) >= 2 && !sec_state.encrypted) {
      |         ^~~~~~~~~~
/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_att_svr.c: In function 'ble_att_svr_rx_indicate':
/esp-idf/components/bt/host/nimble/port/include/esp_nimble_cfg.h:847:34: error: 'CONFIG_BT_NIMBLE_SM_SC_LVL' undeclared (first use in this function); did you mean 'CONFIG_BT_NIMBLE_LOG_LEVEL'?
  847 | #define MYNEWT_VAL_BLE_SM_SC_LVL CONFIG_BT_NIMBLE_SM_SC_LVL
      |                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~
/esp-idf/components/bt/host/nimble/port/include/esp_nimble_cfg.h:18:49: note: in expansion of macro 'MYNEWT_VAL_BLE_SM_SC_LVL'
   18 | #define MYNEWT_VAL(_name)                       MYNEWT_VAL_ ## _name
      |                                                 ^~~~~~~~~~~
/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_att_svr.c:3028:9: note: in expansion of macro 'MYNEWT_VAL'
 3028 |     if (MYNEWT_VAL(BLE_SM_SC_LVL) >= 2 && !sec_state.encrypted) {
      |         ^~~~~~~~~~
ninja: build stopped: subcommand failed.
```
```

RoshanESP, 2024-03-19:
Hi @djytw , Thank you for pointing out this error. The change has been pushed and will be available soon.

rahult-github, 2024-04-06:
Change merged via https://github.com/espressif/esp-idf/commit/1a9f9d69eac9c787cc36dd478a228c927b05e34d

closed 2024-04-06 by rahult-github