espressif/esp-idf #13360: cmock: can't use :subdir option because of the way generated c files are added to the MOCK_GENERATED_SRCS list

BastienGermond, 2024-03-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,

While trying to play with CMock for my project I noticed that it is not possible to mock component that use some kind of namespaced headers (e.g `#include <myproject/logic.h>`) and for that use [:subdir option](https://github.com/ThrowTheSwitch/CMock/blob/eeecc49ce8af123cf8ad40efdb9673e37b56230f/docs/CMock_Summary.md?plain=1#L462) from CMock.

The issue is caused by the way mocked files are added in: https://github.com/espressif/esp-idf/blob/60a2bf6a68867fd468a250966230a067f8d97ea2/tools/cmake/component.cmake#L553

Every mocked files are expected to be directly under `/mocks/`, is it possible to maybe add another configuration option to `idf_component_mock` to specify a `SUBFOLDER` ?

Since the :subdir option is closely related to :mock_path maybe it would be useful to have by default `mocks` and something else if specified otherwise ?

---

Example of mock_config.yaml used :

```
:cmock:
  :subdir: "myproject"
  :plugins:
    - :ignore
    - :callback
```

Currently, the build fails because it will not be able to find the c files generated under `mocks/myproject/Mock<...>.c` and expect c files `mocks/Mock<...>.c`.

0xjakob, 2024-03-12:
Hi @BastienGermond! It seems that there are two different aspects here:
1. The place where the generated mock implementations are placed.
2. Problems including the correct header files.

Do you have problems with both or just one of these aspects? 
Regarding 1.: Usually, it shouldn't matter where the source files (*.c) are placed, the includes are absolute paths. Furthermore, the IDF build system should take care of setting up the compiler calls so all source files are found. Do you have source files with the same name which would conflict with each other after the mocks have been generated?
Regarding 2.: This should also work as long as the include directories  are set in the `CMakeLists.txt` of your mocking component (the one that is creating the mocks).

BastienGermond, 2024-03-12:
Hi @0xjakob,

> 1. The place where the generated mock implementations are placed.

The issue is that the files generated with the cmock_configuration `:subdir: myproject` would be created in this path `${MOCK_GEN_DIR}/mocks/myproject/Mock${file_without_dir}.[ch]`.

What I'm trying to say is that currently the path is hardcoded expecting the default values from CMock (`:subdir: ""`, `:mock_path: "mocks"` and `:mock_prefix: "Mock"`).

In case we have a mock_config.yaml like this :

```
:cmock:
  :subdir: "myproject"
  :mock_prefix: "Espressif"
  :mock_path: "EspMocks"
  :plugins:
    - :ignore
    - :callback
``` 

This would generate files following this path : `${MOCK_GEN_DIR}/EspMocks/myproject/Espressif${file_without_dir}.[ch]`

> the IDF build system should take care of setting up the compiler calls so all source files are found.

https://github.com/espressif/esp-idf/blob/60a2bf6a68867fd468a250966230a067f8d97ea2/tools/cmake/component.cmake#L552-L553

https://github.com/espressif/esp-idf/blob/60a2bf6a68867fd468a250966230a067f8d97ea2/tools/cmake/component.cmake#L558-L560

And because of this different path the c-files are not registered correctly (and thus not found/compiled during compilation causing missing symbols during linking).

> 2. Problems including the correct header files.

I also have this problem caused by not using :subdir, in my project I use a `#include "myproject/nvs.h"` but generating mocks for this without subdir will include "nvs.h" and not the expected "myproject/nvs.h" in `Mocknvs.h`.

I patched it the first time the same way esp-idf does for freertos with adding manually my correct header with :includes_h_pre_orig_header.

https://github.com/espressif/esp-idf/blob/60a2bf6a68867fd468a250966230a067f8d97ea2/tools/mocks/freertos/mock/mock_config.yaml#L2-L4

CMock still add an `#include "nvs.h"` in the mock header generated but at least this added afterward the correct `#include "myproject/nvs.h"`.

However, in case the "nvs.h" was not something found in any include path or something that is not wanted in this header, the compilation will fail with an include not found error or maybe collide with the right header (myproject/nvs.h).

---

I apologize if my explanation was unclear. Please let me know if you did not catch what I mean and I will try to rephrase it :smile:.

0xjakob, 2024-03-12:
It starts to make much more sense now, thanks! Supporting `:subdir:` makes sense then, in particular because of the includes.

Regarding support for custom `:mock_path:` and `:mock_dir:`: I think we could also add that option, although I don't quite understand why they are necessary. Taking `:subdir:` aside, is there a situation where you have a custom `:mock_path:` or `:mock_prefix:` configured in IDF?

BastienGermond, 2024-03-12:
> Taking :subdir: aside, is there a situation where you have a custom :mock_path: or :mock_prefix: configured in IDF?

Not really, and I agree that I don't see a use case right now. Maybe some company may have some convention about filename enforced by tools ?

I was mentioning them only since they will not work with the current process, the real issue for me was the missing `:subdir` support.

closed 2025-03-19 by espressif-bot