espressif/esp-idf #13362: 运行peripherals/usb/device/tusb_serial_device示例报错Serial data stream stopped: Possible serial noise or corruption.

zhouayi, 2024-03-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.0

### Espressif SoC revision.

esp32s3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32-s3-eye

### Power Supply used.

USB

### What is the expected behavior?

原本应该烧录成功，运行程序

### What is the actual behavior?

[image]
[2/3] cmd.exe /C "cd /D E:\Softwares\Espressif\frameworks\esp-idf-v5.0.4\components\esptool_py && E:\Softwares\Espressif\tools\cmake\3.24.0\bin\cmake.exe -D IDF_PATH=E:/Softwares/Espressif/frameworks/esp-idf-v5.0.4 -D SERIAL_TOOL=E:/Softwares/Espressif/python_env/idf5.0_py3.11_env/Scripts/python.exe;;E:/Softwares/Espressif/frameworks/esp-idf-v5.0.4/components/esptool_py/esptool/esptool.py;--chip;esp32s3 -D SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;--no-stub;write_flash;@flash_args -D WORKING_DIRECTORY=E:/Softwares/Espressif/frameworks/esp-idf-v5.0.4/examples/peripherals/usb/device/tusb_serial_device/build -P E:/Softwares/Espressif/frameworks/esp-idf-v5.0.4/components/esptool_py/run_serial_tool.cmake"
esptool esp32s3 -p COM5 -b 460800 --before=default_reset --after=hard_reset --no-stub write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x0 bootloader/bootloader.bin 0x10000 tusb_serial_device.bin 0x8000 partition_table/partition-table.bin
esptool.py v4.6.2
Serial port COM5
Connecting......................................

A fatal error occurred: Failed to connect to ESP32-S3: Serial data stream stopped: Possible serial noise or corruption.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
CMake Error at run_serial_tool.cmake:58 (message):

  E:/Softwares/Espressif/python_env/idf5.0_py3.11_env/Scripts/python.exe;;E:/Softwares/Espressif/frameworks/esp-idf-v5.0.4/components/esptool_py/esptool/esptool.py;--chip;esp32s3
  failed

FAILED: CMakeFiles/flash E:/Softwares/Espressif/frameworks/esp-idf-v5.0.4/examples/peripherals/usb/device/tusb_serial_device/build/CMakeFiles/flash
cmd.exe /C "cd /D E:\Softwares\Espressif\frameworks\esp-idf-v5.0.4\components\esptool_py && E:\Softwares\Espressif\tools\cmake\3.24.0\bin\cmake.exe -D IDF_PATH=E:/Softwares/Espressif/frameworks/esp-idf-v5.0.4 -D SERIAL_TOOL=E:/Softwares/Espressif/python_env/idf5.0_py3.11_env/Scripts/python.exe;;E:/Softwares/Espressif/frameworks/esp-idf-v5.0.4/components/esptool_py/esptool/esptool.py;--chip;esp32s3 -D SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;--no-stub;write_flash;@flash_args -D WORKING_DIRECTORY=E:/Softwares/Espressif/frameworks/esp-idf-v5.0.4/examples/peripherals/usb/device/tusb_serial_device/build -P E:/Softwares/Espressif/frameworks/esp-idf-v5.0.4/components/esptool_py/run_serial_tool.cmake"
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the E:\Softwares\Espressif\frameworks\esp-idf-v5.0.4\examples\peripherals\usb\device\tusb_serial_device\build\log\idf_py_stderr_output_2648 and E:\Softwares\Espressif\frameworks\esp-idf-v5.0.4\examples\peripherals\usb\device\tusb_serial_device\build\log\idf_py_stdout_output_2648

### Steps to reproduce.

1. Step
没有修改任何代码，idf.py set-target esp32s3
2. Step
idf.py flash monitor

zhouayi, 2024-03-12:
发生错误后我其他的程序也全部烧录不了了

peter-marcisovsky, 2024-03-12:
Hello @zhouayi please try a following: 

- connect your ESP to your PC
- press and hold the BOOT button
- while holding the BOOT button press the RST button
- release the BOOT button
- try if the esp32 can be found by running `esptool.py chip_id`

Please let me know what does the output look like.

zhouayi, 2024-03-12:
> Hello @zhouayi please try a following:
> 
> * connect your ESP to your PC
> * press and hold the BOOT button
> * while holding the BOOT button press the RST button
> * release the BOOT button
> * try if the esp32 can be found by running `esptool.py chip_id`
> 
> Please let me know what does the output look like.

Thank you for your reply, the output is :
~$ esptool.py chip_id
esptool.py v4.6.2
Found 2 serial ports
Serial port COM13
Connecting...
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 34:85:18:a8:98:64
Uploading stub...
Running stub...
Stub running...
Warning: ESP32-S3 has no Chip ID. Reading MAC instead.
MAC: 34:85:18:a8:98:64
Hard resetting via RTS pin...

peter-marcisovsky, 2024-03-12:
Thanks for reporting. 
What is happening is, that you are trying to use 2 peripherals connected to the same GPIOs at the same time. 
- esp32-s3-eye has only one USB port, which is connected directly to the D+, D- pins of the esp32s3
- now, you want to use one esp32-s3-eye USB port for 2 things:
    - flash/debug/log (USB-Serial-JTAG peripheral) 
    - TinyUSB application (USB-OTG peripheral)
    
Once you flash your TinyUSB application, the USB-Serial-JTAG automatically turns off, so you are not able to flash/debug/log your applicaton on the esp32-s3-eye board. By performing the BOOT RST buttons sequence, which I asked you to do in my previous message, you manually enter a flash mode. After performing the sequence, you are able to flash your esp32-s3-eye board.

For example the [ESP32-S3-DevKitC-1 v1.1](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/hw-reference/esp32s3/user-guide-devkitc-1.html) features 2 USB ports. One for flash/log, the other one for debug or USB applicatons. Using this board, you are able to run your USB application while logging the output at the same time.

Is there a particular application you are trying to develop using esp32-s3-eye?

closed 2024-03-25 by peter-marcisovsky