espressif/esp-idf #13363: how to config connection event length determination method in menuconfig

luacerbb, 2024-03-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

The config path, Component config → Bluetooth → Controller Options → Connection event length determination method.
I want to know how to config connection event length determination method in menuconfig,  in menuconfig , I can see three choice,ORIGINAL,Use CE parameter for HCI command,Use Espressif self-defined method. I don't know the difference between these three options,help me,thanks

weiyuhannnn, 2024-03-13:
Hello @luacerbb, Thanks for choosing Espressif product! May I know which chip are you using?

luacerbb, 2024-03-13:
> Hello @luacerbb, Thanks for choosing Espressif product! May I know which chip are you using?

Thanks for your reply,I use esp32-s3 chip

luacerbb, 2024-04-07:
@espressif-bot @zhp0406  Why nobody answer my question

esp-cjh, 2024-04-07:
Hi, @luacerbb 

1 `ORIGINAL` mode means that the BT controller will calculate and use the maximum and minimum connection event length.
2 `Use CE parameter for HCI command` mode means you can determine the connection event length by setting the Min_CE_Length and Max_CE_length params when creating connection or updating connection params. Which API you should use depends on which host you are using.
3 `Use Espressif self-defined method` mode is for internal test and no API is provide to the user. This mode is not recommended.

So you should set the config to `Use CE parameter for HCI command` mode if you need to change the connection event length.

luacerbb, 2024-04-11:
@esp-cjh ok,thanks

weiyuhannnn, 2024-11-01:
@luacerbb Thanks for reporting, feel free to reopen with more updates.

closed 2024-11-01 by weiyuhannnn