espressif/esp-idf #13365: 一直连上不wifi

Leadrive, 2024-03-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.6

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3

### Power Supply used.

USB

### What is the expected behavior?

希望能连上wifi

### What is the actual behavior?

开始是可以连上wifi的，大概是过了几个小时或一天（？）出现断开后，一直尝试都连不上。下面是连不上时报的部分错误
````
[12:00:52.608]收←◆I (446016) wifi:starting SA query procedure with 44:56:e2:2f:9e:e1
I (446017) wifi:Send SA Query req with transaction id 65f

[12:00:52.808]收←◆I (446217) wifi:Send SA Query req with transaction id 660

[12:00:53.008]收←◆I (446417) wifi:Send SA Query req with transaction id 661

[12:00:53.208]收←◆I (446617) wifi:Send SA Query req with transaction id 662

[12:00:53.408]收←◆I (446817) wifi:Send SA Query req with transaction id 663

[12:00:53.608]收←◆I (447017) wifi:Send SA Query req with transaction id 664
I (447041) wifi:No response to 6 SA Queries, reset connection sending disassoc
I (447041) wifi:state: run -> init (d100)
I (447041) wifi:pm stop, total sleep time: 25426578 us / 30536170 us

I (447047) wifi:new:<13,0>, old:<13,0>, ap:<255,255>, sta:<13,0>, prof:1
[0;33mW (447054) [comm_wifi, 74]: disconnected reason: 209[0m

[12:00:57.948]收←◆[0;32mI (451356) comm_wifi: WIFI connecting...4[0m
============  在这里，我的程序每隔30秒判断，如果WIFI断开了，做一次 esp_wifi_connect()重连操作，但观察没起作用 ======
[12:00:58.035]收←◆I (451443) wifi:new:<13,0>, old:<13,0>, ap:<255,255>, sta:<13,0>, prof:1
I (451444) wifi:state: init -> auth (b0)
I (451446) wifi:state: auth -> assoc (0)
I (451450) wifi:state: assoc -> run (10)
I (451481) wifi:state: run -> init (ec0)
I (451481) wifi:new:<13,0>, old:<13,0>, ap:<255,255>, sta:<13,0>, prof:1
[0;33mW (451482) [comm_wifi, 74]: disconnected reason: 14[0m
[0;33mW (451485) [comm_wifi, 93]: Wi-Fi 4-way handshake failed, count: 4[0m
[0;31mE (451492) [comm_wifi, 96]: WIFI_REASON_HANDSHAKE_TIMEOUT error, delay 30 seconds, retry...[0m
````
此时路由器上，同一个ip，先后出现不同的mac
[image]
会不会跟路由器某个设置有关？
[image]

### Steps to reproduce.

连上WIFI后，运行一天或更多时间，会出现断开，然后就一直连不上。

mhdong, 2024-03-13:
Hi @Leadrive 
1. 请帮忙提供一下路由器配置的参数，可以截图相关的配置页面。
2. 将log等级开到 debug，并把supplicant的log打开。1）menuconfig->component config->log output-> Debug 2) menuconfig -> component config -> supplicant -> Print debug messages from WPA Supplicant
3. 如果方便的话，请帮忙在连不上的时候使用 wireshark进行抓包。

Leadrive, 2024-03-14:
@mhdong 设备在客户那里，设备剩余内存只有30k，客户也不懂抓包

Leadrive, 2024-03-14:
这是路由器的日志
````
Security Log
------------------------
2021-09-21T19:10:22+08:00 ID 3: Authorized login success::U admin:L 6:N HTTP:P 80:IP 192.168.101.2
2021-09-21T19:10:24+08:00 ID 3: Authorized login success::U admin:L 6:N HTTP:P 80:IP 192.168.101.2
1970-01-01T08:03:18+08:00 ID 4: Authorized login fail::U admin:L 4:N HTTP:P 80:IP 192.168.101.2
1970-01-01T08:03:26+08:00 ID 3: Authorized login success::U admin:L 6:N HTTP:P 80:IP 192.168.101.2
2024-01-26T17:32:18+08:00 ID 3: Authorized login success::U admin:L 6:N HTTP:P 80:IP 192.168.101.3
2024-01-26T17:39:10+08:00 ID 4: Authorized login fail::U root:N HTTP:P 80:IP 192.168.101.3
2024-03-11T12:04:00+08:00 ID 3: Authorized login success::U admin:L 6:N HTTP:P 80:IP 192.168.101.5
2024-03-11T12:04:02+08:00 ID 3: Authorized login success::U admin:L 6:N HTTP:P 80:IP 192.168.101.5
2024-03-11T14:15:31+08:00 ID 3: Authorized login success::U admin:L 6:N HTTP:P 80:IP 192.168.101.5
1970-01-01T08:00:34+08:00 ID 3: Authorized login success::U admin:L 6:N HTTP:P 80:IP 192.168.101.2
1970-01-01T08:00:36+08:00 ID 3: Authorized login success::U admin:L 6:N HTTP:P 80:IP 192.168.101.2
1970-01-01T08:00:40+08:00 ID 3: Authorized login success::U admin:L 6:N HTTP:P 80:IP 192.168.101.2
1970-01-01T08:00:43+08:00 ID 3: Authorized login success::U admin:L 6:N HTTP:P 80:IP 192.168.101.2
2024-03-12T23:01:36+08:00 ID 3: Authorized login success::U admin:L 6:N HTTP:P 80:IP 192.168.101.2
1970-01-01T08:00:33+08:00 ID 3: Authorized login success::U admin:L 6:N HTTP:P 80:IP 192.168.101.2
1970-01-01T08:01:04+08:00 ID 3: Authorized login success::U admin:L 6:N HTTP:P 80:IP 192.168.101.2
2024-03-14T09:54:05+08:00 ID 3: Authorized login success::U admin:L 6:N HTTP:P 80:IP 192.168.101.2
2024-03-14T09:54:07+08:00 ID 3: Authorized login success::U admin:L 6:N HTTP:P 80:IP 192.168.101.2
Mar 14 10:04:52 kern.notice kernel: klogd started: BusyBox v1.30.1 (2020-08-19 14:41:18 CST)
Mar 14 10:04:52 kern.notice CEVENT_APP[3563]: 1710381892892 wl0      68:B6:B3:55:14:68     0 0000 0000 0000 Driver   RX  AUTH_RX                  
````

mhdong, 2024-03-26:
Hi @Leadrive 
同款路由器我们已经购买，在复现问题过程中有什么需要注意的吗？可以提供复现的demo吗？或者只要连接路由器，等到运行一天或更多时间，就会出现断开，然后就一直连不上？

Leadrive, 2024-03-26:
@mhdong 我也买了路由器，目前还在测试中。此路由器没有多余的设置，我目前是wan连局域网，还没测试WAN连光猫的情况。
demo就是 examples/mesh/ip_internal_commuinication
注释了 mesh_netifs_stop() 调用

mhdong, 2024-06-14:
Hi @Leadrive 
我本地没有复现和客户相同的问题。
看到DCHP 服务器配置，起始： 192.168.101.2 结束：192.168.101.10，而路由器已经分配到 192.168.101.10
如果重连时，路由器已经将ip地址分配完，则无法完成连接。

我这边复现的情况是这样的，如果将分配的ip地址分配完，连接在auth阶段由于超时无法成功。

:>sta -C -s coex -p 12345678

+JAP:OK
I (895899) wifi:new:<13,0>, old:<13,0>, ap:<255,255>, sta:<13,0>, prof:1, snd_ch_cfg:0x0
I (895899) wifi:state: init -> auth (0xb0)
I (896899) wifi:state: auth -> init (0x200)
I (896899) wifi:new:<13,0>, old:<13,0>, ap:<255,255>, sta:<13,0>, prof:1, snd_ch_cfg:0x0

+JAP:DISCONNECTED,0,2

+JAP:DISCONNECTED,0,205
I (899349) wifi:new:<13,0>, old:<13,0>, ap:<255,255>, sta:<13,0>, prof:1, snd_ch_cfg:0x0
I (899349) wifi:state: init -> auth (0xb0)
I (900349) wifi:state: auth -> init (0x200)
I (900349) wifi:new:<13,0>, old:<13,0>, ap:<255,255>, sta:<13,0>, prof:1, snd_ch_cfg:0x0

+JAP:DISCONNECTED,0,2

+JAP:DISCONNECTED,0,205

mhdong, 2024-06-14:
Hi @Leadrive 
如果你们那边有复现的话，请分享一下log，和相关的信息，比如STA的mac，路由器的mac。

closed 2024-07-24 by Sherry616