espressif/esp-idf #13368: how to close log output in runtime

luacerbb, 2024-03-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

how I can to close all log output in runtime, like printf and so on. thanks.

luacerbb, 2024-03-13:
esp32 chip

o-marshmallow, 2024-03-18:
Hello @luacerbb !

At runtime, if you want to close all the logs that come from IDF components, you can use the function [esp_log_level_set](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/log.html#_CPPv417esp_log_level_setPKc15esp_log_level_t).

If you want to get rid of the logs that occur when calling `printf`, you can close the standard output:
```
#include <unistd.h>

void app_main(void) {
    close(STDOUT_FILENO);
}
```

luacerbb, 2024-03-18:
> Hello @luacerbb !
> 
> At runtime, if you want to close all the logs that come from IDF components, you can use the function [esp_log_level_set](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/log.html#_CPPv417esp_log_level_setPKc15esp_log_level_t):
> 
> ```
> #include "esp_log.h"
> 
> void app_main(void)
> {
>     esp_log_level_set("*", ESP_LOG_NONE);
> }
> ```
> 
> If you want to get rid of the logs that occur when calling `printf`, you can close the standard output:
> 
> ```
> #include <unistd.h>
> 
> void app_main(void) {
>     close(STDOUT_FILENO);
> }
> 

if I call close(STDOUT_FILENO) close log print,then how to recover log print thanks

mickeyl, 2024-03-18:
FWIW, note that despite closing the output channels, you will still burn cycles to format and transfer the messages. For production software, I recommend `CONFIG_LOG_DEFAULT_LEVEL_NONE` unless you have any redirects or clever logic in place that writes logs to flash.

luacerbb, 2024-03-18:
@mickeyl Thanks for your reply. In addition to using Espressif's SDK, I also integrate SDKs such as Alibaba, all of which will have multiple log output methods, such as likely printf the standard api. By default, production software doesn't enable logging, but it needs to be turned on when troubleshooting, so I need something like how to turn it off and on

o-marshmallow, 2024-03-20:
@luacerbb Since the functions `dup` and `dup2` are not supported yet, let's take another route to stop the standard output and restore it later: we can modify `stdout` to a read-only stream, which will make all the writes to it to be ignored, and restore its original flags when you want to let it recover its original behavior:
```
    // Backup the standard output flags to restore them later
    int flags = stdout->_flags;
    // Set the standard output to read-only
    stdout->_flags = __SRD;

    /* Lot of printfs... */

    // Outputs on the standard output will work again
    stdout->_flags = flags;
```

luacerbb, 2024-03-21:
@o-marshmallow Thanks, I tried what you provided and it worked

closed 2024-03-21 by luacerbb