espressif/esp-idf #13371: Is there a API that can san the all signal channels and report the channel with the least APs?

anmnmnly, 2024-03-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Please see the title for my issue.

I want to change the signal channel when the current selected channel is traffic due to many APs existed. If ESP-IDF is not provide this API, Is there any other way to achieve this goal.

AtkinEngineering, 2024-03-14:
ESP-IDF provide wifi scanning function. For an example, use [https://github.com/espressif/esp-idf/tree/master/examples/wifi/scan](url)

MaxwellAlan, 2024-03-14:
Hi @anmnmnly 

What's the WiFi mode(STA/AP/STA+AP/Sniffer) your board working on?

- STA Mode
  - Using full channel scan methods, that you can get all channels Router count, but the STA channel is depended on Router's channel, only Router change channel that STA can follow.
- AP only Mode
  - AP Mode can not scan the channel, if you want to do this, please using AP+STA mode or AP+Sniffer mode
- AP+STA Mode
  - using STA scan get all channel info, and then setting the AP channel.
- AP+Sniffer
  - Sniffer mode also can do these, but it not easy to use, which also need some basic knowledge about WiFi frame format.

Thanks.

closed 2024-08-09 by Sherry616