espressif/esp-idf #13373: using espressif/idf docker image on Raspberry PI to build application and flash the device

krupis, 2024-03-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello. I am currently experimenting with CI/CD, Jenkins and Docker. My goal is to automate build/flash process and my ESP32 board flashed automatically.

For this experiment, I have setup a simple Raspberry PI 4 and connected it to ESP32 Development board via USB. I can see that new device appears when I check with the command:
``` ls /dev/tty* ```

I can see that new device appears as **/dev/ttyUSB0**

To simply test serial communication, I have executed the following Python script on Raspberry PI:
```
import serial
serialport = serial.Serial("/dev/ttyUSB0", 115200, timeout=0.5)
while(1):
    response = serialport.readlines(None)
    print(response)
```

and I can see that the messages are popping up, so the Serial port is working as expected.

I have then tried to run the docker following the steps below:

1. CD into my project directory
``` cd /home/lukas/Projects/Embedded-FW-Guidelines```

2. Run the docker image
```
sudo docker run --rm --device /dev/ttyUSB0:/dev/ttyUSB0 -v $PWD:/project -w /project -u $UID -e HOME=/tmp -it espressif/idf 
```

3. Build the project using ```idf.py build```

4. Try to flash the device using ```idf.py flash```

The flashing fails because the docker container does not seem to detect the /tty/USB0 serial device. See the full logs below:

```
lukas@raspberrypi:~/Projects/Embedded-FW-Guidelines $ sudo docker run --rm --device /dev/ttyUSB0:/dev/ttyUSB0 -v $PWD:/project -w /project -u $UID -e HOME=/tmp -it espressif/idf
Detecting the Python interpreter
Checking "python3" ...
Python 3.10.12
"python3" has been detected
Checking Python compatibility
Checking other ESP-IDF version.
Adding ESP-IDF tools to PATH...
Checking if Python packages are up to date...
Requirement files:
 - /opt/esp/idf/tools/requirements/requirements.core.txt
Python being checked: /opt/esp/python_env/idf5.3_py3.10_env/bin/python
Python requirements are satisfied.
Added the following directories to PATH:
  /opt/esp/idf/components/espcoredump
  /opt/esp/idf/components/partition_table
  /opt/esp/idf/components/app_update
  /opt/esp/tools/xtensa-esp-elf-gdb/12.1_20231023/xtensa-esp-elf-gdb/bin
  /opt/esp/tools/riscv32-esp-elf-gdb/12.1_20231023/riscv32-esp-elf-gdb/bin
  /opt/esp/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin
  /opt/esp/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin
  /opt/esp/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin
  /opt/esp/tools/cmake/3.24.0/bin
  /opt/esp/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin
  /opt/esp/tools/qemu-xtensa/esp_develop_8.1.3_20231206/qemu/bin
  /opt/esp/tools/qemu-riscv32/esp_develop_8.1.3_20231206/qemu/bin
  /opt/esp/tools/xtensa-esp-elf-gdb/12.1_20231023/xtensa-esp-elf-gdb/bin
  /opt/esp/tools/riscv32-esp-elf-gdb/12.1_20231023/riscv32-esp-elf-gdb/bin
  /opt/esp/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin
  /opt/esp/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin
  /opt/esp/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin
  /opt/esp/tools/cmake/3.24.0/bin
  /opt/esp/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin
  /opt/esp/tools/qemu-xtensa/esp_develop_8.1.3_20231206/qemu/bin
  /opt/esp/tools/qemu-riscv32/esp_develop_8.1.3_20231206/qemu/bin
  /opt/esp/python_env/idf5.3_py3.10_env/bin
  /opt/esp/idf/tools
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

I have no name!@c6266e394d3c:/project$ idf.py build
Executing action: all (aliases: build)
Running ninja in directory /project/build
Executing "ninja all"...
[1/4] cd /project/build/esp-idf/esptool_py && /opt/esp/python_env/idf5.3_py3.10_env/bin/python /opt/esp/idf/componen...-offset 0x8000 partition --type app /project/build/partition_table/partition-table.bin /project/build/my_project.bi
my_project.bin binary size 0x2b200 bytes. Smallest app partition is 0x100000 bytes. 0xd4e00 bytes (83%) free.
[1/1] cd /project/build/bootloader/esp-idf/esptool_py && /opt/esp/python_env/idf5.3_py3.10_env/bin/python /opt/esp/i...components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 /project/build/bootloader/bootloader.bi
Bootloader binary size 0x6880 bytes. 0x780 bytes (7%) free.

Project build complete. To flash, run:
 idf.py flash
or
 idf.py -p PORT flash
or
 python -m esptool --chip esp32 -b 460800 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_size 2MB --flash_freq 40m 0x1000 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0x10000 build/my_project.bin
or from the "/project/build" directory
 python -m esptool --chip esp32 -b 460800 --before default_reset --after hard_reset write_flash "@flash_args"
I have no name!@c6266e394d3c:/project$ idf.py flash
Executing action: flash
Serial port /dev/ttyUSB0
/dev/ttyUSB0 failed to connect: Could not open /dev/ttyUSB0, the port is busy or doesn't exist.
([Errno 13] could not open port /dev/ttyUSB0: [Errno 13] Permission denied: '/dev/ttyUSB0')

Hint: Try to add user into dialout group: sudo usermod -a -G dialout $USER

No serial ports found. Connect a device, or use '-p PORT' option to set a specific port.
I have no name!@c6266e394d3c:/project$
```

I have followed the espressif instructions and passed the serial port using the device option as described here:
[image]

Perhaps I am not doing it correctly? I would appreciate if someone could help me out :)

fhrbata, 2024-03-13:
Hello @krupis ,

have you tried to follow the hint message in the output.

> Hint: Try to add user into dialout group: sudo usermod -a -G dialout $USER

It seems you are running the docker image as $USER and it may not have access rights to the `/dev/ttyUSB0`.

krupis, 2024-03-13:
> Hello @krupis ,
> 
> have you tried to follow the hint message in the output.
> 
> > Hint: Try to add user into dialout group: sudo usermod -a -G dialout $USER
> 
> It seems you are running the docker image as $USER and it may not have access rights to the `/dev/ttyUSB0`.

I assume $USER should be replaced with my Raspberry PI user ```lukas``` 

I have tried to run the following command on my Raspberry PI:
```
sudo usermod -a -G dialout lukas
```

And then try to run the docker container, build and flash again but the same error persists

fhrbata, 2024-03-13:
> I assume $USER should be replaced with my Raspberry PI user lukas

yes, if `lukas` is the user running the `docker` command

Could you please check who owns the device with e.g. `ls -al /dev/ttyUSB0`. It's possible that the group is not `dialout`, but something else.

krupis, 2024-03-13:
> > I assume $USER should be replaced with my Raspberry PI user lukas
> 
> yes, if `lukas` is the user running the `docker` command
> 
> Could you please check who owns the device with e.g. `ls -al /dev/ttyUSB0`. It's possible that the group is not `dialout`, but something else.

I have run the command that you have suggested and this is the result:
```
lukas@raspberrypi:~/Projects/Embedded-FW-Guidelines $ ls -al /dev/ttyUSB0
crw-rw---- 1 root dialout 188, 0 Mar 13 13:22 /dev/ttyUSB0
```

fhrbata, 2024-03-13:
Have you tried this from a new shell? IIUC adding user to a group will not propagated to the current shell. You can check e.g. with `groups` command, in which groups your user is added. You should see `dialout` in the output.

There is also some docs related to this in the esptool https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/basic-options.html?highlight=dialout#serial-port

> In Linux, the current user may not have access to serial ports and a “Permission Denied” or “Port doesn’t exist” errors may appear. On most Linux distributions, the solution is to add the user to the dialout group (check e.g. ls -l /dev/ttyUSB0 to find the group) with a command like sudo usermod -a -G dialout $USER. You can call su - $USER to enable read and write permissions for the serial port without having to log out and back in again. Check your Linux distribution’s documentation for more information.

fhrbata, 2024-03-13:
Ok, I think I understand now. The group is not propagated into the docker, so the user in docker is not in the `dialout` and hence cannot access the device. When I run the docker with `-u $UID:18` where `18` is group id on my machine, I can flash and access the device from within the container.

erhankur, 2024-03-13:
@krupis you can start docker in privileged mode with '--privileged' option. It should help.

fhrbata, 2024-03-13:
@krupis if you don't want to run the container as privileged you can maybe use something like the following
```
$ sudo docker run --rm --device /dev/ttyUSB0:/dev/ttyUSB0  -u $UID --group-add $(getent group dialout | cut -d':' -f3) -e HOME=/tmp -v $PWD:/project -it espressif/idf
```

It adds the user in container into a `dialout` group from host.

krupis, 2024-03-14:
I have checked which groups my user ```lukas``` is in :
lukas@raspberrypi:~/Projects/Embedded-FW-Guidelines $ groups lukas
lukas : lukas adm dialout cdrom sudo audio video plugdev games users input render netdev spi i2c gpio lpadmin

I have tried to run the command with ```--privileged``` option but I am still not able to flash the device. See the logs below:
```
lukas@raspberrypi:~/Projects/Embedded-FW-Guidelines $ sudo docker run --rm --device /dev/ttyUSB0:/dev/ttyUSB0 -v $PWD:/project -w /project -u $UID -e HOME=/tmp -it --privileged  espressif/idf
Detecting the Python interpreter
Checking "python3" ...
Python 3.10.12
"python3" has been detected
Checking Python compatibility
Checking other ESP-IDF version.
Adding ESP-IDF tools to PATH...
Checking if Python packages are up to date...
Requirement files:
 - /opt/esp/idf/tools/requirements/requirements.core.txt
Python being checked: /opt/esp/python_env/idf5.3_py3.10_env/bin/python
Python requirements are satisfied.
Added the following directories to PATH:
  /opt/esp/idf/components/espcoredump
  /opt/esp/idf/components/partition_table
  /opt/esp/idf/components/app_update
  /opt/esp/tools/xtensa-esp-elf-gdb/12.1_20231023/xtensa-esp-elf-gdb/bin
  /opt/esp/tools/riscv32-esp-elf-gdb/12.1_20231023/riscv32-esp-elf-gdb/bin
  /opt/esp/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin
  /opt/esp/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin
  /opt/esp/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin
  /opt/esp/tools/cmake/3.24.0/bin
  /opt/esp/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin
  /opt/esp/tools/qemu-xtensa/esp_develop_8.1.3_20231206/qemu/bin
  /opt/esp/tools/qemu-riscv32/esp_develop_8.1.3_20231206/qemu/bin
  /opt/esp/tools/xtensa-esp-elf-gdb/12.1_20231023/xtensa-esp-elf-gdb/bin
  /opt/esp/tools/riscv32-esp-elf-gdb/12.1_20231023/riscv32-esp-elf-gdb/bin
  /opt/esp/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin
  /opt/esp/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin
  /opt/esp/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin
  /opt/esp/tools/cmake/3.24.0/bin
  /opt/esp/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin
  /opt/esp/tools/qemu-xtensa/esp_develop_8.1.3_20231206/qemu/bin
  /opt/esp/tools/qemu-riscv32/esp_develop_8.1.3_20231206/qemu/bin
  /opt/esp/python_env/idf5.3_py3.10_env/bin
  /opt/esp/idf/tools
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

I have no name!@a3d73aa44d4d:/project$ ls
CMakeLists.txt  Embedded_workflow.drawio  Jenkinsfile  Readme.md  build  images  main  node_modules  package-lock.json package.json  sdkconfig  sdkconfig.ci  sdkconfig.old  templates
I have no name!@a3d73aa44d4d:/project$ ls dev/tty*
ls: cannot access 'dev/tty*': No such file or directory
I have no name!@a3d73aa44d4d:/project$ idf.py flash -p dev/ttyUSB0
Executing action: flash
Running ninja in directory /project/build
Executing "ninja flash"...
[1/5] cd /project/build/esp-idf/esptool_py && /opt/esp/python_env/idf5.3_py3.10_env/bin/python /opt/...artition --type app /project/build/partition_table/partition-table.bin /project/build/my_project.bin
my_project.bin binary size 0x2b200 bytes. Smallest app partition is 0x100000 bytes. 0xd4e00 bytes (83%) free.
[1/1] cd /project/build/bootloader/esp-idf/esptool_py && /opt/esp/python_env/idf5.3_py3.10_env/bin/p...tion_table/check_sizes.py --offset 0x8000 bootloader 0x1000 /project/build/bootloader/bootloader.bin
Bootloader binary size 0x6880 bytes. 0x780 bytes (7%) free.
[2/3] cd /opt/esp/idf/components/esptool_py && /opt/esp/tools/cmake/3.24.0/bin/cmake -D IDF_PATH=/op...rgs" -D WORKING_DIRECTORY=/project/build -P /opt/esp/idf/components/esptool_py/run_serial_tool.cmake
esptool.py --chip esp32 -p dev/ttyUSB0 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB 0x1000 bootloader/bootloader.bin 0x10000 my_project.bin 0x8000 partition_table/partition-table.bin
esptool.py vv4.8.dev1
Serial port dev/ttyUSB0

A fatal error occurred: Could not open dev/ttyUSB0, the port is busy or doesn't exist.
([Errno 2] could not open port dev/ttyUSB0: [Errno 2] No such file or directory: 'dev/ttyUSB0')

Hint: Check if the port is correct and ESP connected

[... 101 lines cut ...]
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00007fff...
Flash will be erased from 0x00010000 to 0x0003bfff...
Flash will be erased from 0x00008000 to 0x00008fff...
Compressed 26752 bytes to 16373...
Writing at 0x00001000... (100 %)
Wrote 26752 bytes (16373 compressed) at 0x00001000 in 0.8 seconds (effective 268.1 kbit/s)...
Hash of data verified.
Compressed 176640 bytes to 94141...
Writing at 0x000363d0... (100 %)
Wrote 176640 bytes (94141 compressed) at 0x00010000 in 2.7 seconds (effective 526.0 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 103...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.1 seconds (effective 427.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Done
```

2. When I run the espressif/esp-idf docker image from my repository. I was expecting to require to build the project everytime the docker is run as it should create a new instance ? But that is not the case, I have built the project just once and I do not need to build it anymore. Is that expected behaviour?

The next step is to perform the same actions (run the docker image, build the project and flash the device) via the Gitlab CI/CD pipeline. I will update on how that goes.

krupis, 2024-03-14:
**UPDATE**

I have setup gitlab-runner with docker executor on my Raspberry PI and trying to run the CI/CD pipeline as shown below:
```

# List of stages for jobs, and their order of execution
stages:
  - test

flash_job:
  image: espressif/idf:latest
  variables:
    GIT_SUBMODULE_STRATEGY: normal
  stage: test
  tags:
    - raspberrypi
  before_script:
    - idf.py build
  script:
    - idf.py -p /dev/ttyUSB0 flash
```
It has failed as expected and the output is as following:
```
Running with gitlab-runner 16.9.1 (782c6ecb)
  on Raspberry PI HAL Docker SUDO test xWAkiP4wq, system ID: s_0ae6b[2](https://ems-gitlab.teltonika.lt/embeded-programmers/Embedded-FW-Guidelines/-/jobs/791#L2)582ae5
Preparing the "docker" executor
00:11
Using Docker executor with image espressif/idf:latest ...
Pulling docker image espressif/idf:latest ...
Using docker image sha256:7460bcd444cc5fc9c7b6504b7c8f862a5[3](https://ems-gitlab.teltonika.lt/embeded-programmers/Embedded-FW-Guidelines/-/jobs/791#L3)556fe2f2[4](https://ems-gitlab.teltonika.lt/embeded-programmers/Embedded-FW-Guidelines/-/jobs/791#L4)ac07bb01c[5](https://ems-gitlab.teltonika.lt/embeded-programmers/Embedded-FW-Guidelines/-/jobs/791#L5)f1501a1[6](https://ems-gitlab.teltonika.lt/embeded-programmers/Embedded-FW-Guidelines/-/jobs/791#L6)8a6 for espressif/idf:latest with digest espressif/idf@sha256:fd20cf143fe2afa5dd4c5fd8d8df823cccfafe48fc165762541e0df96c12935b ...
Preparing environment
00:01
Running on runner-xwakip4wq-project-121-concurrent-0 via raspberrypi...
Getting source from Git repository
00:03
Fetching changes with git depth set to 20...
Initialized empty Git repository in /builds/embeded-programmers/Embedded-FW-Guidelines/.git/
Created fresh repository.
Checking out f2fc6e19 as detached HEAD (ref is main)...
Updating/initializing submodules with git depth set to 20...
Updated submodules
Executing "step_script" stage of the job script
04:28
Using docker image sha256:7460bcd444cc5fc9c7b6504b7c8f862a53556fe2f24ac07bb01c5f1501a168a6 for espressif/idf:latest with digest espressif/idf@sha256:fd20cf143fe2afa5dd4c5fd8d8df823cccfafe48fc165762541e0df96c12935b ...
Detecting the Python interpreter
Checking "python3" ...
Python 3.10.12
"python3" has been detected
Checking Python compatibility
Checking other ESP-IDF version.
Adding ESP-IDF tools to PATH...
Checking if Python packages are up to date...
Python requirements are satisfied.
Requirement files:
 - /opt/esp/idf/tools/requirements/requirements.core.txt
Python being checked: /opt/esp/python_env/idf5.3_py3.10_env/bin/python
Added the following directories to PATH:
  /opt/esp/idf/components/espcoredump
  /opt/esp/idf/components/partition_table
  /opt/esp/idf/components/app_update
  /opt/esp/tools/xtensa-esp-elf-gdb/12.1_20231023/xtensa-esp-elf-gdb/bin
  /opt/esp/tools/riscv32-esp-elf-gdb/12.1_20231023/riscv32-esp-elf-gdb/bin
  /opt/esp/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin
  /opt/esp/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin
  /opt/esp/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin
  /opt/esp/tools/cmake/3.24.0/bin
  /opt/esp/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin
  /opt/esp/tools/qemu-xtensa/esp_develop_8.1.3_20231206/qemu/bin
  /opt/esp/tools/qemu-riscv32/esp_develop_8.1.3_20231206/qemu/bin
  /opt/esp/tools/xtensa-esp-elf-gdb/12.1_20231023/xtensa-esp-elf-gdb/bin
[... 1268 lines cut ...]
[session_server]
  session_timeout = 1800

[[runners]]
  name = "Raspberry PI HAL Docker SUDO test"
  url = "https://xxx-test.test.lt/"
  id = 19
  token = "glrt-xWAkiP4wqx7GUyEGjn-o"
  token_obtained_at = 2024-03-14T06:03:24Z
  token_expires_at = 0001-01-01T00:00:00Z
  executor = "docker"
  [runners.cache]
    MaxUploadedArchiveSize = 0
  [runners.docker]
    tls_verify = false
    image = "espressif/esp-idf"
    privileged = false
    disable_entrypoint_overwrite = false
    oom_kill_disable = false
    disable_cache = false
    volumes = ["/cache"]
    shm_size = 0
    network_mtu = 0
```
Perhaps you could clarify how to modify this as I am not sure how to add ```--group-add $(getent group dialout | cut -d':' -f3)``` option to this.

fhrbata, 2024-03-14:
Hello @krupis ,

> I have checked which groups my user `lukas` is in : lukas@raspberrypi:~/Projects/Embedded-FW-Guidelines $ groups lukas lukas : lukas adm dialout cdrom sudo audio video plugdev games users input render netdev spi i2c gpio lpadmin
> 
> I have tried to run the command with `--privileged` option but I am still not able to flash the device. See the logs below:
> 
> ```
> lukas@raspberrypi:~/Projects/Embedded-FW-Guidelines $ sudo docker run --rm --device /dev/ttyUSB0:/dev/ttyUSB0 -v $PWD:/project -w /project -u $UID -e HOME=/tmp -it --privileged  espressif/idf
> ...
> I have no name!@a3d73aa44d4d:/project$ idf.py flash -p dev/ttyUSB0
> Serial port dev/ttyUSB0
>  ...

I believe you made a typo `dev/ttyUSB0` in the device path, because it's missing the leading `/`, but I guess that you probably noticed it.

> ...
> I have no name!@a3d73aa44d4d:/project$ idf.py flash
> Executing action: flash
> Serial port /dev/ttyUSB0
> /dev/ttyUSB0 failed to connect: Could not open /dev/ttyUSB0, the port is busy or doesn't exist.
> ([Errno 13] could not open port /dev/ttyUSB0: [Errno 13] Permission denied: '/dev/ttyUSB0')

IMHO the problem is that you are starting the container as privileged, but still setting the user to `lukas`(`-u $UID`), so the user within the container still doesn't have access to the device. Also IIUC the `--device` is not needed, because `--privileged` automatically makes all devices available from within the container.

You can try the following, which shoudl work.
```
sudo docker run --rm  -v $PWD:/project -w /project  -e HOME=/tmp -it --privileged  espressif/idf
```

But note that all the files generated during build will be owned by root in the host directory, but maybe it's not a problem for you.

> ...
> I have also tried what @fhrbata suggested `--group-add $(getent group dialout | cut -d':' -f3)` **and I have managed to sucessfully flash the device!**
>
> Although there are a couple of things [...]

Probably because the `-w /project` option is missing from to command. It's possible I forgot/removed it.

> ...

> 2. When I run the espressif/esp-idf docker image from my repository. I was expecting to require to build the project everytime the docker is run as it should create a new instance ? But that is not [...]

The project is build within the host directory, which is mounted into the docker container(`-v $PWD:/project`). So even when the container is destroyed, the project is not affected by this. So yes, this is expected and how it works 
https://docs.docker.com/storage/bind-mounts/
> 
> 
> The next step is to perform the same actions (run the docker image, build the project and flash the device) via the Gitlab CI/CD pipeline. I will update on how that goes.

fhrbata, 2024-03-14:
> **UPDATE**
>
> I have setup gitlab-runner with docker executor on my Raspberry PI and trying to run the CI/CD pipeline as shown below:
>
> ```
>
> # List of stages for jobs, and their order of execution [...]

> ...

> My current `config.toml` is as following:
>
> ```
> lukas@raspberrypi:~/Projects/Embedded-FW-Guidelines $ sudo cat /etc/gitlab-runner/config.toml
> concurrent = 1
> check_interval = 0
> connection_max_age = [...]

TBH I'm not that familiar with `gitlab-runner` settings, but you can try to run it in privileged mode `privileged = true` as described here https://docs.gitlab.com/runner/executors/docker.html#privileged-mode. Or you can try to set the `user` and `group_add` as described in the docs here https://docs.gitlab.com/runner/configuration/advanced-configuration.html#the-runnersdocker-section. But I don't know if it's possible to set this dynamically based on some output command. But maybe in your case it would be enough to just add the `group_add` statically based on the `getent group dialout | cut -d':' -f3`. Meaning something like the following in my case
```
[runners.docker]
...
   group_add = 18
```
Note I haven't tried this at all, just a quick look at the docs and my guess. Anyway, I believe this is not directly related to the esp-idf docker image and it's more about your ci configuration. I'm not sure if I'm able to help with this more.

HTH

krupis, 2024-03-14:
> TBH I'm not that familiar with `gitlab-runner` settings, but you can try to run it in privileged mode `privileged = true`

Yeah that was it :) I have posted my reply above few seconds before you did. Thanks for all the help

fhrbata, 2024-03-14:
> > TBH I'm not that familiar with `gitlab-runner` settings, but you can try to run it in privileged mode `privileged = true`
> 
> Yeah that was it :) I have posted my reply above few seconds before you did. Thanks for all the help

You're welcome, of course. I'll keep this discussion open for now. I'll discuss with other members of our team to see if we need to enhance our documentation regarding the scenario where the container is initiated under a specific non-root user, as recommended in our documentation. In this case, flashing may not be possible because the container user might not be added to the necessary 'dialout' group.

Thank you!

closed 2024-03-22 by espressif-bot