espressif/esp-idf #13378: v5.0.x: CONFIG_UART_ISR_IN_IRAM incompatible with CONFIG_RINGBUF_PLACE_FUNCTIONS_INTO_FLASH

jimmo, 2024-03-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm from the MicroPython project, investigating an issue that was raised here https://github.com/micropython/micropython/issues/14005#issuecomment-1996507230

The summary is:
* MicroPython enables `CONFIG_UART_ISR_IN_IRAM` and `CONFIG_RINGBUF_PLACE_FUNCTIONS_INTO_FLASH`
* This is an allowed combination according to the Kconfig. Clearly `RINGBUF_PLACE_ISR_FUNCTIONS_INTO_FLASH` would not be allowed, as the UART ISRs require this (and the Kconfig enforces this).
* However, `uart_rx_intr_handler_default` (in IRAM) calls `xRingbufferSendFromISR` (in IRAM) whcih calls `prvGetFreeSize` (*not* in IRAM). (i.e. `prvGetFreeSize` is considered to be a "ringbuf function" but not a "ringbuf ISR function").
* This leads to a crash `Cache disabled but cached memory region accessed` if the UART ISR fires while a flash operation is in progress.

You can see the decoded stack trace here: https://github.com/micropython/micropython/issues/14005#issuecomment-1982399075

This code changed in v5.1.x, see 443d87c2ff0939860e94e425ba8394c283159adf, but reporting this to you because 5.0.x is still a supported release. I haven't verified whether the 5.1.x fixes the issue though (not sure if it calls a different non-IRAM function now), but are planning to move to 5.2 soon so will report back.

We don't have an easy workaround because we don't have enough IRAM available to disable `CONFIG_RINGBUF_PLACE_FUNCTIONS_INTO_FLASH`.

Thanks, please let me know if I can provide any more information.

o-marshmallow, 2024-03-18:
Hello @jimmo ,

Thank you for reporting this issue, in newer versions, 5.1 and above, the function `xRingbufferSendFromISR` doesn't use `prvGetFreeSize` anymore as the ringbuffer internals have been modified. We are going to investigate what the proper solution is for the version v5.0 of IDF.

AxelLin, 2024-05-06:
@o-marshmallow 
How is the status of this issue?
The fix 4a22a5b only available in v5.0 branch, just wondering if v5.1+ need fix or not?

o-marshmallow, 2024-05-24:
Hello @AxelLin,

This is now merged and closed. As I said above, in IDF v5.1 and above, the function `xRingbufferSendFromISR` doesn't use `prvGetFreeSize` anymore as the ringbuffer internals have been modified.

closed 2024-05-24 by o-marshmallow