espressif/esp-idf #13379: After updating to esp v5.2.1 the nrfMesh app is not working for the nodes. It Stucks on Composition data Get.

Builtfor-developers, 2024-03-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I updated my esp idf to version 5.2.1 as because it has the features like remote provisioning. Before updating I was using the nrfMesh app to provision my device. But after the update, I first ran into the problem that the esp image which was created was large and memory overflow error but I resolved it by going to menuconfig -> partition -> and single factory app (large).     But next after it got solved I flashed my firmware to the esp32 and started to provision it by nrfMesh app but it got stuck on the sending composition data get step. I tried the newer version of examples too but got the same error. But when I try to provision by the other esp32 device it gets provisioned without any error. So any help regarding this issue will be appreciated. Thank in advance.

forx157, 2024-03-15:
@Builtfor-developers 
This is a known issue and we will fix this in the future, for the time being you can refer to [https://github.com/espressif/esp-idf/issues/12871#issuecomment-1991478141](url) for a temporary solution.

Builtfor-developers, 2024-03-15:
> @Builtfor-developers This is a known issue and we will fix this in the future, for the time being you can refer to [https://github.com/espressif/esp-idf/issues/12871#issuecomment-1991478141](url) for a temporary solution.

Oh that's Sad. New release will take time so, I was wondering that if by any chance you know what the problem really is so that I can maybe work on it. Any specific point perhaps that can give me some idea about the main cause of problem. Thanks for quick reply.

Builtfor-developers, 2024-03-15:
Never Mind. Just solved it. Thanks to this one  https://github.com/espressif/esp-idf/issues/12871#issuecomment-1991478141
    if (IS_ENABLED(CONFIG_BLE_MESH_GATT_PROXY_SERVER) &&
       bt_mesh_private_gatt_proxy_state_get() != BLE_MESH_PRIVATE_GATT_PROXY_ENABLED &&
       net_if == BLE_MESH_NET_IF_PROXY) {

it works now. Thank You.

closed 2024-03-15 by Builtfor-developers