espressif/esp-idf #13380: After an update of Partition Table ESP keeps restarting.

meakashrao, 2024-03-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1

### Espressif SoC revision.

ESP32-WROOM-32E-N8 (revision v3.0 and V3.1)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom hardware based on Espressif boards of made with reference to above chips

### Power Supply used.

External 3.3V

### What is the expected behavior?

The device to start correctly from ota0 partition

### What is the actual behavior?

The Device is in a loop of SW_Reset

[image]
This image basically tells there was a 4MB Flash partition Table written and it failed the check for 8MB

[image]
This is what we get for Silicon Version 3.1 chips

For chips of Silicon Version 3.0 it reboots from ota0 partition and the correct firmware is implemented

### Steps to reproduce.

For Silicon Revision 3.0 everything works fine

For Silicon Revision 3.1 we see the reset after repartition step is executed(SPI Erase and Write of the protected segment returns true)

### More Information.

<html>
<body>
<li>
<p>4MB and 8MB Flash Partition Table</p>
<p>4MB</p>

 Name | Type | SubType | Offset | Size
nvs | data | nvs | 0x9000 | 0x5000
otadata | data | ota | 0xe000 | 0x2000
app0 | app | ota_0 | 0x10000 | 0x1E0000
app1 | app | ota_1 | 0x1F0000 | 0x1E0000
spiffs | data | spiffs | 0x3D0000 | 0x30000

<p>8MB</p>
<html>
<body>

Name | Type | SubType | Offset | Size

nvs | data | nvs | 0x9000 | 0x5000
otadata | data | ota | 0xe000 | 0x2000
app0 | app | ota_0 | 0x10000 | 0x330000
app1 | app | ota_1 | 0x340000 | 0x330000
spiffs | data | spiffs | 0x670000 | 0x190000

</body>
</html>
</li>
</ul>
<p>As can be seen from above Table, the Table till app0 offset address is same for both 4MB and 8MB and it is a blessing in disguise, why?</p>
<p>Because when we update from 4MB to 8MB, the ota1 partition which was at offset address 0x1F0000 is now suddenly a part of ota0, so the firmware after restarting has to jump and read stuff from ota0 as ota1 location won’t contain any firmware or would be empty as the space was initially in SPIFFS and and we didn’t use spiffs for anything.</p>

</body>
</html>

The basic Steps are
a) Check the running Partition to make sure that ota_0 of 4MB has the correct Firmware (Latest)

b) If running from ota1(i.e ota_0 has the correct firmware) then update the partition table size

c) The update includes and SPI Erase and SPI Write and the restart. (I thought NVS was necessary, still might be WIFI bug explained below, but I removed NVS, just serial prints are there)

hmalpani, 2024-03-15:
Hello @meakashrao 
Thanks for reporting the issue. Can you please share an application which can be used to reproduce this issue? Also the images you shared are not visible. Can you re-share them?

meakashrao, 2024-03-15:
Hey sorry for the Images

This image basically tells there was a 4MB Flash partition Table written and it failed the check for 8MB, and then when Value R is sent from Bluetooth(NVS Erase and write, after SPI Write was later removed but I forgot to remove the comments)
[image]
 

**This is what we get for Silicon Version 3.1 chips**
[image]

**For chips of Silicon Version 3.0 it reboots from ota0 partition after changing the repartition and the correct firmware is implemented(which is in ota0 partition)**

```c
#include "Arduino.h"
#include "spi_flash_mmap.h"
#include "esp_flash.h"
#include "esp_partition.h"
#include "esp_ota_ops.h"
#include "esp_err.h"
#include "esp_log.h"
#include "Bluetooth.h"
#include "Repartition.h"

static const char *PT_TAG = "Repart";

char partitionMessages[MAX_PARTITIONS][PARTITION_MSG_LENGTH]; // Array to hold partition messages
int partitionStart = 0;

// The flash chip pointer (NULL to use the default flash chip)
esp_flash_t *flash_chip = NULL;

typedef struct
{
    esp_partition_type_t type; // The partition type
    uint32_t address;          // The starting address of the partition
    uint32_t size;             // The size of the partition
} partition_info_t;

// Define an array to hold the partition information
static partition_info_t partition_info_array[MAX_PARTITIONS];
int partition_count = 0; // To keep track of the number of partitions saved

//This is the table for 8MB Flash with 2 OTA partitions and no coredump
//Size of ota_0 and OTA_1 is 0X330000=3.34MB, SPIFFS=1.64MB
//the advantage here he is nvs, otadata is same as min_spiffs.csv (which was used earlier)
//the offset of ota_0 is same as min_spiffs.csv at 0x10000
static const char twoota_spiffs_no_coredump_8mb[] = {
    0xaa, 0x50, 0x01, 0x02, 0x00, 0x90, 0x00, 0x00, 0x00, 0x50, 0x00, 0x00, 0x6e, 0x76, 0x73, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
	0xaa, 0x50, 0x01, 0x00, 0x00, 0xe0, 0x00, 0x00, 0x00, 0x20, 0x00, 0x00, 0x6f, 0x74, 0x61, 0x64, 0x61, 0x74, 0x61, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
	0xaa, 0x50, 0x00, 0x10, 0x00, 0x00, 0x01, 0x00, 0x00, 0x00, 0x33, 0x00, 0x61, 0x70, 0x70, 0x30, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
	0xaa, 0x50, 0x00, 0x11, 0x00, 0x00, 0x34, 0x00, 0x00, 0x00, 0x33, 0x00, 0x61, 0x70, 0x70, 0x31, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
	0xaa, 0x50, 0x01, 0x82, 0x00, 0x00, 0x67, 0x00, 0x00, 0x00, 0x19, 0x00, 0x73, 0x70, 0x69, 0x66, 0x66, 0x73, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
	0xeb, 0xeb, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x00, 0x46, 0x5c, 0x1b, 0xea, 0xb8, 0x90, 0x09, 0x70, 0x64, 0x10, 0xb5, 0x75, 0x12, 0xab, 0x92,
    0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,
    0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,
    0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,
    0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,
    0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,
};

void check_Partition();
bool write_partition_table(const char *new_table);
void alter_default_partitions();
void check_upgraded_partitions();
void repartition_device();
void check8MBPartition();

void check_Partition()
{
    esp_partition_iterator_t mypartiterator;
    const esp_partition_t *mypart;
[... 256 lines cut ...]
    {
        addErrorMessage("-otadata check OK--");
        Serial.println("OTA Data partition check passed");
    }
    else
    {
        addErrorMessage("otadata check NOTOK");
        Serial.println("OTA Data partition check failed");
    }

    if(nvs_data && nvs_data->address == 0x9000 && nvs_data->size == 0x5000)
    {
        addErrorMessage("---nvs check OK----");
        Serial.println("NVS partition check passed");
    }
    else
    {
        addErrorMessage("-nvs check NOT-OK--");
        Serial.println("NVS partition check failed");
    }
}
``` 

void repartition_device()  is called when 'R' is sent from Bluetooth
void check8MBPartition() is called when 'C' is sent from Bluetooth

hmalpani, 2024-03-15:
@meakashrao 
Can you also share full logs?

meakashrao, 2024-03-15:
Logs for Silicon Revision 3.0

Microsoft Windows [Version 10.0.22631.3296]
(c) Microsoft Corporation. All rights reserved.

D:\UP\GitHubVSArea\ESP_IDF\ESP32FW_OTA_Repartition\ESP32FW-OTA-Repartition\ESP32FW-OTA-REPARTITION>set IDF_PATH=C:\Users\Akash\esp\v5.1\esp-idf

D:\UP\GitHubVSArea\ESP_IDF\ESP32FW_OTA_Repartition\ESP32FW-OTA-Repartition\ESP32FW-OTA-REPARTITION>c:\Users\Akash\esp\Tools_Directory\python_env\idf5.1_py3.11_env\Scripts\python.exe C:\Users\Akash\esp\v5.1\esp-idf\tools\idf_monitor.py -p COM3 -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32 d:\UP\GitHubVSArea\ESP_IDF\ESP32FW_OTA_Repartition\ESP32FW-OTA-Repartition\ESP32FW-OTA-REPARTITION\build\ESP32FW-OTA-REPARTITION.elf
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM3 instead...
--- esp-idf-monitor 1.3.4 on \\.\COM3 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jul 29 2019 �ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1240
load:0x40078000,len:13012
load:0x40080400,len:3648
0x40080400: _init at ??:?

entry 0x400805f8
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1240
load:0x40078000,len:13012
load:0x40080400,len:3648
0x40080400: _init at ??:?

entry 0x400805f8
I (470) cpu_start: Multicore app
I (470) cpu_start: Pro cpu up.
I (470) cpu_start: Starting app cpu, entry point is 0x40081364
0x40081364: call_start_cpu1 at C:/Users/Akash/esp/v5.1/esp-idf/components/esp_system/port/cpu_start.c:159

I (0) cpu_start: App cpu up.
I (488) cpu_start: Pro cpu start user code
I (488) cpu_start: cpu freq: 160000000 Hz
I (488) cpu_start: Application information:
I (493) cpu_start: Project name:     ESP32FW-OTA-REPARTITION
I (499) cpu_start: App version:      1
I (504) cpu_start: Compile time:     Feb 16 2024 11:15:17
I (510) cpu_start: ELF file SHA256:  813b5db61818843a...
I (516) cpu_start: ESP-IDF:          v5.1.2-666-ga5e3e0b90e
I (522) cpu_start: Min chip rev:     v0.0
I (527) cpu_start: Max chip rev:     v3.99
I (532) cpu_start: Chip rev:         v3.0
I (536) heap_init: Initializing. RAM available for dynamic allocation:
I (544) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (550) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (556) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (562) heap_init: At 3FFCD2D0 len 00012D30 (75 KiB): DRAM
I (568) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (574) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (581) heap_init: At 4009ECD0 len 00001330 (4 KiB): IRAM
I (588) spi_flash: detected chip: generic
I (592) spi_flash: flash io: dio
I (597) coexist: coex firmware version: 77cd7f8
I (601) app_start: Starting scheduler on CPU0
         (606) app_start: Starting scheduler on CPU1
32mI (606) main_task: Started on CPU0
I (615) main_task: Calling app_main()
I (643) maRed Pin: 15
Green Pin: 17
[... 60 lines cut ...]
I (31310) Repart: size: 1E0000
I (31313) Repart:
I (31316) Repart: ota_0 and ota_1, not NULL
I (31321) Repart: Default Partition
W (31326) repartition: Running from OTA1, modify partition table
I (31332) Repart: SPI Erase Starts
I (31395) Repart: Flash Erase OK
I (31395) Repart: SPI Erase Ends, Write Starts
I (31425) Repart: Flash Write OK
I (31425) Repart: SPI Write Ends, NVS Starts
I (31425) Repart: NVS Ends, Rebooting
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1240
load:0x40078000,len:13012
load:0x40080400,len:3648
0x40080400: _init at ??:?

entry 0x400805f8
E (530) esp_core_dump_flash: N♠→�ɕ�dump partition found!
E (531) esp_core_dump_flash: No core dump partition found!

meakashrao, 2024-03-15:
### _**Logs for Silicon Revision 3.0**_
```
Microsoft Windows [Version 10.0.22631.3296]
(c) Microsoft Corporation. All rights reserved.

D:\UP\GitHubVSArea\ESP_IDF\ESP32FW_OTA_Repartition\ESP32FW-OTA-Repartition\ESP32FW-OTA-REPARTITION>set IDF_PATH=C:\Users\Akash\esp\v5.1\esp-idf

D:\UP\GitHubVSArea\ESP_IDF\ESP32FW_OTA_Repartition\ESP32FW-OTA-Repartition\ESP32FW-OTA-REPARTITION>c:\Users\Akash\esp\Tools_Directory\python_env\idf5.1_py3.11_env\Scripts\python.exe C:\Users\Akash\esp\v5.1\esp-idf\tools\idf_monitor.py -p COM3 -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32 d:\UP\GitHubVSArea\ESP_IDF\ESP32FW_OTA_Repartition\ESP32FW-OTA-Repartition\ESP32FW-OTA-REPARTITION\build\ESP32FW-OTA-REPARTITION.elf
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM3 instead...
--- esp-idf-monitor 1.3.4 on \\.\COM3 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jul 29 2019 �ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1240
load:0x40078000,len:13012
load:0x40080400,len:3648
0x40080400: _init at ??:?

entry 0x400805f8
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1240
load:0x40078000,len:13012
load:0x40080400,len:3648
0x40080400: _init at ??:?

entry 0x400805f8
I (470) cpu_start: Multicore app
I (470) cpu_start: Pro cpu up.
I (470) cpu_start: Starting app cpu, entry point is 0x40081364
0x40081364: call_start_cpu1 at C:/Users/Akash/esp/v5.1/esp-idf/components/esp_system/port/cpu_start.c:159

I (0) cpu_start: App cpu up.
I (488) cpu_start: Pro cpu start user code
I (488) cpu_start: cpu freq: 160000000 Hz
I (488) cpu_start: Application information:
I (493) cpu_start: Project name:     ESP32FW-OTA-REPARTITION
I (499) cpu_start: App version:      1
I (504) cpu_start: Compile time:     Feb 16 2024 11:15:17
I (510) cpu_start: ELF file SHA256:  813b5db61818843a...
I (516) cpu_start: ESP-IDF:          v5.1.2-666-ga5e3e0b90e
I (522) cpu_start: Min chip rev:     v0.0
I (527) cpu_start: Max chip rev:     v3.99
I (532) cpu_start: Chip rev:         v3.0
I (536) heap_init: Initializing. RAM available for dynamic allocation:
I (544) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (550) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (556) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (562) heap_init: At 3FFCD2D0 len 00012D30 (75 KiB): DRAM
I (568) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (574) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (581) heap_init: At 4009ECD0 len 00001330 (4 KiB): IRAM
I (588) spi_flash: detected chip: generic
I (592) spi_flash: flash io: dio
I (597) coexist: coex firmware version: 77cd7f8
I (601) app_start: Starting scheduler on CPU0
         (606) app_start: Starting scheduler on CPU1
32mI (606) main_task: Started on CPU0
I (615) main_task: Calling app_main()
I (643) maRed Pin: 15
Green Pin: 17
[... 659 lines cut ...]
rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1240
load:0x40078000,len:13012
load:0x40080400,len:3648
0x40080400: _init at ??:?

entry 0x400805f8
ets Jul 29 2019 12:21:46

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1240
load:0x40078000,len:13012
load:0x40080400,len:3648
0x40080400: _init at ??:?

entry 0x400805f8
None
Waiting for the device to reconnectets Jul 29 2019 12:21:46
```

meakashrao, 2024-03-19:
Hello. 

I recently verified that in Silicon Revision 3.1 as well the partition table is updated as indicated by ESP Erase and Write success. I read the whole flash and checked 0x8000 address before and after sending the command

Before Partition Update
[image]

After Partition Update
[image]

Please help me figure out why is it restarting.

hmalpani, 2024-03-20:
Hello @meakashrao 
One thing I found wrong is that you have set flash size in sdkconfig as 4MB but the new ota table is configured for flash size 8MB. Can you update the flash size in menuconfig and let me know if that makes any difference?

hmalpani, 2024-03-26:
Hello @meakashrao 
Can you try enabling the verbose logs in bootloader and send the log output when the crash occurs. Also it will be helpful if you could share the flash contents by reading the flash using `esptool.py read_flash`. It will be useful in reproducing the issue.
Thanks!

meakashrao, 2024-03-26:
Hello,

Since the initial development was done on Arduino the bootloader logs options isn't available. 
Also, the ESP version used in Arduino was v2.0.9 (I tried to check for bootloader version incompatibility but failed to do so

As far as reading the flash is concerned. I did that already and after the partition update command 8MB of Flash was updated  and then Silicon Version 3.1 is in a loop and not starting the application. 

[image]
[image]
)

I have the bootloader binary used in our applications
[Bootloader.zip](https://github.com/espressif/esp-idf/files/14755521/Bootloader.zip)

mahavirj, 2024-04-01:
@meakashrao 

Would it be possible for you to build the bootloader from ESP-IDF with verbose logging? I think it is difficult to comment on the issue with the current information you supplied. Once we have logs from the bootloader, it should be clear as why the control is not getting passed to the application image.

meakashrao, 2024-04-01:
OK so two points

a) The machine that are already out, have a bootloader that was compiled on Arduino IDE, so in any way will the bootloader.bin file help?
b) I tried doing what you asked(although the bootloader version is in question here) but building the bootloader with verbose logging asks for extra space(it asks for the partition table to be shifted from 0x8000) 

So will the verbose logging still be useful after all these?

meakashrao, 2024-04-01:
Why I am asking is, will different bootloader won't create an issue

My view is a new bootloader with ESP IDF might work but the problem is same bootloader.bin generated and compiled with Arduino works for Silicon Revision Chips 3.0 and won't work for Silicon Revision Chips 3.1

mahavirj, 2024-04-12:
> My view is a new bootloader with ESP IDF might work but the problem is same bootloader.bin generated and compiled with Arduino works for Silicon Revision Chips 3.0 and won't work for Silicon Revision Chips 3.1

For this case, I suggest that you create a new issue tracker in https://github.com/espressif/arduino-esp32 project

meakashrao, 2024-04-12:
Sure. Thank you

closed 2024-04-12 by meakashrao