espressif/esp-idf #13381: usb hid device barcode scanner can not use

yel-best, 2024-03-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

* usb_host_lib 
After reading the barcode gun, 57 hungry key_code appears directly, and it enters a dead loop

[image]

I'm using this example
https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host/hid

Please tell me some ways or reasons,thanks

Through another project, **usb_host_lib**, the parameters of the scanning gun I am currently using are read as follows:

```
I (13615) CLASS: Opening device at address 1
I (13615) CLASS: Getting device information
I (13615) CLASS:        Full speed
I (13615) CLASS:        bConfigurationValue 1
I (13615) CLASS: Getting device descriptor
*** Device descriptor ***
bLength 18
bDescriptorType 1
bcdUSB 1.10
bDeviceClass 0x0
bDeviceSubClass 0x0
bDeviceProtocol 0x0
bMaxPacketSize0 64
idVendor 0x218
idProduct 0x210
bcdDevice 0.00
iManufacturer 1
iProduct 2
iSerialNumber 3
bNumConfigurations 1
I (13645) CLASS: Getting config descriptor
*** Configuration descriptor ***
bLength 9
bDescriptorType 2
wTotalLength 34
bNumInterfaces 1
bConfigurationValue 1
iConfiguration 0
bmAttributes 0x80
bMaxPower 400mA
        *** Interface descriptor ***
        bLength 9
        bDescriptorType 4
        bInterfaceNumber 0
        bAlternateSetting 0
        bNumEndpoints 1
        bInterfaceClass 0x3
        bInterfaceSubClass 0x1
        bInterfaceProtocol 0x1
        iInterface 0
                *** Endpoint descriptor ***
                bLength 7
                bDescriptorType 5
                bEndpointAddress 0x83   EP 3 IN
                bmAttributes 0x3        INT
                wMaxPacketSize 8
                bInterval 1
I (13695) CLASS: Getting Manufacturer string descriptor
USBScn Chip
I (13705) CLASS: Getting Product string descriptor
USBScn Module$
I (13715) CLASS: Getting Serial Number string descriptor
2027300413413333

```

peter-marcisovsky, 2024-03-14:
Closing the issue due to the duplicity. 
The same issue has been opened and is being solved in the `espressif/esp-usb` 
https://github.com/espressif/esp-usb/issues/26

closed 2024-03-14 by peter-marcisovsky

yel-best, 2024-03-15:
> Closing the issue due to the duplicity. The same issue has been opened and is being solved in the `espressif/esp-usb` [espressif/esp-usb#26](https://github.com/espressif/esp-usb/issues/26)

Thank you for your reply. Yes, you are right. I submitted this question repeatedly because I was really troubled. I hope you can understand. Thanks again.

Oh, I didn't modify the logic code, I just added some esp_log, and I printed out every function execution point that the program ran. Then I found that the content of my **QR code was `"a" `but the result of scanning the code was keyCode = 57**, and in the function:
```
hid_host_interface_callback
hid_host_keyboard_report_callback
hid_host_interface_callback
hid_host_keyboard_report_callback
key_event_callback 57
```
The loop jump does not output the result character `"a" `I want.

I don’t know the reason, but the result I understand from the code scanner is actually characters such as abcdefg in the keyboard. Do I need to modify anything? I think the driver and character set in hid_usage_keyboard can be applied to the code scanner.