espressif/esp-idf #13382: Rpc(ClientError(Transport(Failed to load system certs: No valid certificate found)))-[xtensa-esp32s3-espidf]

blue-freedom-technologies, 2024-03-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2

### Espressif SoC revision.

ESP32-S3 (revision: v0.1)

### Operating System used.

Linux

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-DevKitC-1-N32R8V

### Power Supply used.

USB

### What is the expected behavior?

Connect to a RPC node (A RPC Node is a server in a network that can process RPC requests from clients) using my ESP32-S3 board.

### What is the actual behavior?

```text
I (11148) mcu_app: Connection error: Rpc(ClientError(Transport(Failed to load system certs: No valid certificate found)))
```

### Steps to reproduce.

1. cargo run --bin mcu-app --release
I (11148) mcu_app: Connection error: Rpc(ClientError(Transport(Failed to load system certs: No valid certificate found)))
...

### Debug Logs.

```plain
I (9908) mcu_app: Wifi connected...!!!
I (9918) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (9918) wifi:<ba-add>idx:0 (ifx:0, fc:34:97:8a:61:c0), tid:6, ssn:1, winSize:64
I (10908) esp_netif_handlers: sta ip: 192.168.50.97, mask: 255.255.255.0, gw: 192.168.50.1
I (10908) mcu_app: Wifi netif up...!!!
I (10908) mcu_app: Wifi DHCP info: IpInfo { ip: 192.168.50.97, subnet: Subnet { gateway: 192.168.50.1, mask: Mask(24) }, dns: Some(192.168.50.1), secondary_dns: Some(0.0.0.0) }
I (10928) wifi:<ba-add>idx:1 (ifx:0, fc:34:97:8a:61:c0), tid:0, ssn:1, winSize:64
I (11148) mcu_app: Connection error: Rpc(ClientError(Transport(Failed to load system certs: No valid certificate found)))
```

### More Information.

cargo run --bin mcu-app --release

sdkconfig.defaults
```text
CONFIG_ESP_MAIN_TASK_STACK_SIZE=8000

CONFIG_ESP_TASK_WDT_TIMEOUT_S=15

CONFIG_MBEDTLS_CERTIFICATE_BUNDLE=y

CONFIG_SPIRAM=y

CONFIG_SPIRAM_MODE_OCT=y
CONFIG_SPIRAM_TYPE_AUTO=y
CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY=y
CONFIG_SPIRAM_CLK_IO=30
CONFIG_SPIRAM_CS_IO=26
CONFIG_SPIRAM_SPEED_40M=y
CONFIG_SPIRAM_SPEED=40
CONFIG_SPIRAM_BOOT_INIT=y
CONFIG_SPIRAM_USE_MALLOC=y
CONFIG_SPIRAM_MEMTEST=y
CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL=16384
CONFIG_SPIRAM_MALLOC_RESERVE_INTERNAL=32768

```

MabezDev, 2024-03-15:
I'm not sure why you closed https://github.com/esp-rs/esp-idf-hal/issues/393 and posted it here without adding more context? For instance, we don't know what APIs you're using, or ones you've tried. Can you post a minimal reproduction and then we can help you.

blue-freedom-technologies, 2024-03-15:
Hi,

For my understanding, the ESP x509 Certificate Bundle API provides an way to include a bundle of custom x509 root certificates for TLS server verification.

[CONFIG_MBEDTLS_CERTIFICATE_BUNDLE=y](https://docs.espressif.com/projects/esp-idf/en/stable/esp32s2/api-reference/kconfig.html#config-mbedtls-certificate-bundle)

When enabled this option allows us to store default root certificates. I this correct ?

If yes, we should not have  this error message "Failed to load system certs: No valid certificate found".

I am doing something wrong here ? Or I need to create a custom certificate just for this to work ?

blue-freedom-technologies, 2024-03-17:
ok, after some changings we have a proper error in generating of the x509 crt bundle.

I am using this [cacert.pem](https://curl.se/ca/cacert.pem)  bundle and executing this command.

```
RUST_BACKTRACE=full cargo build --bin mcu-app --release
```

<details>

<summary>Generating x509_crt_bundle</summary>

```text
The following warnings were emitted during compilation:

warning: esp-idf-sys@0.34.1: Compiler version doesn't include clang or GCC: "./espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-cc" "--version"
warning: esp-idf-sys@0.34.1: Compiler version doesn't include clang or GCC: "c++" "--version"

error: failed to run custom build command for `esp-idf-sys v0.34.1`
note: To improve backtraces for build dependencies, set the CARGO_PROFILE_RELEASE_BUILD_OVERRIDE_DEBUG=true environment variable to enable debug information generation.

Caused by:
  process didn't exit successfully: `/mcu-app/target/release/build/esp-idf-sys-567c6d69b47f1bee/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=EXTRA-COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENT_MANAGER
  IDF_PYTHON_ENV_PATH=/mcu-app/.embuild/espressif/python_env/idf5.1_py3.10_env
  ESP_IDF_VERSION=5.1
  PATH=/mcu-app/.embuild/espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin:/mcu-app/.embuild/espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/mcu-app/.embuild/espressif/tools/cmake/3.24.0/bin:/mcu-app/.embuild/espressif/tools/ninja/1.10.2/:/mcu-app/.embuild/espressif/python_env/idf5.1_py3.10_env/bin:/mcu-app/.embuild/espressif/esp-idf/v5.1.2/tools:$PATH
  IDF_DEACTIVATE_FILE_PATH=/tmp/tmp_vw0xvzvidf_81157
  Current system platform: linux-amd64
  Skipping xtensa-esp32s3-elf@esp-12.2.0_20230208 (already installed)
  Skipping cmake@3.24.0 (already installed)
  Skipping ninja@1.10.2 (already installed)
  Skipping esp32ulp-elf@2.35_20220830 (already installed)
  IDF_PYTHON_ENV_PATH=/mcu-app/.embuild/espressif/python_env/idf5.1_py3.10_env
  ESP_IDF_VERSION=5.1
  PATH=/mcu-app/.embuild/espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin:/mcu-app/.embuild/espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/mcu-app/.embuild/espressif/tools/cmake/3.24.0/bin:/mcu-app/.embuild/espressif/tools/ninja/1.10.2/:/mcu-app/.embuild/espressif/python_env/idf5.1_py3.10_env/bin:/mcu-app/.embuild/espressif/esp-idf/v5.1.2/tools:$PATH
  IDF_DEACTIVATE_FILE_PATH=/tmp/tmp95e2ueljidf_81157
  cargo:rerun-if-changed=/mcu-app/sdkconfig.defaults
  cargo:warning=Compiler version doesn't include clang or GCC: "./espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-cc" "--version"
  cargo:warning=Compiler version doesn't include clang or GCC: "c++" "--version"
  CMAKE_PREFIX_PATH_xtensa-esp32s3-espidf = None
  CMAKE_PREFIX_PATH_xtensa_esp32s3_espidf = None
  TARGET_CMAKE_PREFIX_PATH = None
  CMAKE_PREFIX_PATH = None
  CMAKE_xtensa-esp32s3-espidf = None
  CMAKE_xtensa_esp32s3_espidf = None
  TARGET_CMAKE = None
  CMAKE = None
  running: cd "/mcu-app/target/xtensa-esp32s3-espidf/release/build/esp-idf-sys-79fed1f10b9c81ba/out/build" && CMAKE_PREFIX_PATH="" EXTRA_COMPONENT_DIRS="" IDF_COMPONENT_MANAGER="1" IDF_PATH="/mcu-app/.embuild/espressif/esp-idf/v5.1.2" IDF_TARGET="esp32s3" IDF_TOOLS_PATH="/mcu-app/.embuild/espressif" PATH="/mcu-app/.embuild/espressif/python_env/idf5.1_py3.10_env/bin:/mcu-app/.embuild/espressif/tools/ninja/1.10.2/:/mcu-app/.embuild/espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin:/mcu-app/.embuild/espressif/esp-idf/v5.1.2/tools:/mcu-app/.embuild/espressif/tools/cmake/3.24.0/bin:/mcu-app/.embuild/espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:./nvm/versions/node/v18.14.2/bin:./cargo/bin:./local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin:/Android/Sdk/emulator:/Android/Sdk/platform-tools:/opt/riscv/bin:/usr/local/bin/" PROJECT_DIR="/mcu-app" SDKCONFIG_DEFAULTS="/mcu-app/target/xtensa-esp32s3-espidf/release/build/esp-idf-sys-79fed1f10b9c81ba/out/gen-sdkconfig.defaults;/mcu-app/sdkconfig.defaults" "cmake" "/mcu-app/target/xtensa-esp32s3-espidf/release/build/esp-idf-sys-79fed1f10b9c81ba/out" "-G" "Ninja" "-DCMAKE_TOOLCHAIN_FILE=/mcu-app/.embuild/espressif/esp-idf/v5.1.2/tools/cmake/toolchain-esp32s3.cmake" "-DCMAKE_BUILD_TYPE=" "-DPYTHON=/mcu-app/.embuild/espressif/python_env/idf5.1_py3.10_env/bin/python" "-DCMAKE_INSTALL_PREFIX=/mcu-app/target/xtensa-esp32s3-espidf/release/build/esp-idf-sys-79fed1f10b9c81ba/out" "-DCMAKE_C_FLAGS= -mlongcalls -ffunction-sections -fdata-sections -mlongcalls" "-DCMAKE_CXX_FLAGS= -mlongcalls -ffunction-sections -fdata-sections" "-DCMAKE_ASM_FLAGS= -mlongcalls -ffunction-sections -fdata-sections -mlongcalls"
  -- Found Git: /usr/bin/git (found version "2.34.1") 
  -- The C compiler identification is GNU 12.2.0
  -- The CXX compiler identification is GNU 12.2.0
[... cut ...]
                                 at ./csu/../csu/libc-start.c:392:3
    25:     0x5851e0339b15 - _start
    26:                0x0 - <unknown>
warning: build failed, waiting for other jobs to finish...

```
</details>

ninja: error: '/mcu-app/target/xtensa-esp32s3-espidf/release/build/esp-idf-sys-79fed1f10b9c81ba/out/certs', needed by 'esp-idf/mbedtls/x509_crt_bundle', missing and no known rule to make it
  thread 'main' panicked at ./cargo/registry/src/index.crates.io-6f17d22bba15001f/cmake-0.1.50/src/lib.rs:1098:5:

Cheers,

blue-freedom-technologies, 2024-03-19:
This error is clearly a bug:

```text
ninja: error: '/mcu-app/target/xtensa-esp32s3-espidf/release/build/esp-idf-sys-79fed1f10b9c81ba/out/certs', needed by 'esp-idf/mbedtls/x509_crt_bundle', missing and no known rule to make it
thread 'main' panicked at ./cargo/registry/src/index.crates.io-6f17d22bba15001f/cmake-0.1.50/src/lib.rs:1098:5:
```

Because as soon as we manually copy the certificates into that path, everything goes well.

[image]

But my main error stays the same:
```text
Rpc(ClientError(Transport(Failed to load system certs: No valid certificate found)))
```

No certificate is installed in the ESP32-S3 certificate store.

blue-freedom-technologies, 2024-03-25:
Unfortunately, The [native certificate store](https://docs.rs/jsonrpsee-ws-client/latest/jsonrpsee_ws_client/struct.WsClientBuilder.html#method.use_native_rustls) crate does not support the ESP-IDF framework.

closed 2024-03-25 by blue-freedom-technologies