espressif/esp-idf #13383: Inquiry about Broadcast Interval and Message Sending Limitation in BLE Mesh Testing

lessely, 2024-03-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

idf version:v4.4.6-323-g979a56386d

I want to test the communication success rate, and since BLE mesh uses 1Mbps modulation, I need to create 10% background traffic in the network. Based on the rate of 1Mbps = 125,000 bytes/s, with a message length of 47 bytes including an 11-byte information link layer, I need to send 265 messages per second to achieve this goal.

During testing, I used the Vendor Server Model and called the `esp_ble_mesh_server_model_send_msg()` function to send undivided messages with a payload length of 11 bytes (including a 3-byte opcode, with actual data length of 8 bytes), and these were unacknowledged messages. I have already disabled the message retransmission mechanism.

I would like to know if there is a broadcast interval when I call this function consecutively twice to send messages. Additionally, during testing, I found that when using a for loop to continuously send messages, I could send a maximum of 50 messages before the sending functionality became inactive, requiring device reboot.

Thank you for your help!

lessely, 2024-03-14:
The following image shows capturing a data packet using a Bluetooth sniffer. It can be observed that a data packet can be sent within 1ms. In theory, 1000 data packets should be able to be sent within 1 second.

[image]

forx157, 2025-04-07:
Hi, @lessely 
Sorry for the long delay in getting back to you and many thanks for being patient.
Unfortunately, there is currently no support for creating such high traffic in mesh networks.