espressif/esp-idf #13384: [ble mesh] Out of segment buffers

chegewara, 2024-03-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3-dev-2087-g5f42493573

### Espressif SoC revision.

esp32S3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

S3 devkits

### Power Supply used.

USB

### What is the expected behavior?

When sent message cant be delivered and is not ACK tx buffer should be discarded maybe?

### What is the actual behavior?

When sent message cant be delivered due to node not being in range, power loss or other reason and is not ACKnowledged, trying to resend eventually leads to `BLE_MESH: Out of segment buffers` in transport layer.
From now on any action is not possible. Here i am trying to provisioning another node
```
E (937154) BLE_MESH: Out of network buffers
E (937158) BLE_MESH: Failed to send client message 0x0000803d
I (937167) mesh-cb: example_ble_mesh_cfg_client_cb, error_code = 0xffffff97, event = 0x01, addr: 0x0005, opcode: 0x803d
E (937179) mesh-cb: Send config client message failed, opcode 0x803d
```

### Steps to reproduce.

1. Provisioning node
2. turn off node simulating out of range
3. send data to vendor model - for testing i am sending very long data
4. resend data to vendor model

### Debug Logs.

```plain
some relevant logs (not all)

I (89107) BLE_MESH: Segment interval should be at least 90 (cur 60)
I (89108) BLE_MESH: Send next segment 25 after 60ms
I (89169) BLE_MESH: Send next seg 25, cred 0
I (89202) BLE_MESH: Segment interval should be at least 90 (cur 60)
I (89203) BLE_MESH: Send next segment 26 after 60ms
I (89264) BLE_MESH: Send next seg 26, cred 0
I (89297) BLE_MESH: Segment interval should be at least 90 (cur 60)
I (89298) BLE_MESH: Send next segment 27 after 60ms
I (89359) BLE_MESH: Send next seg 27, cred 0
I (89392) BLE_MESH: Segment interval should be at least 90 (cur 60)
I (89393) BLE_MESH: Send next segment 28 after 60ms
I (89454) BLE_MESH: Send next seg 28, cred 0
I (89488) BLE_MESH: All segments sent, resend after 500ms
W (89989) BLE_MESH: Ran out of retransmission to 0x0005 (0/0)
I (106011) ble-mesh-dev: ble device: vendor model found
W (106017) sendMsg: opcode: 0xc11234, len: 336, status: ESP_OK
I (106017) BLE_MESH: send, app_idx 0x0000 src 0x0001 dst 0x0005
W (106024) clients_model_action_set: missing parameters
I (106030) BLE_MESH: send, len 339: c13412536564207574207065727370696369617469732c20756e6465206f6d6e69732069737465206e61747573206572726f722073697420766f6c7570746174
I (106051) BLE_MESH: Seg, send_tag 0x01, send_szmic 0, aszmic 0
E (106064) BLE_MESH: Out of segment buffers
I (161608) ble-mesh-dev: ble device: vendor model found
W (161615) sendMsg: opcode: 0xc11234, len: 336, status: ESP_OK
I (161615) BLE_MESH: send, app_idx 0x0000 src 0x0001 dst 0x0005
W (161621) clients_model_action_set: missing parameters
I (161628) BLE_MESH: send, len 339: c13412536564207574207065727370696369617469732c20756e6465206f6d6e69732069737465206e61747573206572726f722073697420766f6c7570746174
I (161649) BLE_MESH: Seg, send_tag 0x01, send_szmic 0, aszmic 0
E (161660) BLE_MESH: Out of segment buffers
I (625301) ble-mesh-dev: ble device: vendor model found
W (625307) sendMsg: opcode: 0xc11234, len: 336, status: ESP_OK
I (625308) BLE_MESH: send, app_idx 0x0000 src 0x0001 dst 0x0005
W (625314) clients_model_action_set: missing parameters
I (625320) BLE_MESH: send, len 339: 
I (625341) BLE_MESH: Seg, send_tag 0x01, send_szmic 0, aszmic 0
E (625353) BLE_MESH: Out of segment buffers
```
```

forx157, 2024-03-18:
Hi, @chegewara 
The reason for this is due to insufficient resources, but whether this is an issue needs further verification, so could you please change the log in function `bt_mesh_adv_create_from_pool` to
```c
    BT_DBG("pool %p, buf_count %d, uinit_count %d",
            buf->pool, pool->buf_count, pool->uninit_count);
```
Change `BT_DBG` to `BT_WARN` , after that run your programme again and share the output log with me.

chegewara, 2024-03-21:
Hi @forx157 
sorry for late reply.
It is happening whenever i set vendor model publish to address different than unicast address, like group address or 0xFFFF
```
I (61876) ble-mesh-dev: ble device: model found
location event: 2, OP code: 0x8227
local location event
{"north":0,"east":0,"alt":0,"floor":0,"rssi":-55}
W (61876) BLE_MESH: pool 0x3ffc1d20, buf_count 60, uinit_count 59
W (61886) BLE_MESH: pool 0x3ffc1d20, buf_count 60, uinit_count 58
W (61896) BLE_MESH: pool 0x3ffc1d20, buf_count 60, uinit_count 57
W (61896) BLE_MESH: pool 0x3ffc1d20, buf_count 60, uinit_count 56
W (61906) BLE_MESH: pool 0x3ffc1d20, buf_count 60, uinit_count 55
W (69146) BLE_MESH: pool 0x3ffc1d20, buf_count 60, uinit_count 55
W (69296) BLE_MESH: No OpCode 0x00008227 for elem 0
generic callback event: 0x0002
I (69296) ble-mesh-dev: ble device: model found
location event: 2, OP code: 0x8227
local location event
{"north":0,"east":0,"alt":0,"floor":0,"rssi":-47}
W (69306) BLE_MESH: pool 0x3ffc1d20, buf_count 60, uinit_count 55
W (69316) BLE_MESH: pool 0x3ffc1d20, buf_count 60, uinit_count 54
W (69326) BLE_MESH: pool 0x3ffc1d20, buf_count 60, uinit_count 53
W (69326) BLE_MESH: pool 0x3ffc1d20, buf_count 60, uinit_count 52
W (69336) BLE_MESH: pool 0x3ffc1d20, buf_count 60, uinit_count 51
W (76646) BLE_MESH: pool 0x3ffc1d20, buf_count 60, uinit_count 51
```
With unicast address buffers are reclaimed/freed 

Thanks

chegewara, 2024-03-21:
And, like i mentioned in OP, it is also happening when publish address is unicast, but node is out of range and cant ACK
```
I (421796) ble-mesh-dev: ble device: model found
location event: 2, OP code: 0x8227
local location event
{"north":0,"east":0,"alt":0,"floor":0,"rssi":-62}
W (421806) BLE_MESH: pool 0x3ffc1d20, buf_count 60, uinit_count 55
W (421816) BLE_MESH: pool 0x3ffc1d20, buf_count 60, uinit_count 54
W (421826) BLE_MESH: pool 0x3ffc1d20, buf_count 60, uinit_count 53
W (421826) BLE_MESH: pool 0x3ffc1d20, buf_count 60, uinit_count 52
W (421836) BLE_MESH: pool 0x3ffc1d20, buf_count 60, uinit_count 51
W (421866) BLE_MESH: pool 0x3ffc1d20, buf_count 60, uinit_count 50
W (424636) BLE_MESH: Ran out of retransmission to 0x0006 (0/0)
W (429166) BLE_MESH: pool 0x3ffc1d20, buf_count 60, uinit_count 51
W (429286) BLE_MESH: No OpCode 0x00008227 for elem 0
generic callback event: 0x0002
I (429296) ble-mesh-dev: ble device: model found
location event: 2, OP code: 0x8227
local location event
{"north":0,"east":0,"alt":0,"floor":0,"rssi":-57}
W (429296) BLE_MESH: pool 0x3ffc1d20, buf_count 60, uinit_count 51
W (429306) BLE_MESH: pool 0x3ffc1d20, buf_count 60, uinit_count 50
W (429316) BLE_MESH: pool 0x3ffc1d20, buf_count 60, uinit_count 49
W (429326) BLE_MESH: pool 0x3ffc1d20, buf_count 60, uinit_count 48
W (429326) BLE_MESH: pool 0x3ffc1d20, buf_count 60, uinit_count 47
W (431866) BLE_MESH: pool 0x3ffc1d20, buf_count 60, uinit_count 46
W (432026) BLE_MESH: Ran out of retransmission to 0x0006 (0/0)
W (436666) BLE_MESH: pool 0x3ffc1d20, buf_count 60, uinit_count 47
```

Which i believe should not happen when publish, only when sending with ACK.

forx157, 2024-03-21:
@chegewara 
Looking through the logs, it doesn't look like there's anything wrong, the resources are in a dynamic process of allocation and free.
In addition, based on the situation you described, it should be that the segment message is sent and the segment message is turned on for retransmission due to the node being out of communication range, so the segment message will continue to take up resources until the end of the retransmission.
Maybe you can reduce the message advertising frequency to avoid the resource shortage situation.
Or you could streamline the broadcast messages to be advertising as non-packetised messages.

chegewara, 2024-03-21:
I think i dont understand something.

Why it is working when publish to unicast address and not working with the same code and publishing to group address?

Does segmented message require ACK after every segment transmission? There is no way to ACK when receiver address is other than unicast, because its against specs (which node in the group should send ACK, what if address is 0xFFFF).

Thanks

chegewara, 2024-03-21:
I continue to read thru the code. Is it not that after this the buffers should be free?
```
I (16856) BLE_MESH: Publish Retransmit Count 0 Interval 100ms
I (16866) BLE_MESH: send, app_idx 0x0000 src 0x0005 dst 0xc002
I (16876) BLE_MESH: send, len 52: c134127b226e6f727468223a302c2265617374223a302c22616c74223a302c22666c6f6f72223a302c2272737369223a2d36337d
I (16886) BLE_MESH: Seg, send_tag 0x01, send_szmic 0, aszmic 0
W (16896) BLE_MESH: pool 0x3ffc1d20, buf_count 60, uinit_count 51
W (16906) BLE_MESH: pool 0x3ffc1d20, buf_count 60, uinit_count 50
W (16906) BLE_MESH: pool 0x3ffc1d20, buf_count 60, uinit_count 49
W (16916) BLE_MESH: pool 0x3ffc1d20, buf_count 60, uinit_count 48
W (16926) BLE_MESH: pool 0x3ffc1d20, buf_count 60, uinit_count 47
I (16936) BLE_MESH: Settings store, waiting 0 seconds
I (16946) BLE_MESH: Use security credentials 0x00, proxy 0
I (16946) BLE_MESH: Network PDU, count 2, interval 20
I (16956) BLE_MESH: Segment interval should be at least 90 (cur 60)
I (16956) BLE_MESH: Send next segment 1 after 60ms
I (17026) BLE_MESH: Send next seg 1, cred 0
I (17026) BLE_MESH: Settings store, waiting 0 seconds
I (17036) BLE_MESH: Use security credentials 0x00, proxy 0
I (17036) BLE_MESH: Network PDU, count 2, interval 20
I (17056) BLE_MESH: Segment interval should be at least 90 (cur 60)
I (17066) BLE_MESH: Send next segment 2 after 60ms
I (17126) BLE_MESH: Send next seg 2, cred 0
I (17126) BLE_MESH: Settings store, waiting 0 seconds
I (17136) BLE_MESH: Use security credentials 0x00, proxy 0
I (17136) BLE_MESH: Network PDU, count 2, interval 20
I (17156) BLE_MESH: Segment interval should be at least 90 (cur 60)
I (17166) BLE_MESH: Send next segment 3 after 60ms
I (17226) BLE_MESH: Send next seg 3, cred 0
I (17226) BLE_MESH: Settings store, waiting 0 seconds
I (17236) BLE_MESH: Use security credentials 0x00, proxy 0
I (17236) BLE_MESH: Network PDU, count 2, interval 20
I (17266) BLE_MESH: Segment interval should be at least 90 (cur 60)
I (17266) BLE_MESH: Send next segment 4 after 60ms
I (17326) BLE_MESH: Send next seg 4, cred 0
I (17326) BLE_MESH: Settings store, waiting 0 seconds
I (17336) BLE_MESH: Use security credentials 0x00, proxy 0
I (17336) BLE_MESH: Network PDU, count 2, interval 20
I (17356) BLE_MESH: All segments sent, resend after 250ms
W (17616) BLE_MESH: Multicast Retransmission Count left: 1
I (17616) BLE_MESH: Resend seg 0, cred 0
I (17616) BLE_MESH: Settings store, waiting 0 seconds
I (17636) BLE_MESH: Segment interval should be at least 90 (cur 60)
I (17636) BLE_MESH: Send next segment 1 after 60ms
I (17696) BLE_MESH: Send next seg 1, cred 0
I (17696) BLE_MESH: Settings store, waiting 0 seconds
I (17726) BLE_MESH: Segment interval should be at least 90 (cur 60)
I (17736) BLE_MESH: Send next segment 2 after 60ms
I (17796) BLE_MESH: Send next seg 2, cred 0
I (17796) BLE_MESH: Settings store, waiting 0 seconds
I (17826) BLE_MESH: Segment interval should be at least 90 (cur 60)
I (17836) BLE_MESH: Send next segment 3 after 60ms
I (17896) BLE_MESH: Send next seg 3, cred 0
I (17896) BLE_MESH: Settings store, waiting 0 seconds
I (17926) BLE_MESH: Segment interval should be at least 90 (cur 60)
I (17936) BLE_MESH: Send next segment 4 after 60ms
I (17996) BLE_MESH: Send next seg 4, cred 0
I (17996) BLE_MESH: Settings store, waiting 0 seconds
I (18036) BLE_MESH: All segments sent, resend after 250ms
W (18286) BLE_MESH: Multicast Retransmission Count left: 0
I (18286) BLE_MESH: Resend seg 0, cred 0
I (18286) BLE_MESH: Settings store, waiting 0 seconds
I (18316) BLE_MESH: Segment interval should be at least 90 (cur 60)
I (18316) BLE_MESH: Send next segment 1 after 60ms
I (18376) BLE_MESH: Send next seg 1, cred 0
I (18376) BLE_MESH: Settings store, waiting 0 seconds
I (18406) BLE_MESH: Segment interval should be at least 90 (cur 60)
I (18416) BLE_MESH: Send next segment 2 after 60ms
I (18476) BLE_MESH: Send next seg 2, cred 0
I (18476) BLE_MESH: Settings store, waiting 0 seconds
I (18506) BLE_MESH: Segment interval should be at least 90 (cur 60)
I (18516) BLE_MESH: Send next segment 3 after 60ms
I (18576) BLE_MESH: Send next seg 3, cred 0
I (18576) BLE_MESH: Settings store, waiting 0 seconds
I (18606) BLE_MESH: Segment interval should be at least 90 (cur 60)
I (18616) BLE_MESH: Send next segment 4 after 60ms
I (18676) BLE_MESH: Send next seg 4, cred 0
I (18676) BLE_MESH: Settings store, waiting 0 seconds
I (18716) BLE_MESH: All segments sent, resend after 250ms
W (18966) BLE_MESH: Complete tx sdu to multicast 0xc002           <--------------- HERE
```

Should be fragments resend at all when its publish to non-unicast address? (just asking, because i have no idea, sorry)

chegewara, 2024-03-21:
@forx157 
I believe i found where the bug is and its hard to believe it (i think)

In this loop all nack buffers should be freed (line 379)
https://github.com/espressif/esp-idf/blob/master/components/bt/esp_ble_mesh/core/transport.enh.c#L379
```
    for (size_t i = 0; i <= tx->seg_n && tx->nack_count; i++) {
        if (tx->seg[i] == NULL) {
            continue;
        }

        seg_tx_done(tx, i);
    }
```

But look at line 371 (whats going on there?)
```
    tx->seg_n = 0;
```

How is the impact on loop?

forx157, 2024-03-22:
> @forx157 I believe i found where the bug is and its hard to believe it (i think)
> 
> In this loop all nack buffers should be freed (line 379) https://github.com/espressif/esp-idf/blob/master/components/bt/esp_ble_mesh/core/transport.enh.c#L379
> 
> ```
>     for (size_t i = 0; i <= tx->seg_n && tx->nack_count; i++) {
>         if (tx->seg[i] == NULL) {
>             continue;
>         }
> 
>         seg_tx_done(tx, i);
>     }
> ```
> 
> But look at line 371 (whats going on there?)
> 
> ```
>     tx->seg_n = 0;
> ```
> 
> How is the impact on loop?
> 
> This code works much better (to me)
> 
> ```
>     for (size_t i = 0; i <= tx->seg_n && tx->nack_count; i++) {
>         if (tx->seg[i] == NULL) {
>             continue;
>         }
> 
>         seg_tx_done(tx, i);
>     }
> 
>     tx->cb = NULL;
>     tx->cb_data = NULL;
>     tx->seq_auth = 0U;
>     tx->sub = NULL;
>     tx->seg_n = 0;
>     tx->last_seg_n = 0;
>     tx->lsn_updated = 0;
>     tx->dst = BLE_MESH_ADDR_UNASSIGNED;
>     tx->surc = 0;
>     tx->surwpc = 0;
>     tx->smrc = 0;
> 
>     tx->nack_count = 0U;
> ```
> 
> Thanks
@chegewara 
Yes, there is indeed a problem here, I've submitted the MR with the fix and it should be synced to github soon. Once again, thanks for the work you've put in.

forx157, 2024-03-22:
> I think i dont understand something.
> 
> Why it is working when publish to unicast address and not working with the same code and publishing to group address?
> 
> Does segmented message require ACK after every segment transmission? There is no way to ACK when receiver address is other than unicast, because its against specs (which node in the group should send ACK, what if address is 0xFFFF).
> 
> Thanks
> 
> PS maybe in that case it would be better to send message with no ACK, what do you think

@chegewara 
When advertising a segment message to a group address, receivers is not asked to reply to the segment ack, but the message will be advertised five times to ensure that the receivers can receive the message, which results in a longer resource usage time when advertising a segmented message to a group address than when the destination address is a unicast address.

chegewara, 2024-03-23:
Thanks for acknowledging it is a bug and fixing it.

ShahinHaque97-CT, 2024-03-23:
@forx157 Would you be able to post the git commit on this thread when it is completed please, 
Thank you,

chegewara, 2024-03-27:
Hi @forx157 
i dont want to open new issue every time i think i find a bug (or maybe i should?), so this time i post it here.
This condition is wrong:
https://github.com/espressif/esp-idf/blob/master/components/bt/esp_ble_mesh/api/core/esp_ble_mesh_networking_api.c#L579-L583

It should fail either source address is not unicast or dst address is wrong, but now it fails only when all 3 conditions fail.

Thanks

forx157, 2024-03-27:
@chegewara 
Thanks for the detailed bug report and letting us know about this. we will fix this soon.
And it is recommended that you create a new bug report so that it is easy for our team to track.

closed 2024-04-04 by espressif-bot