espressif/esp-idf #13385: Failed to download https://dl.espressif.com/dl/esp-idf/espidf.constraints.v4.4.tx

naveen-nayan, 2024-03-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

master

### Operating System used.

Linux

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Issue in install.sh master branch woth python 3.10.12
WARNING: Failed to download https://dl.espressif.com/dl/esp-idf/espidf.constraints.v4.4.txt 

### What is the actual behavior?

Issue in install.sh master branch woth python 3.10.12
WARNING: Failed to download https://dl.espressif.com/dl/esp-idf/espidf.constraints.v4.4.txt 

### Steps to reproduce.

1. run install.sh 
...

### Build or installation Logs.

```plain
Detecting the Python interpreter
Checking "python3" ...
Python 3.10.12
"python3" has been detected
Checking Python compatibility
Installing ESP-IDF tools
Selected targets are: esp32c3, esp32h2, esp32s3, esp32c6, esp32p4, esp32c5, esp32c61, esp32, esp32c2, esp32s2
Current system platform: linux-amd64
Installing tools: xtensa-esp-elf-gdb, riscv32-esp-elf-gdb, xtensa-esp-elf, riscv32-esp-elf, esp32ulp-elf, openocd-esp32, esp-rom-elfs
Skipping xtensa-esp-elf-gdb@12.1_20231023 (already installed)
Skipping riscv32-esp-elf-gdb@12.1_20231023 (already installed)
Skipping xtensa-esp-elf@esp-13.2.0_20230928 (already installed)
Skipping riscv32-esp-elf@esp-13.2.0_20230928 (already installed)
Skipping esp32ulp-elf@2.35_20220830 (already installed)
Skipping openocd-esp32@v0.12.0-esp32-20230921 (already installed)
Skipping esp-rom-elfs@20240305 (already installed)
Installing Python environment and packages
Python 3.10.12
pip 24.0 from /home/nayan/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/pip (python 3.10)
Upgrading pip and setuptools...
Requirement already satisfied: pip in /home/nayan/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages (24.0)
Requirement already satisfied: setuptools in /home/nayan/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages (69.2.0)
Downloading https://dl.espressif.com/dl/esp-idf/espidf.constraints.v4.4.txt
Destination: /home/nayan/.espressif/espidf.constraints.v4.4.txt.tmp
WARNING: Download failure: HTTP Error 404: Not Found
WARNING: Failed to download https://dl.espressif.com/dl/esp-idf/espidf.constraints.v4.4.txt to /home/nayan/.espressif/espidf.constraints.v4.4.txt.tmp
Downloading https://dl.espressif.com/dl/esp-idf/espidf.constraints.v4.4.txt
Destination: /home/nayan/.espressif/espidf.constraints.v4.4.txt.tmp
WARNING: Download failure: HTTP Error 404: Not Found
WARNING: Failed to download https://dl.espressif.com/dl/esp-idf/espidf.constraints.v4.4.txt to /home/nayan/.espressif/espidf.constraints.v4.4.txt.tmp
Downloading https://dl.espressif.com/dl/esp-idf/espidf.constraints.v4.4.txt
Destination: /home/nayan/.espressif/espidf.constraints.v4.4.txt.tmp
WARNING: Download failure: HTTP Error 404: Not Found
WARNING: Failed to download https://dl.espressif.com/dl/esp-idf/espidf.constraints.v4.4.txt to /home/nayan/.espressif/espidf.constraints.v4.4.txt.tmp
ERROR: Failed to download, and retry count has expired
Please make sure you have a working Internet connection.
See the help on how to disable constraints in order to work around this issue.
```

dobairoland, 2024-03-15:
How did you obtain ESP-IDF? Did you clone it from here? If yes then what is the output of `git describe`?

If you updated from v4.4 to the master branch by checking it out then you also have to fetch recent changes in the master branch including fetching the tags. The following might help (in that order).

```
git fetch
git fetch --tags
git checkout master
git merge origin/master
git describe
```

fhrbata, 2024-03-15:
@dobairoland is right, the culprit seems to be missing tags, meaning `git fetch --tags` should fix this for you. I tried deleting v5.* tags and I got the same error message. The reason is the the esp-idf version detection is primary based on `git describe` output and as a fallback we are using version from `components/esp_common/include/esp_idf_version.h`. I'm wondering if it's really necessary to use `git describe` and not to rely purely on `components/esp_common/include/esp_idf_version.h`. IIUC this should be less error prone.

igrr, 2024-03-15:
@fhrbata +1 on that, since we only need the major and minor version component in idf_tools.py, we can remove `git describe` usage and get the version only from esp_idf_version.h.

naveen-nayan, 2024-03-15:
@dobairoland 
 I First forked and cloned from my fork and it is updated from master branch
output of `git describe`  is `v4.4-dev-19146-ga7d1da94b9`

dobairoland, 2024-03-15:
Your fork doesn't have the v5.* tags: https://github.com/naveen-nayan/esp-idf/tags

fhrbata, 2024-03-15:
`$ git fetch https://github.com/espressif/esp-idf/ --tags` in your fork clone should fetch the missing tags from the `esp-idf` repo. Hopefully after this the `install.sh` should work as expected. You can later push them to your fork remote.

naveen-nayan, 2024-03-15:
@dobairoland resolved thanks