espressif/esp-idf #13386: Please document esp_rom_gpio_connect_out_signal() and explain how to disconnect a signal

kriegste, 2024-03-14:
### Is your feature request related to a problem?

I have to use the function
esp_rom_gpio_connect_out_signal()
to combine two signals. There is no real documentation for this function. In esp_rom_gpio.h it just says "Combine a peripheral signal which tagged as output attribute with a GPIO. There's no limitation on the number of signals that a GPIO can combine with."
How can I undo this connection? I do not want to reset all the config for this GPIO since it is used in a bus initialization. I need just to undo the change that esp_rom_gpio_connect_out_signal() did.

### Describe the solution you'd like.

A esp_rom_gpio_disconnect_out_signal() or a similar function.

### Describe alternatives you've considered.

I am using esp_rom_gpio_connect_out_signal() to internally connect MISO and MOSI of a SPI bus. Theoretically there's the SPI flag SPI_DEVICE_3WIRE, but this seems broken. The chips I use only have SDIO (MISO and MOSI combined) and only work with the help of esp_rom_gpio_connect_out_signal().
I need to connect a second chip to the bus, which has separate MISO and MOSI, so I need to switch between both modes.

esp-wzh, 2024-03-15:
Hi, @kriegste, you can still use this API to undo the configuration you made, just assign the signal of IO back to SIG_GPIO_OUT_IDX.
e.g. `esp_rom_gpio_connect_out_signal(io_num, SIG_GPIO_OUT_IDX, false, false);`

BTW:
here is the source code of `esp_rom_gpio_connect_out_signal`:

```
/*
 * set signal output to gpio
 * one signal can output to several gpios
 * If signal_idx == SIG_GPIO_OUT_IDX, cancel output put to the gpio
 */
void gpio_matrix_out(uint32_t gpio, uint32_t signal_idx, bool out_inv, bool oen_inv)
{
    uint32_t addr = GPIO_FUNC0_OUT_SEL_CFG_REG + (gpio * 4);
    uint32_t pin_reg;
    uint32_t value = signal_idx << GPIO_FUNC0_OUT_SEL_S;

    if (gpio >= GPIO_PIN_COUNT) {
        return;
    }

    if (out_inv)
        value |= GPIO_FUNC0_OUT_INV_SEL;
    if (oen_inv)
        value |= GPIO_FUNC0_OEN_INV_SEL;
    REG_WRITE(addr, value);

    if (gpio < 32) {
        pin_reg = (1 << gpio);
        GPIO_REG_WRITE(GPIO_ENABLE_W1TS_REG, pin_reg);
    } else {
#if MAX_GPIO_NUM >= 32
        pin_reg = (1 << (gpio - 32));
        GPIO_REG_WRITE(GPIO_ENABLE1_W1TS_REG, pin_reg);
#endif
    }
}
```

ESP-Marius, 2024-08-19:
Marking this as closed as there doesnt seem to be any outstanding issues.

closed 2024-08-19 by ESP-Marius