espressif/esp-idf #13389: Synchronizing 2 I2S controllers with one RX microphone each

PrathamG, 2024-03-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi! I have one microphone working in mono mode connected to each of the I2S controllers on ESP32S3. I want these microphones to capture audio synchronously, meaning start collecting on the same channel at the same time. What options are possible to collect the I2S samples time synchronized?

L-KAYA, 2024-03-19:
For the generic case, you can set one controller to master and another to slave, so that all four I2S devices (2 mic and 2 I2S controllers) can share a same clock (i.e. BCLK and WS that generated by  the master I2S controller).
e.g.,

```
    ESP                        MIC1          
┌───────────┐            ┌──────────────────┐
│           │            │                  │
│    M_MCLK ├──┬─────────┤ MCLK             │
│           │  │         │                  │
│    M_BCLK ├──┼──┬──────┤ BCLK             │
│           │  │  │      │                  │
│      M_WS ├──┼──┼──┬───┤ WS               │
│           │  │  │  │   │                  │
│     M_DIN ├──┼──┼──┼───┤ DOUT             │
│           │  │  │  │   └──────────────────┘
│           │  │  │  │                       
│           │  │  │  │                       
│           │  │  │  │         MIC1          
│           │  │  │  │   ┌──────────────────┐
│           │  │  │  │   │                  │
│    S_MCLK ├──┴──┼──┼───┤ MCLK             │
│           │     │  │   │                  │
│    S_BCLK ├─────┴──┴───┤ BCLK             │
│           │            │                  │
│      S_WS ├────────────┤ WS               │
│           │            │                  │
│     S_DIN ├────────────┤ DOUT             │
└───────────┘            └──────────────────┘
```

For some specific codec (like [es7210](http://www.everest-semi.com/pdf/ES7210%20PB.pdf)), multiple codecs can be chained together and use TDM to read all the channels of these codecs.

e.g.,

```
    ESP                            
┌───────────┐         ┌───────────┐
│           │ optional│           │
│      MCLK ├─┬───────┤ MCLK      │
│           │ │       │           │
│      BCLK ├─┼─┬─────┤ BCLK      │
│           │ │ │     │           │
│        WS ├─┼─┼─┬───┤ WS        │
│           │ │ │ │   │           │
│       DIN ├─┼─┼─┼───┤ DOUT      │
│           │ │ │ │   │           │
└───────────┘ │ │ │ ┌─┤ DIN       │
              │ │ │ │ └───────────┘
              │ │ │ │              
              │ │ │ │ ┌───────────┐
              │ │ │ │ │           │
              └─┼─┼─┼─┤ MCLK      │
                │ │ │ │           │
                └─┼─┼─┤ BCLK      │
                  │ │ │           │
                  └─┼─┤ WS        │
                    │ │           │
                    └─┤ DOUT      │
                      │           │
                      └───────────┘
```

PrathamG, 2024-03-20:
Hi @L-KAYA , Thank you for your helpful response! In my setup I'm using INMP441 mics which only have BCLK, WS, and DOUT pins. Would the setup you provided above still work?

L-KAYA, 2024-03-21:
Yes, I checked `INMP441` datasheet, it has no MCLK requirement. So MCLK still optional, if you want to use MCLK, then you can only connect the `M_MCLK` and `S_MCLK` together and don't forget to choose `i2s_std_clk_config_t::clk_src` to `I2S_CLK_SRC_EXTERNAL` and set the `i2s_std_clk_config_t::ext_clk_freq_hz` for the slave I2S controller.

Except MCLK, you still need to connect **ALL** BCLK and WS together.

PrathamG, 2024-03-22:
Oh I'm using ESP32S3. There isn't an option to use I2S_CLK_SRC_EXTERNAL for i2s_std_clk_config_t::clk_src.

L-KAYA, 2024-03-22:
Oh sorry, `I2S_CLK_SRC_EXTERNAL` should be a `IDF v5.2` feature. If you are not convenient to migrate your IDF version, you can take a try without MCLK first, it should work as well if your sample rate is not too high (for example, higher than 48KHz)

PrathamG, 2024-03-22:
Okay understood. Thank you very much for your help!

closed 2024-03-22 by PrathamG

xiqwertyui, 2024-09-09:
I wonder if here exist a typo, 
    ```
ESP                        MIC1          
┌───────────┐            ┌──────────────────┐
│           │            │                  │
│    M_MCLK ├──┬─────────┤ MCLK             │
│           │  │         │                  │
│    M_BCLK ├──┼──┬──────┤ BCLK             │
│           │  │  │      │                  │
│      M_WS ├──┼──┼──┬───┤ WS               │
│           │  │  │  │   │                  │
│     M_DIN ├──┼──┼──┼───┤ DOUT             │
│           │  │  │  │   └──────────────────┘
│           │  │  │  │                       
│           │  │  │  │                       
│           │  │  │  │         MIC1          
│           │  │  │  │   ┌──────────────────┐
│           │  │  │  │   │                  │
│    S_MCLK ├──┴──┼──┼───┤ MCLK             │
│           │     │  │   │                  │
│    S_BCLK ├─────┴──┴───┤ BCLK             │
│           │            │                  │
│      S_WS ├────────────┤ WS               │
│           │            │                  │
│     S_DIN ├────────────┤ DOUT             │
└───────────┘            └──────────────────┘

```
Should the M_WS connect to S_BCLK instead of S_WS?

L-KAYA, 2024-09-09:
> I wonder if [here](https://github.com/espressif/esp-idf/issues/13389#issuecomment-2006120249) exist a typo,
> 
> Should the M_WS connect to S_BCLK instead of S_WS?

Ah yes!