espressif/esp-idf #13390: WIPI AP not allowing a device to latch to it (the config and setup returns true

meakashrao, 2024-03-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1

### Espressif SoC revision.

ESP32-WROOM-32E-N8 (revision v3.0 and V3.1)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP-WROOM-32

### Power Supply used.

USB

### What is the expected behavior?

The initial firmware was developed for a product on Platform IDE and later we ported the same to ESP IDF, but since products were in the field and we used ArduinoOTA for updating the firmware in the field using WiFi AP, we used Arduino as an IDF component to be able to do this in the IDF as well

I was expecting the WiFi AP to behave exactly the same.

```c
#include <Arduino.h>
#include <WiFi.h>
#include <WiFiClient.h>
#include <WebServer.h>
#include <Update.h>
#include <EEPROM.h>
#include <ArduinoOTA.h>
#include <WiFiUdp.h>
#include <ESPmDNS.h>
#include "esp_log.h"

const char *host = "esp32";
const char *ssid_ap = "ReactHardwareCB";
const char *password_ap = "IfTzeitelMarriesLazarwolf";

static const char *TAG_O = "OTA";

void OTA_Handler::begin()
{
  ESP_LOGI(TAG_O, "OTA Begin");
  delay(5);

  IPAddress local_ip(192, 168, 1, 1);
  IPAddress local_mask(255, 255, 255, 0);
  IPAddress gateway(192, 168, 1, 1);

  bool apConfigSuccess = WiFi.softAPConfig(local_ip, gateway, local_mask);
  ESP_LOGI(TAG_O, "AP IP Config success: %s", apConfigSuccess ? "true" : "false");

  bool apStartSuccess = WiFi.softAP(ssid_ap, NULL);
  ESP_LOGI(TAG_O, "AP Start success: %s", apStartSuccess ? "true" : "false");

  ESP_LOGI(TAG_O, "AP SSID: %s", WiFi.softAPSSID().c_str());
  ESP_LOGI(TAG_O, "IP address: %s", WiFi.softAPIP().toString().c_str());
  ESP_LOGI(TAG_O, "Subnet Mask: %s", WiFi.softAPSubnetMask().toString().c_str());
  ESP_LOGI(TAG_O, "Broadcast IP address: %s", WiFi.softAPBroadcastIP().toString().c_str());
  ESP_LOGI(TAG_O, "Network ID: %s", WiFi.softAPNetworkID().toString().c_str());
  
  delay(15);
  ArduinoOTA
      .onStart([]()
               {
    String type;
    if (ArduinoOTA.getCommand() == U_FLASH)
        type = "sketch";
      else // U_SPIFFS
        type = "filesystem";

      // NOTE: if updating SPIFFS this would be the place to unmount SPIFFS using SPIFFS.end()
      Serial.println("Start updating " + type); })
      .onEnd([]()
             {
      Serial.println("\nEnd");
      EEPROM.write(0, 0);
      EEPROM.commit();
      ESP.restart(); })
      .onProgress([](unsigned int progress, unsigned int total)
                  { Serial.printf("Progress: %u%%\r", (progress / (total / 100))); })
      .onError([](ota_error_t error)
               {
      Serial.printf("Error[%u]: ", error);
      if (error == OTA_AUTH_ERROR) Serial.println("Auth Failed");
      else if (error == OTA_BEGIN_ERROR) Serial.println("Begin Failed");
      else if (error == OTA_CONNECT_ERROR) Serial.println("Connect Failed");
      else if (error == OTA_RECEIVE_ERROR) Serial.println("Receive Failed");
      else if (error == OTA_END_ERROR) Serial.println("End Failed"); });

  ArduinoOTA.begin();
  ArduinoOTA.setPort(3232);
  ArduinoOTA.setPassword("admin");
}

void OTA_Handler::loop()
{
  ArduinoOTA.handle();
}
``` 

The code int's and setup the correct AP and other stuff and that is evident by the following Image
[image]
So, I verified that the SSID, IP Address, Subnet Mask and others are set accurately.

### What is the actual behavior?

The Laptop tries to connect to the AP(evident from join from wifi library) but then fails to latch(evident from leave from wifi library)

I also need help to properly understand the meaning of AID, bgn, bss_flags and bss in the image above.

### Steps to reproduce.

As said initially, we ported from Platform IO to ESP IDF(both of them done using VS Code IDE)

1. Upload Firmware version developed with Platform IO
2. Upload second Firmware version developed with Platform IO (both 1 and 2 are inter changeable as we have did more than 3 versions of the Platform IO Firmware and we were able to switch between them on the field easily)
3. Upload the ESP IDF Build, changing from Platform IO to ESP IDF gets successfully updated.
4. Jumping back to Platform IO Build(any version) or to a second ESP IDF build (to verify OTA) does not works properly

### More Information.

I also erased the whole flash of the ESP Module and then I was able to connect from my Laptop to the ESP IDF built code and jump between multiple builds of ESP IDF firmware

The logs for the same below:

I (1241) wifi:mode : softAP (c0:49:ef:bf:eb:b5)
I (1246) wifi:Total power save buffer number: 16
I (1247) wifi:Init max length of beacon: 752/752
I (1248) wifi:Init max length of beacon: 752/752
I (1252) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.1.1
I (1250) OTA: AP IP Config success: true
I (1529) wifi:Total power save buffer number: 16
I (1530) OTA: AP Start success: true
I (1530) OTA: AP set to ReactHardwareCB
I (1532) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.1.1
I (1541) OTA: IP address: 192.168.1.1
I (1576) Main: OTA_Setup Done
I (16764) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1
I (16765) wifi:station: 70:cf:49:87:e8:ee join, AID=1, bgn, 40U
I (16985) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.1.2
I (17107) wifi:idx:2 (ifx:1, 70:cf:49:87:e8:ee), tid:0, ssn:21, winSize:64

Also, some reading gave me the idea the WiFi SSID and Password are saved in NVS, am I right? is the saved ID and Password getting corrupted?

mhdong, 2024-03-18:
Hi @meakashrao 
https://github.com/espressif/esp-idf/blob/6b9c2fca79a7b77a42242d48b0cfa82d612e9b85/components/esp_wifi/include/esp_wifi_types_generic.h#L299
Do not support AUTH_WEP, AUTH_WAPI_PSK and AUTH_OWE in soft-AP mode.

If it has nothing to do with WAPI and is only a problem of disconnection, could you provide packet capture for successful connection and unsuccessful connection separately?

mhdong, 2024-03-25:
Hi @meakashrao 
Sorry for mistaking WiFi for WAPI encryption method.
Is it convenient for you to use IDF's menuconfig for configuration?
Please set ESP_WIFI_DEBUG_PRINT to y and LOG_DEFAULT_LEVEL_DEBUG to Debug
idf.py menuconfig -> Component config -> Wi-Fi -> Print debug messages from WPA Supplicant
idf.py menuconfig -> Component config -> Log output -> Default log verbosity -> Debug

AxelLin, 2024-05-19:
@meakashrao Could you provide debug log based on the comment in https://github.com/espressif/esp-idf/issues/13390#issuecomment-2017136827 ?

meakashrao, 2024-05-21:
This has been discontinued for now

closed 2024-05-21 by meakashrao